mavlink/mavros #1826: RTT To High for Timesync(Trying to send messages using ROS Topics)

EmirhanKeles, 2023-02-09:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
Hi. I am trying to build a simple simulation using ROS Noetic, MAVROS and ArduPilot for fixed wing UAVs. I am trying to send messages for example velocity, acceleration change, using mavros topics but i dont give any reactions from simulation. I think the error RTT to high for timesync is related to this problem because i can send messages using ROS services perfectly. How to solve this problem and connect simulation normally.

Thanks a lot.

### MAVROS version and platform

Mavros: Latest version for ROS Noetic
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: 4.0.4

### Node logs

```
attached all log files to end of the text
```

### Diagnostics

```
header: 
  seq: 56
  stamp: 
    secs: 1675944633
    nsecs: 779404227
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Received packets:"
        value: "6304"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "46"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "217040"
      - 
        key: "Tx total bytes:"
        value: "21133"
      - 
        key: "Rx speed:"
        value: "4001.000000"
      - 
        key: "Tx speed:"
        value: "320.000000"
  - 
    level: 0
    name: "mavros: GPS"
    message: "3D fix"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Satellites visible"
        value: "10"
      - 
        key: "Fix type"
        value: "6"
      - 
        key: "EPH (m)"
        value: "1.21"
      - 
        key: "EPV (m)"
        value: "2.00"
  - 
    level: 1
    name: "mavros: Mount"
    message: "Can not diagnose in this targeting mode"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Mode"
        value: "255"
  - 
    level: 0
    name: "mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "63"
      - 
        key: "Frequency (Hz)"
        value: "1.039967"
      - 
        key: "Vehicle type"
        value: "Fixed wing aircraft"
      - 
        key: "Autopilot type"
        value: "ArduPilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
  - 
    level: 0
    name: "mavros: System"
    message: "Normal"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Sensor present"
        value: "0x1361FC3F"
      - 
        key: "Sensor enabled"
        value: "0x1261803F"
      - 
        key: "Sensor health"
        value: "0x1371803F"
      - 
        key: "3D gyro"
        value: "Ok"
      - 
        key: "3D accelerometer"
        value: "Ok"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "differential pressure"
        value: "Ok"
      - 
        key: "GPS"
        value: "Ok"
      - 
        key: "motor outputs / control"
        value: "Ok"
      - 
        key: "rc receiver"
        value: "Ok"
      - 
        key: "AHRS subsystem health"
        value: "Ok"
      - 
        key: "Terrain subsystem health"
        value: "Ok"
      - 
        key: "Battery"
        value: "Ok"
      - 
        key: "pre-arm check status. Always healthy when armed"
        value: "Ok"
      - 
        key: "CPU Load (%)"
        value: "0.0"
      - 
        key: "Drop rate (%)"
        value: "0.0"
      - 
        key: "Errors comm"
        value: "0"
      - 
        key: "Errors count #1"
        value: "0"
      - 
        key: "Errors count #2"
        value: "0"
      - 
        key: "Errors count #3"
        value: "0"
      - 
        key: "Errors count #4"
        value: "0"
  - 
    level: 0
    name: "mavros: Battery"
    message: "Normal"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Voltage"
        value: "12.59"
      - 
        key: "Current"
        value: "-0.0"
      - 
        key: "Remaining"
        value: "100.0"
  - 
    level: 0
    name: "mavros: Time Sync"
    message: "Normal"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "630"
      - 
        key: "Frequency (Hz)"
        value: "9.999619"
      - 
        key: "Last RTT (ms)"
        value: "8.225267"
      - 
        key: "Mean RTT (ms)"
        value: "13.983896"
      - 
        key: "Last remote time (s)"
        value: "72.252012000"
      - 
        key: "Estimated time offset (s)"
        value: "1675944561.517242908"
  - 
    level: 0
    name: "mavros: APM Memory"
    message: "Normal"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Free memory (B)"
        value: "131072"
      - 
        key: "Heap top"
        value: "0x0000"
  - 
    level: 2
    name: "mavros: APM Hardware"
    message: "Not initialised"
    hardware_id: "udp://127.0.0.1:14551@14555"
    values: 
      - 
        key: "Core voltage"
        value: "-1.000000"
      - 
        key: "I2C errors"
        value: "0"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 1829 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 129, 2, 136, 137, 11020, 147, 11030, 22, 152, 24, 27, 29, 30, 32, 33, 163, 36, 164, 168, 42, 49, 178, 65, 193, 73, 74, 111, 241, 242, 116, 125

```

[image]

[logs.zip](https://github.com/mavlink/mavros/files/10696950/logs.zip)

amilcarlucas, 2023-02-09:
Ignore the "RTT too high for timesync: XX ms." messages.

EmirhanKeles, 2023-02-09:
I will try right away. thanks a lot

closed 2026-10-21 by github-actions[bot]