mavlink/mavros #1828: how to send reset_counter through mavros odometry

shubham-shahh, 2023-02-14:
----

### Issue details

Hi, 
I am willing to send SLAM odometry to FCU through `/mavros/odometry/out` topic, since it is of type `<nav_msgs::Odometry>` I am unable to send the resent counter which is part of the odometry mavlink message definition as mentioned [here](https://mavlink.io/en/messages/common.html#ODOMETRY). how can I send the resent counter

### MAVROS version and platform

Mavros: ?1.13.0?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: ?4.1.5?

thanks

vooon, 2023-02-14:
I'm afraid that's not possible to achieve with nav_msgs directly. But possible to define new message with additional fields, then use message filters to have synchronised call to out.

https://github.com/ros2/message_filters

shubham-shahh, 2023-02-16:
> I'm afraid that's not possible to achieve with nav_msgs directly. But possible to define new message with additional fields, then use message filters to have synchronised call to out.
> 
> https://github.com/ros2/message_filters

thanks for the response, 
I am currently using ROS1, I was wondering if, we can add reset count in 
`string child_frame_id` field if it is not used further for transformations?

vooon, 2023-02-16:
Message filters do exist in ROS1 as well.

I do not recommend to change frame ids, as it potentially may violate expectations of other components.

shubham-shahh, 2023-02-16:
> Message filters do exist in ROS1 as well.
> 
> I do not recommend to change frame ids, as it potentially may violate expectations of other components.

yes, I misunderstood that part, I later realized it. and yes it makes sense, I will try the way you mentioned

thanks

shubham-shahh, 2023-02-16:
I was wondering, how can I set/check source system id for a mavros instance running on on board computer

vooon, 2023-02-16:
https://github.com/mavlink/mavros/blob/master/mavros/src/lib/mavros.cpp#L58

closed 2023-02-17 by shubham-shahh