mavlink/mavros #1830: is there a way to listen NAMED_VALUE_FLOAT through mavros?

shubham-shahh, 2023-02-16:
### Issue details

is there a way to listen to [NAMED_FLOAT](https://mavlink.io/en/messages/common.html#NAMED_VALUE_FLOAT) through mavros?

### MAVROS version and platform

Mavros: 1.14.0 (can move to latest version available for ROS1)
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: 4.1.5

vooon, 2023-02-16:
https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/debug_value.cpp#L221

shubham-shahh, 2023-02-16:
> https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/debug_value.cpp#L221

Hi,
Thanks for the info, how can I enable this?

shubham-shahh, 2023-02-16:
> > https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/debug_value.cpp#L221
> 
> Hi, Thanks for the info, how can I enable this?

I was able to enable it from the plugin list, thanks again

closed 2023-02-16 by shubham-shahh