mavlink/mavros #1837: HIGHRES_IMU stream rate not going over 182Hz

ilitev, 2023-03-06:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hello,
I have an issue where when i try to increase the HIGHRES_IMU data rate to 200Hz it only goes up to 182Hz. With older px4 version (1.12) i get 200Hz stable rate. Any suggestions?

### MAVROS version and platform

Mavros: Latest
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: 1.13

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

FaboNo, 2023-03-18:
@ilitev Sorry to bother you but may I ask you how did you increase the IMU data rate at 180Hz? 
I am running PX 1.13.2 on a Pixhawk 6X mini

I tried to use the following command
`rosrun mavros mavsys rate --all 100`

But it did not change the rate

I used the trick to write on the SD (/etc/extras.txt):
```
mavlink stream -d /dev/ttyUSB0 -s ATTITUDE_QUATERNION -r 100
mavlink stream -d /dev/ttyUSB0 -s HIGHRES_IMU -r 100
```
But it did not work either

So I am out of option ... and open to suggestions

ilitev, 2023-03-18:
Hello,
You can use the same commands you wrote in the QGroundcontrol console and it should work but you will have to do it every time your flight controller restarts

I did it permanently by changing some source code and rebuilding the firmware. You can do that by git clone the repository of px4, go in to the folder, then src/modules/mavlink and open the mavlink_main.cpp, in there you can ctlr+f and search for HIGHRES_IMU. When you change to the desired number you save exit and go to back to the PX4 folder in a terminal and type “make <your board> “ and it should build new firmware. That firmware you will find in the build folder in PX4. The. You can flash your flight controller with QGroundcontrol and you should have higher rate.

Hope it helps

ilitev, 2023-03-21:
> 

The mavlink analyzer in QGC shows the same rate aprox.185Hz. With the old firmware (1.12) the SD card with the etc/extras.txt works but not with the newer firmware. What i tried now is to change the rate in the firmware (modules/mavlink/mavlink_main.cpp) to 200Hz but no luck.

closed 2026-10-21 by github-actions[bot]