mavlink/mavros #1841: ROS2 humble Ardupilot unsupported Autopilot_version

chngdickson, 2023-03-13:
### Issue details

Mavros not working with the current Ardupilot version, FCU don't support version request and no messages are being published to mavros

### MAVROS version and platform

Mavros: v0.5.0
ROS2: Humble
Ubuntu: ?22.04?

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: master

### Node logs

```
![image](https://user-images.githubusercontent.com/20728313/224721908-6d07623b-96df-4f35-9223-6231ed820351.png)

```

closed 2023-03-13 by chngdickson

ZhongmouLi, 2023-11-12:
In fact, I meet this problem in SITL.

@chngdickson , did you find a solution for this?

chngdickson, 2023-11-14:
> In fact, I meet this problem in SITL.
> 
> @chngdickson , did you find a solution for this?
> 

Yes i have. Its actually related to the **understanding** of what is an FCU. Theoretically, we should only have one FCU running on 1 pc.

List of FCU : Mavros, QGroundControl, MissionPlanner.

Running any of them concurrently on one pc will cause issues. Mavros is sort of friends with QGroundControl. so they will allow it.

However, what we really want is the capability to run MissionPlanner and Mavros concurrently.

The solution?
Run mavros and MissionPlanner with 2 different ports with each connecting to different computers. 

Connections
- Mavros: _Uart to tty_ with a **companion computer** connected directly to Pixhawk/cube.
- MissionPlanner: _Telemetry_ connected with a **laptop** with telemetry.