mavlink/mavros #1843: QGC can't display full interface under UDP connection (”MAVConnUDP::send_message: TX queue overflow“) and can't read x & y position from FCU

YX-G1881, 2023-03-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----
Hi, guys! I'm a beginner with px4 & mavros and this is my first time asking for help here, so I'm sorry if i didn't follow the rules.

### Issue details

I'm using CUAV V5+ FCU with Nvidia Jetson Nano 4GB on the drone, and QGroundControl on my laptop. The FCU communicates with the Nano board via serial port (only GND, Tx, and Rx are connected).  The Nano board communicates with QGC on the laptop via UDP connection. 

The first problem is that the QGC can't display the full interface when I run _roslaunch mavros px4.launch fcu_url:="/dev/ttyTHS1:921600" gcs_url:="udp://@192.168.3.43"_ (192.168.3.43 is my laptop ip). It says "vehicle1 did not respond to the request for parameters". Sometimes the full interface shows up in QGC but I can't change any value of parameters or restart the FCU via QGC.  And it keeps saying "Communication lost" and "Data link lost". Also, I often get `DROPPED Message-Id xx IDs: 1.1 Seq xx: MAVConnUDP::send_message: TX queue overflow` in the mavros terminal. If I use USB to connect the FCU and the laptop, everything seems alright in QGC. I can change the value of the parameters and restart the FCU via QGC. 

My settings in QGC (v4.0.11) are:
MAV_0_CONFIG    TELEM 1   
MAV_1_CONFIG    TELEM 2
MAV_1_FORWARD    1
MAV_1_MODE    Onboard 
MAV_1_RATE    0  B/s
SER_TEL2_BAUD    921600  8N1

I have tried the MAV_1_RATE with 80000 B/s and other values, but I still got the ”MAVConnUDP::send_message: TX queue overflow“ error eventually. It is just a matter of time. I am using EW-7811UN wireless network equipment on one of the Nano board USB ports and I think the wifi is ok. And in QGC, the loss rate of mavlink is low (usually below 5%). Is this a network quality problem?

Another problem is that I am able to read the imu data in either QGC MAVLink Inspector or Nano board terminal. The velocity x, y, and z is not zero (move the drome with my hands, not flying) but meanwhile, the position x and y remain zero (position z is measured by a range sensor and it works fine) or sometimes position x and y have values but it seems they are not matched with the velocity values (looks like have a delay or stuck). Is this a hardware issue or a problem with the wrong mavros configuration?

### MAVROS version and platform

Mavros: 1.5.0
ROS: melodic
Ubuntu: 18.04LTS

### Autopilot type and version

[  ] ArduPilot
[ √ ] PX4

Version: v1.9.2
(firmware: px4_fmu_v5_default.px4)

### Node logs

```
roslaunch mavros px4.launch fcu_url:="/dev/ttyTHS1:921600" gcs_url:="udp://@192.168.3.43"
... logging to /home/bcsh/.ros/log/a02a375a-c666-11ed-9ca7-1e3ef778179d/roslaunch-bcsh-desktop-10661.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.3.24:45391/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
[... 445 lines cut ...]
        value: "3.890858"
      - 
        key: "Last remote time (s)"
        value: "5856.937410000"
      - 
        key: "Estimated time offset (s)"
        value: "1679233360.536698818"
---

```

### Check ID

```
rosrun mavros checked
OK. I got messages from 1:1.

---
Received 5331 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 140, 141, 147, 4, 30, 31, 32, 36, 74, 331, 76, 83, 340, 230, 105, 111, 241, 245, 253

```

I'm stuck with the issue for over a week, please help me! Thanks!

vooon, 2023-03-20:
Well, that's really unexpected to happen on UDP links. Could you try to run without gcs_url?
And can you please try to use latest version built from source?

YX-G1881, 2023-03-21:
> Well, that's really unexpected to happen on UDP links. Could you try to run without gcs_url? And can you please try to use latest version built from source?

Hello, @vooon! Thanks for your reply. 
I tried to run without gcs_url just now. Here are the results:

### Node logs 

```
roslaunch mavros px4.launch fcu_url:="/dev/ttyTHS1:921600"
... logging to /home/bcsh/.ros/log/cf737ee6-c78c-11ed-87d5-1e3ef778179d/roslaunch-bcsh-desktop-13995.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.3.24:32803/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyTHS1:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
[... 996 lines cut ...]
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
```

### Check ID
```
rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 4682 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 132, 8, 140, 141, 147, 22, 4, 411, 30, 31, 36, 74, 331, 83, 230, 105, 111, 241, 245
```

By the way, I also tried another cable for the telem2 port (FCU - Nano board) but that didn't help.  What does this `mavconn: serial0: recv: v2.0 !CRC Message-Id: 30 [28 bytes] IDs: 1.1 Seq: 219` debug info mean? Is it something wrong with the FCU hardware?

closed 2026-10-20 by github-actions[bot]