mavlink/mavros #1844: Manual Control doesn't work

thunderstorm010, 2023-03-20:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_
Manual control doesn't work. I can check rostopic echo /mavros/manual_control/send, my app sends messages and when i start mavros i see nothing changing

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: Kinetic
Ubuntu: 22.04 (RoboStack)

### Autopilot type and version

[+] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```

... logging to /home/thunderstorm010/.ros/log/7e9a733a-c73c-11ed-8056-ac74b1f1ea14/roslaunch-pop-os-16126.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pop-os:34709/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
[... 416 lines cut ...]
      - 
        key: "Mean RTT (ms)"
        value: "5.959974"
      - 
        key: "Last remote time (s)"
        value: "2887.001744000"
      - 
        key: "Estimated time offset (s)"
        value: "1679328158.115167856"
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 183 messages, from 2 addresses
sys:comp   list of messages
  1:1     0, 111
 255:230   0, 66

```

LiangHuangBC, 2023-10-19:
same issue here with sailboat, ros2

pavlovator, 2024-05-31:
Hi, did you manage to solve the issue?

thunderstorm010, 2024-06-01:
Nope, we just stopped using ROS.

closed 2026-10-20 by github-actions[bot]