mavlink/mavros #1845: Drone jerks in flight when MAV_SYS_ID different of 1

jfaure6, 2023-03-21:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

When I control my drone using mavros with a python script, everything works as expected when I enter in OFFBOARD mode. But when I want to add a new drone with a different MAV_SYS_ID (2,3 or 4 as I have 4 drones), the drone enters in OFFBOARD Mode but starts to behave strangely in flight, with jerks. Has anyone ever faced this scenario?
I can work around the problem by running mavros on each onboard computer of each drone and setting MAV_SYS_ID to 1 for all but that prevents me from using QGC to track the drones at the same time.

### MAVROS version and platform

Mavros: ?1.15.0?
ROS: ?Melodic?
Ubuntu: ?18.04?

### Autopilot type and version

[ x ] PX4

Version: ?1.12.3?

vooon, 2023-03-22:
What about sysid configuration for mavros?

vooon, 2023-03-22:
I suggest first to try run only one system with different ID.

closed 2026-10-20 by github-actions[bot]