mavlink/mavros #1847: Undefined behaviour with Mavros v1.15.0

shubham-shahh, 2023-03-31:
### Issue details

I have noticed that if I run the following command
`rosservice call /mavros/set_stream_rate 0 10 1` suddenly it triggers the `/move_base_simple/goal` topic. this is not happening if I use mavros version 1.13.0. now I am unable to install through apt, is there a way to install 1.13.0 through apt?

### MAVROS version and platform

Mavros: 1.15.0
ROS: Melodic
Ubuntu: 18.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 4.1.5

vooon, 2023-03-31:
That sounds like something really nasty happening. I'm unsure if you can find old debs, so suppose to build from source.

shubham-shahh, 2023-04-01:
> That sounds like something really nasty happening. I'm unsure if you can find old debs, so suppose to build from source.

Oh okay, do you know of any possible cause for this or any debug steps I could use to find out how this is happening? And what's the latest and stable release for ros melodic? 

Thanks

ZeeshanshariffNS, 2023-05-22:
Hi @vooon @shubham-shahh 
I'm facing the same issue as well with mavros 1.15
I would like to add that i found that mavros publishes to /move_base_simple/goal once the EK3 IMU0 and IMU1 origin is set and it publishes huge value of -3269205.28506, -3189791.07343 along x and y 

[image]
[image]

closed 2026-10-20 by github-actions[bot]