mavlink/mavros #1855: Possible error in imu.cpp plugin code

Alvar007, 2023-04-22:
Hello, I think there may be an error in the code that handles the scaled IMU message. In the mavlink website you can see that the scaled IMU message contains the magnetic field data in mgauss. But the code for the plugin uses a coeff of 1000.0 and it's called MILLIT_TO_TESLA, as it if was interpreting that the data was in mtesla instead of mgauss.

So naturally when I check the topic on ROS I get my magnetic field data in mgauss multiplied by 1000.0, which is clearly wrong as it's supposed to show in Tesla according to the ROS documentation.

Also I should mention that in the same code the _publish_imu_data_raw_ function is used for the scaled IMU instead of the _publish_imu_data_ one that is also there. So when I check my topics I have to go to _/mavros/imu/data_raw_ when it should be _/mavros/imu/data_.

If this is indeed an error I would appreciate that it would get fixed in the source as we are trying to avoid to create our own package for the project I'm working on.

Thanks in advance for your consideration.

venabled, 2024-03-14:
Second this, it appears to be converting the wrong way.

vooon, 2024-03-15:
Perhaps that's some ancient leftover or early RAW_IMU usage. Should be corrected to `MILLIG_TO_TESLA = 1e-7`.

On today's code APM presents mgauss:
- https://github.com/ArduPilot/ardupilot/blob/533eb73622b11bc0d4841ae5268c05079bcb33dc/libraries/GCS_MAVLink/GCS_Common.cpp#L2098
- https://github.com/ArduPilot/ardupilot/blob/533eb73622b11bc0d4841ae5268c05079bcb33dc/libraries/AP_Compass/AP_Compass.h#L158

As for use `publish_imu_data_raw` - it's correct. Your imu data should be combined from ATTITUDE/ATTITUDE_QUATERNION and *_IMU.

You're free to open a MR.