mavlink/mavros #1860: ROS 2 geographic_msgs issue

dottantgal, 2023-05-02:
### Issue details

When I launch my own mavros 2 launch file I get this error on my platform
`symbol lookup error: /home/nvidia/ros2_ws/install/mavros_msgs/lib/libmavros_msgs__rosidl_typesupport_fastrtps_cpp.so: undefined symbol: _ZN15geographic_msgs3msg24typesupport_fastrtps_cpp28max_serialized_size_GeoPointERbm `
This is the launch file
```
<launch>
	<!-- vim: set ft=xml noet : -->
	<!-- base node launch file-->

	<arg name="fcu_url" default="udp://@192.168.50.242:16550?ids=255,240" />
	<arg name="gcs_url" default="" />
	<arg name="tgt_system" default="1" />
	<arg name="tgt_component" default="1" />
	<arg name="log_output" default="screen" />
	<arg name="fcu_protocol" default="v2.0" />
	<arg name="respawn_mavros" default="false" />

	<node pkg="mavros" exec="mavros_node" namespace="mavros" output="screen">

		<param name="fcu_url" value="$(var fcu_url)" />
		<param name="gcs_url" value="$(var gcs_url)" />
		<param name="target_system_id" value="$(var tgt_system)" />
		<param name="target_component_id" value="$(var tgt_component)" />
		<param name="fcu_protocol" value="$(var fcu_protocol)" />

		<!-- load blacklist, config -->
		<param from="/home/nvidia/ros2_ws/src/mavros/mavros/launch/apm_pluginlists.yaml" />
		<param from="/home/nvidia/ros2_ws/src/mavros/mavros/launch/apm_config.yaml" />
	</node>
</launch>
```
I performed the source installation described [here](https://github.com/mavlink/mavros/tree/ros2/mavros#source-installation) and I've adjusted all the config files to the new version.
I want to highlight that I've installed it on my laptop and It works if I connect to SITL, but the installation on my nvidia jetson xavier to connect to a real ardupilot fails as I shown.

### MAVROS version and platform

Mavros: 2.4.0
ROS: Foxy
Ubuntu: 20.04

### Autopilot type and version

[ X] ArduPilot

### Node logs

```
[INFO] [mavros_node-1]: process started with pid [45378]
[mavros_node-1] [INFO] [1683018579.387337504] [mavros.mavros_node]: Starting mavros_node container
[mavros_node-1] [INFO] [1683018579.387715328] [mavros.mavros_node]: FCU URL: udp://@192.168.50.242:16550?ids=255,240
[mavros_node-1] [INFO] [1683018579.387803936] [mavros.mavros_node]: GCS URL: 
[mavros_node-1] [INFO] [1683018579.387868096] [mavros.mavros_node]: UAS Prefix: /uas1
[mavros_node-1] [INFO] [1683018579.387927168] [mavros.mavros_node]: Starting mavros router node
[mavros_node-1] [INFO] [1683018579.411115712] [mavros.mavros_router]: Built-in SIMD instructions: ARM NEON
[mavros_node-1] [INFO] [1683018579.411284928] [mavros.mavros_router]: Built-in MAVLink package version: 2022.12.30
[mavros_node-1] [INFO] [1683018579.411366944] [mavros.mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[mavros_node-1] [INFO] [1683018579.411435008] [mavros.mavros_router]: MAVROS Router started
[mavros_node-1] [INFO] [1683018579.411821248] [mavros.mavros_router]: Requested to add endpoint: type: 0, url: udp://@192.168.50.242:16550?ids=255,240
[mavros_node-1] [INFO] [1683018579.412745472] [mavros.mavros_router]: Endpoint link[1000] created
[mavros_node-1] [INFO] [1683018579.420173152] [mavros.mavros_router]: link[1000] opened successfully
[mavros_node-1] [INFO] [1683018579.420746432] [mavros.mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[mavros_node-1] [INFO] [1683018579.421282592] [mavros.mavros_router]: Endpoint link[1001] created
[mavros_node-1] [INFO] [1683018579.428729440] [mavros.mavros_router]: link[1001] opened successfully
[mavros_node-1] [INFO] [1683018579.429038208] [mavros.mavros_node]: Starting mavros uas node
[mavros_node-1] [INFO] [1683018579.596906784] [mavros.mavros]: UAS Executor started, threads: 8
[mavros_node-1] [INFO] [1683018579.687384320] [mavros.mavros]: Plugin actuator_control created
[mavros_node-1] [INFO] [1683018579.688661024] [mavros.mavros]: Plugin actuator_control initialized
[mavros_node-1] [INFO] [1683018579.774774144] [mavros.mavros]: Plugin adsb created
[mavros_node-1] [INFO] [1683018579.775662144] [mavros.mavros]: Plugin adsb initialized
[mavros_node-1] [INFO] [1683018579.798570944] [mavros.mavros]: Plugin altitude created
[mavros_node-1] [INFO] [1683018579.800514656] [mavros.mavros]: Plugin altitude initialized
[mavros_node-1] [INFO] [1683018579.823962176] [mavros.mavros]: Plugin cam_imu_sync created
[mavros_node-1] [INFO] [1683018579.824787872] [mavros.mavros]: Plugin cam_imu_sync initialized
[mavros_node-1] /home/nvidia/ros2_ws/install/mavros/lib/mavros/mavros_node: symbol lookup error: /home/nvidia/ros2_ws/install/mavros_msgs/lib/libmavros_msgs__rosidl_typesupport_fastrtps_cpp.so: undefined symbol: _ZN15geographic_msgs3msg24typesupport_fastrtps_cpp28max_serialized_size_GeoPointERbm
[ERROR] [mavros_node-1]: process has died [pid 45378, exit code 127, cmd '/home/nvidia/ros2_ws/install/mavros/lib/mavros/mavros_node --ros-args -r __ns:=/mavros --params-file /tmp/launch_params_ox05zj09 --params-file /tmp/launch_params_6gzurebw --params-file /tmp/launch_params_1fwo4wfg --params-file /tmp/launch_params_rb13aw8a --params-file /tmp/launch_params_uno6dete --params-file /home/nvidia/ros2_ws/src/mavros/mavros/launch/apm_pluginlists.yaml --params-file /home/nvidia/ros2_ws/src/mavros/mavros/launch/apm_config.yaml'].

```

vooon, 2023-05-02:
Looks strange, but maybe worth to try to add geographic_msgs to the workspace src.

dottantgal, 2023-05-04:
Thanks for the suggestion. I added the geographic_msgs from [here](https://github.com/ros-geographic-info/geographic_info/tree/ros2), compiled without problems, sourced the ws, but nothing, same error occurred.

dottantgal, 2023-05-04:
If it could help these are the printouts of
`ldd /home/nvidia/ros2_ws/install/mavros_msgs/lib/libmavros_msgs__rosidl_typesupport_fastrtps_cpp.so`
which shows
```
linux-vdso.so.1 (0x0000ffff88bca000)
	librosidl_typesupport_fastrtps_cpp.so => /opt/ros/foxy/lib/librosidl_typesupport_fastrtps_cpp.so (0x0000ffff88ad1000)
	libbuiltin_interfaces__rosidl_typesupport_fastrtps_cpp.so => /opt/ros/foxy/lib/libbuiltin_interfaces__rosidl_typesupport_fastrtps_cpp.so (0x0000ffff88abe000)
	librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so => /opt/ros/foxy/lib/librcl_interfaces__rosidl_typesupport_fastrtps_cpp.so (0x0000ffff88a96000)
	libgeometry_msgs__rosidl_typesupport_fastrtps_cpp.so => /opt/ros/foxy/lib/libgeometry_msgs__rosidl_typesupport_fastrtps_cpp.so (0x0000ffff88a74000)
	libstd_msgs__rosidl_typesupport_fastrtps_cpp.so => /opt/ros/foxy/lib/libstd_msgs__rosidl_typesupport_fastrtps_cpp.so (0x0000ffff88a4f000)
	libfastcdr.so.1 => /opt/ros/foxy/lib/libfastcdr.so.1 (0x0000ffff88a2d000)
	librosidl_runtime_c.so => /opt/ros/foxy/lib/librosidl_runtime_c.so (0x0000ffff88a13000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff887ee000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff887ca000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff88657000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff88b9a000)
	librcutils.so => /opt/ros/foxy/lib/librcutils.so (0x0000ffff88632000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff88587000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffff88573000)
```
and 
`ldd /home/nvidia/ros2_ws/install/mavros/lib/mavros/mavros_node`
which shows
```
linux-vdso.so.1 (0x0000ffff814f0000)
	libmavros.so => /home/nvidia/ros2_ws/install/mavros/lib/libmavros.so (0x0000ffff80c4d000)
	libmavros_msgs__rosidl_typesupport_cpp.so => /home/nvidia/ros2_ws/install/mavros_msgs/lib/libmavros_msgs__rosidl_typesupport_cpp.so (0x0000ffff80c07000)
	libdiagnostic_msgs__rosidl_typesupport_cpp.so => /opt/ros/foxy/lib/libdiagnostic_msgs__rosidl_typesupport_cpp.so (0x0000ffff80bf3000)
	libmavconn.so => /home/nvidia/ros2_ws/install/libmavconn/lib/libmavconn.so (0x0000ffff80aa9000)
	librclcpp.so => /opt/ros/foxy/lib/librclcpp.so (0x0000ffff808e1000)
	librcl.so => /opt/ros/foxy/lib/librcl.so (0x0000ffff8089a000)
	libtracetools.so => /opt/ros/foxy/lib/libtracetools.so (0x0000ffff80888000)
	librcutils.so => /opt/ros/foxy/lib/librcutils.so (0x0000ffff80863000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff8063e000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff8061a000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff804a7000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff814c0000)
	libconsole_bridge.so.1.0 => /opt/ros/foxy/lib/aarch64-linux-gnu/libconsole_bridge.so.1.0 (0x0000ffff80492000)
	libtf2_msgs__rosidl_typesupport_cpp.so => /opt/ros/foxy/lib/libtf2_msgs__rosidl_typesupport_cpp.so (0x0000ffff8047c000)
	libtf2_ros.so => /opt/ros/foxy/lib/libtf2_ros.so (0x0000ffff803ec000)
	libGeographic.so.19 => /lib/aarch64-linux-gnu/libGeographic.so.19 (0x0000ffff8034e000)
	libtinyxml2.so.6 => /lib/aarch64-linux-gnu/libtinyxml2.so.6 (0x0000ffff8032a000)
	libament_index_cpp.so => /opt/ros/foxy/lib/libament_index_cpp.so (0x0000ffff80312000)
	libclass_loader.so => /opt/ros/foxy/lib/libclass_loader.so (0x0000ffff802ef000)
	liblibstatistics_collector.so => /opt/ros/foxy/lib/liblibstatistics_collector.so (0x0000ffff802d9000)
	libstatistics_msgs__rosidl_typesupport_cpp.so => /opt/ros/foxy/lib/libstatistics_msgs__rosidl_typesupport_cpp.so (0x0000ffff802c6000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000ffff80295000)
	libtf2.so => /opt/ros/foxy/lib/libtf2.so (0x0000ffff80264000)
	librcpputils.so => /opt/ros/foxy/lib/librcpputils.so (0x0000ffff8024d000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff801a2000)
	librosidl_typesupport_cpp.so => /opt/ros/foxy/lib/librosidl_typesupport_cpp.so (0x0000ffff8018d000)
	librcl_yaml_param_parser.so => /opt/ros/foxy/lib/librcl_yaml_param_parser.so (0x0000ffff80174000)
	librosgraph_msgs__rosidl_typesupport_cpp.so => /opt/ros/foxy/lib/librosgraph_msgs__rosidl_typesupport_cpp.so (0x0000ffff80162000)
	librcl_interfaces__rosidl_typesupport_cpp.so => /opt/ros/foxy/lib/librcl_interfaces__rosidl_typesupport_cpp.so (0x0000ffff80149000)
	librmw_implementation.so => /opt/ros/foxy/lib/librmw_implementation.so (0x0000ffff8012f000)
	librmw.so => /opt/ros/foxy/lib/librmw.so (0x0000ffff80118000)
	librcl_interfaces__rosidl_typesupport_c.so => /opt/ros/foxy/lib/librcl_interfaces__rosidl_typesupport_c.so (0x0000ffff80101000)
	librcl_logging_spdlog.so => /opt/ros/foxy/lib/librcl_logging_spdlog.so (0x0000ffff800e6000)
	librcl_interfaces__rosidl_generator_c.so => /opt/ros/foxy/lib/librcl_interfaces__rosidl_generator_c.so (0x0000ffff800ba000)
	librosidl_runtime_c.so => /opt/ros/foxy/lib/librosidl_runtime_c.so (0x0000ffff800a0000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000ffff8008c000)
	libaction_msgs__rosidl_typesupport_cpp.so => /opt/ros/foxy/lib/libaction_msgs__rosidl_typesupport_cpp.so (0x0000ffff80078000)
	librclcpp_action.so => /opt/ros/foxy/lib/librclcpp_action.so (0x0000ffff8004b000)
	libyaml.so => /opt/ros/foxy/lib/libyaml.so (0x0000ffff8001c000)
	librosidl_typesupport_c.so => /opt/ros/foxy/lib/librosidl_typesupport_c.so (0x0000ffff80008000)
	libspdlog.so.1 => /lib/aarch64-linux-gnu/libspdlog.so.1 (0x0000ffff7ff5c000)
	libbuiltin_interfaces__rosidl_generator_c.so => /opt/ros/foxy/lib/libbuiltin_interfaces__rosidl_generator_c.so (0x0000ffff7ff48000)
	librcl_action.so => /opt/ros/foxy/lib/librcl_action.so (0x0000ffff7ff26000)
```