mavlink/mavros #1861: Load Params for Ros2 launchfile?

chngdickson, 2023-05-08:
----

### Issue details

_How to Load param file using ros2? The commented code below works, but when loading params using the yaml file, it doesn't work._

### MAVROS version and platform

ROS: ?Humble?
Ubuntu: ?22.04?

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

### Ros2 launch.py code

```
import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription, RegisterEventHandler
from launch.conditions import IfCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch.launch_context import LaunchContext
from launch.events.process.process_exited import ProcessExited
from launch.event_handlers.on_process_exit import OnProcessExit
from launch_ros.actions import Node, node

def generate_launch_description():

  pth_my_world_ros = get_package_share_directory('tutorial_2')
  pth_mavros_configs = os.path.join(pth_my_world_ros,"config","mavros_configs")
  pth_param1 = os.path.join(pth_mavros_configs,"apm_pluginlists.yaml")
  pth_param2 = os.path.join(pth_mavros_configs,"apm_config2.yaml")
  
  config_param1 = LaunchConfiguration('config_dir',default=pth_param1)
  config_param2 = LaunchConfiguration('config_dir',default=pth_param2)
  mavros_node = Node(
    package='mavros',
    executable='mavros_node',
    namespace='mavros',
    output="screen",
    parameters=[config_param2,config_param1
      # {
      #   "fcu_url":"udp://:14551@",
      #   "gcs_url":"udp://:115200@", # QGC use [14550], MissionPlanner [115200]
      #   "target_system_id":1,
      #   "target_component_id":1,
      #   "fcu_protocol":"v2.0",
      # },
      # node.ParameterFile(param1, allow_substs=True)
      # node.ParameterFile(
      #  param1
      # )
      ]
  )
```

vooon, 2023-05-08:
Unusure about python-launch. Check https://github.com/mavlink/mavros/blob/ros2/mavros/launch/ xml was recently upgraded for ros2.

Also note that each plugin is a node, so check prefixes, e.g. in apm_config.yml

kribe48, 2023-05-25:
Not sure if it solves the issue, but uou should not name both of the launch configurations to "config_dir" - you should use a unique name for each of the launch configurations.

antonikaras, 2023-05-30:
Can you try something like this : 

* mavros_launcher.launch.py
```
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import Node

def launch_setup(context, *args, **kwargs):
        
    # Define input variables
    log_level = LaunchConfiguration('log_level')

    # Get the dev parameters
    dev_params = get_package_share_directory('mavros_launcher') + '/params/params.yaml'

    tmp = Node(package="mavros", 
                executable="mavros_node", 
                parameters=[dev_params], 
                arguments=['--ros-args', '--log-level', log_level],
                output="screen")

    return [tmp]

def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument('log_level', default_value="info"),
        OpaqueFunction(function = launch_setup)
        ])
```

* params.yaml
```
mavros_node:
  ros__parameters:
    fcu_url: "serial:///dev/ttyACM0:115200"
    gcs_url: "udp://@192.168.2.1:14550"
```

kribe48, 2023-05-30:
@antonikaras this looks like something that will work, but you are missing the apm specific configuration and pluginlists files. Perhaps this could work? 
```
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch.substitutions import LaunchConfiguration
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import Node

def launch_setup(context, *args, **kwargs):
        
    # Define input variables
    log_level = LaunchConfiguration('log_level')

    # Get all parameter files
    dev_params = get_package_share_directory('mavros_launcher') + '/params/params.yaml'
    config_params = get_package_share_directory('mavros_launcher') + /params/apm_config.yaml'
    pluginlist_params = get_package_share_directory('mavros_launcher') + /params/apm_pluginlists.yaml'

    tmp = Node(package="mavros", 
                executable="mavros_node", 
                parameters=[dev_params, config_params, pluginlist_params], 
                arguments=['--ros-args', '--log-level', log_level],
                output="screen")

    return [tmp]
def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument('log_level', default_value="info"),
        OpaqueFunction(function = launch_setup)
        ])
```

chngdickson, 2023-06-10:
> Can you try something like this :
>
> * mavros_launcher.launch.py
>
> ```
> from ament_index_python.packages import get_package_share_directory
> from launch import LaunchDescription
> from launch.actions [...]

Sorry for the 2 weeks reply, was extremely busy on site.

I've tried the implementation as suggested, but it sadly didn't work. Changed the params.yaml to something more suitable as fcu_url on my pc is 14550 and gcs_url is 115200.

chngdickson, 2023-06-10:
Both of your suggestion worked but requires some tweaking to work in a simulation environment.

Launch file
```
import os
from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch.launch_context import LaunchContext
from launch.events.process.process_exited import ProcessExited
from launch.event_handlers.on_process_exit import OnProcessExit
from launch.actions import DeclareLaunchArgument, OpaqueFunction
from launch_ros.actions import Node

def launch_setup(context, *args, **kwargs):

  pth_mavros_launcher = get_package_share_directory("mavros_launcher_package")
  pth_param0 = pth_mavros_launcher + "/config/mavros_configs/params.yaml"
  pth_param1 = pth_mavros_launcher + "/config/mavros_configs/apm_config.yaml"
  pth_param2 = pth_mavros_launcher + "/config/mavros_configs/apm_pluginlists.yaml"  
  log_level = LaunchConfiguration("log_level")

  mavros_node = Node(
    package="mavros",
    executable="mavros_node",
    output="screen",
    parameters=[pth_param0, pth_param1, pth_param2],
    arguments=["--ros-args", "--log-level", log_level]
    
  )
  
  return [mavros_node]
  
def generate_launch_description():
  def on_exit_restart(event:ProcessExited, context:LaunchContext):
    print("\n\nProcess [{}] exited, pid: {}, return code: {}\n\n".format(
      event.action.name, event.pid, event.returncode))
    if event.returncode != 0 and "mavros_node" in event.action.name:
      return generate_launch_description() # respawn node action
      
  return LaunchDescription([
    DeclareLaunchArgument("log_level", default_value="info"),
    OpaqueFunction(function = launch_setup)
    ])
```

params.yaml 
```# --- system plugins ---
mavros_node:
  ros__parameters:
    fcu_url: "udp://:14550@"
    gcs_url: "udp://:115200@"
    tgt_system: 1
    tgt_component: 1
    heartbeat_rate: 1.0
    publish_sim_time: false
```

apm_config.yaml and apm_pluginlists.yaml  is set from the default

closed 2023-06-10 by chngdickson