mavlink/mavros #1863: Executing commands on Pixhawk with ArduPilot via MAVROS2

OliverZanone, 2023-05-31:
### Issue details
Finding information for MAVROS2 has been pretty frustrating as relevant information seems scarce. 
As much as I have struggled to get some of this working, I would be more than happy to write up some tutorials after all of this. 
The best source of information I've found came from this post [here](https://github.com/mavlink/mavros/issues/1718).

From this, I have been attempting to manually deploy the commands to a Pixhawk FMU via a rpi companion computer using a combination of services and especially commandLong, i.e. for arming:
`ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: True}"`
For setting mode to guided:
`ros2 service call /mavros/cmd/command mavros_msgs/srv/CommandLong "{command: 176, param1: 1, param2: 4}"` 
I later realized you must also follow your firmware's documentation as well, and the supported parameters found [here](https://ardupilot.org/dev/docs/mavlink-get-set-flightmode.html) for ardupilot, thus the values for param1 and param2.
I can confirm the mode is being set in mission planner flight modes tab. 

Since I was able to get those working, I have tried working through other commands found in the mavlink [docs](https://mavlink.io/en/messages/common.html) and then crossreferencing those with ardupilot documentation. 
However, all other commands I've tested have failed. 

For example, taking off: 
`ros2 service call /mavros/cmd/command mavros_msgs/srv/CommandLong "{command: 22, param7: 5}"`
With broadcast set to false, nothing happens/no output. Setting broadcast to true shows "Unexpected command 22, result 4" I have all of the prearm safety checks unchecked, other than the hardware button found on my gps unit. 

Trying the built in mavros service for taking off:
`ros2 service call /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL "{altitude: 5}"`
this gives success=false result=4, which according to mavlink docs, 4 means good command but failed to execute, as in the prior method. 

My goal is to implement all of the guided commands found [here](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html) programatically. Some of these fall under the message category, not the command category on mavlink docs, which commandLong I assume is only for the command category. 

I assume some of these commands are built into the other mavros services, but as I cannot find any documentation on how to use them, it has really been a game of trial and error on using just the commandLong.

If someone could point me in the right direction on either finding documentation on some of the mavros services or figuring out how to implement some of these commands and messages found in mavlink/ardupilot documentation that would be greatly appreciated. 

### MAVROS version and platform

Mavros2: 2.4.0-1jammy.20230429.052850
ROS2: Humble
Ubuntu: 22.04.2 LTS (in docker container running on raspberry pi bullseye)

### Autopilot type and version

[ x ] ArduPilot
[  ] PX4

Version: V4.0.6

### Node logs

```
ros2 run mavros mavros_node --ros-args --param fcu_url:=/dev/serial0:921600
[INFO] [1685560039.392028775] [mavros_node]: Starting mavros_node container
[INFO] [1685560039.393001800] [mavros_node]: FCU URL: /dev/serial0:921600
[INFO] [1685560039.393076130] [mavros_node]: GCS URL:
[INFO] [1685560039.393122974] [mavros_node]: UAS Prefix: /uas1
[INFO] [1685560039.393175300] [mavros_node]: Starting mavros router node
[INFO] [1685560039.444330369] [mavros_router]: Built-in SIMD instructions: ARM NEON
[INFO] [1685560039.444455785] [mavros_router]: Built-in MAVLink package version: 2022.12.30
[INFO] [1685560039.444505481] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1685560039.444946280] [mavros_router]: MAVROS Router started
[INFO] [1685560039.445124022] [mavros_router]: Requested to add endpoint: type: 0, url: /dev/serial0:921600
[INFO] [1685560039.445441961] [mavros_router]: Endpoint link[1000] created
[INFO] [1685560039.447051659] [mavros_router]: link[1000] opened successfully
[INFO] [1685560039.447727656] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1685560039.447875133] [mavros_router]: Endpoint link[1001] created
[INFO] [1685560039.473473311] [mavros_router]: link[1001] opened successfully
[INFO] [1685560039.473701175] [mavros_node]: Starting mavros uas node
[INFO] [1685560040.019120526] [mavros]: UAS Executor started, threads: 4
[INFO] [1685560040.105053786] [mavros_router]: link[1000] detected remote address 1.1
[INFO] [1685560040.197345404] [mavros]: Plugin actuator_control created
[INFO] [1685560040.197727226] [mavros]: Plugin actuator_control initialized
[INFO] [1685560040.219377512] [mavros]: Plugin altitude created
[INFO] [1685560040.219758593] [mavros]: Plugin altitude initialized
[INFO] [1685560040.253212990] [mavros]: Plugin command created
[INFO] [1685560040.253993712] [mavros]: Plugin command initialized
[... 420 lines cut ...]
  - key: Tx total bytes
    value: '129655'
  - key: Rx speed
    value: inf
  - key: Tx speed
    value: inf
  - key: Remotes count
    value: '5'
  - key: Remote [0]
    value: '0.0'
  - key: Remote [1]
    value: '1.0'
  - key: Remote [2]
    value: '1.1'
  - key: Remote [3]
    value: '255.0'
  - key: Remote [4]
    value: '255.190'
```

### Check ID

```
I dont remember the equivalent check id command for ROS2 MAVROS
```

vooon, 2023-05-31:
Hi,

Check id - `ros2 run mavros mav checkid`. But i already see ids in router diag. A bit strange that all system stats are zero, and that there is not battery.

https://mavlink.io/en/messages/common.html#MAV_RESULT_FAILED why that happened most likely be sent as STATUSTEXT, which printed with `FCU: ` prefix.

OliverZanone, 2023-06-02:
Hello, I added my check ID to the first post. It seems to be outputting correctly. I actually was able to get the takeoff and land commands to work. However, there is still very limited output on the mavros_node console. Is there a way to make this more verbose, i.e. have it output when the mode changes and when it is performing a command?

My prior issues were as follows: I was indoors with no GPS signal. I was attempting to test the Guided_NoGPS mode, and have it take off in that mode. I am unsure of why this was not working, I am guessing there is a different protocol/command to use for this mode. After moving to an area with GPS and retesting in guided mode, takeoff and landing worked as expected. 
For others, the following is for setting mode, arming, taking off, then landing: 
```
ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: True}"
ros2 service call /mavros/cmd/command mavros_msgs/srv/CommandLong "{command: 176, param1: 1, param2: 4}"
# Can also use this:
# ros2 service call /mavros/set_mode mavros_msgs/srv/SetMode "{base_mode: 0, custom_mode: 4}"
ros2 service call /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL "{altitude: 5}"
ros2 service call /mavros/cmd/land mavros_msgs/srv/CommandTOL
```
What mavros command would be for [SET_POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED)?
I see a long list of mavros2 services (using ros2 list services -t), but with no documentation on them, have no idea which would be the correct one. 

Would these messages maybe be params that can then be set? Is there something similar to CommandLong but for messages?
Thanks

vooon, 2023-06-02:
Version 2 mostly have same messages and plugins as v1. So you can list topics and see that you're asking about setpoint* plugins.

Actually checkid doesn't looks correct, to few messages coming from 1.1. Should be much more than just a heartbeat.
Maybe you need to check port configuration and also try to request data streams.

wntun, 2024-08-06:
Hello @OliverZanone , I'm using the Cube Blue and cannot even change the flight mode. Could you share the parameters you used for serial?