mavlink/mavros #1864: How to install mavros inside the container with ros2 humble preinstalled?

reinzler, 2023-06-08:
Hi, how to install mavros inside the container with ros2 humble preinstalled? I tried to find an installation tutorial, but haven't managed.
----

### Issue details
Installation support?

### MAVROS version and platform

Mavros: latest?
ROS2 Humble (inside NVIDIA ISAAC ROS container)
Ubuntu: 20.04
NVIDIA Jetson Orin Nano

vooon, 2023-06-09:
I suppose just simple `docker exec -ti ros bash` and then use `apt install`?

reinzler, 2023-06-11:
> I suppose just simple `docker exec -ti ros bash` and then use `apt install`?

Thanks for response, but installation guidelines a for catkin, looks like for old version of ROS. Will it works with ROS2 Humble?

vooon, 2023-06-12:
https://github.com/mavlink/mavros/blob/ros2/mavros/README.md#installation

You sure?

NatanelBirarov, 2023-06-13:
> https://github.com/mavlink/mavros/blob/ros2/mavros/README.md#installation
> 
> You sure?

Thanks for response, but - E: Unable to locate package ros-humble-mavros

rockstarartist, 2023-09-07:
You are welcome to use my container if you like:
https://hub.docker.com/r/rockstarartist/mavros2

ZhongmouLi, 2023-11-26:
@rockstarartist Hi, I find your container so helpful and it frees from installing everything from the beginning.

But when I begin a simulation of Ardupilot using
`sim_vehicle.py -v ArduCopter --console` and run your container like 
`docker run -v /dev/shm:/dev/shm --net=host rockstarartist/mavros2:humble-x86 ros2 run mavros mavros_node --ros-args -p fcu_url:=udp://:14561@localhost:5901 -p tgt_system:=1`

It says `ta
[INFO] [1701006354.890800104] [mavros]: MAVROS UAS via /uas1 started. MY ID 1.191, TARGET ID 1.1
[INFO] [1701006355.847949284] [mavros_router]: link[1001] detected remote address 1.191
[WARN] [1701006356.855918982] [mavros.sys]: VER: broadcast request timeout, retries left 4
[WARN] [1701006357.854969064] [mavros.sys]: VER: broadcast request timeout, retries left 3
[WARN] [1701006358.854882772] [mavros.sys]: VER: unicast request timeout, retries left 2
[WARN] [1701006359.856221560] [mavros.sys]: VER: unicast request timeout, retries left 1
[WARN] [1701006360.855133240] [mavros.sys]: VER: unicast request timeout, retries left 0
[WARN] [1701006361.855991097] [mavros.sys]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities`

Do you know how to solve it?

rockstarartist, 2023-11-26:
Hi!

I'll take a look at it this evening!

-Felix

rockstarartist, 2023-11-26:
You may have your Arducopter on a different port, try changing the fcu url
from 14561 to 14550 or 14551 and see if the mavros container connects.

ZhongmouLi, 2023-12-16:
@rockstarartist Hi, sorry to reply so late.

But I tried `docker run -v /dev/shm:/dev/shm --net=host rockstarartist/mavros2:humble-x86 ros2 run mavros mavros_node --ros-args -p fcu_url:=udp://:14551@localhost:5901 -p tgt_system:=1` and also with 14550. Nothing works and it says

`[INFO] [1702742438.605417883] [mavros]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1702742438.605422979] [mavros]: MAVROS UAS via /uas1 started. MY ID 1.191, TARGET ID 1.1
[INFO] [1702742439.577656563] [mavros_router]: link[1001] detected remote address 1.191
[WARN] [1702742440.586103605] [mavros.sys]: VER: broadcast request timeout, retries left 4
[WARN] [1702742441.590521673] [mavros.sys]: VER: broadcast request timeout, retries left 3
[WARN] [1702742442.589055109] [mavros.sys]: VER: unicast request timeout, retries left 2
[WARN] [1702742443.584184938] [mavros.sys]: VER: unicast request timeout, retries left 1
[WARN] [1702742444.586135757] [mavros.sys]: VER: unicast request timeout, retries left 0
[WARN] [1702742445.582410087] [mavros.sys]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
`

By the way, when I run `sim_vehicle.py -v ArduCopter --console` on my pc, there is the what shows in terminal
`SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/robot/RobotFirmware/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/robot/RobotFirmware/ardupilot 
Setting out to                           : /home/robot/RobotFirmware/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
CXX Compiler                             : g++ 9.4.0 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Enabled OpenDroneID                      : no 
Enabled firmware ID checking             : no 
GPS Debug Logging                        : no 
Enabled custom controller                : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Configured VSCode Intellisense:                : no 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 3.8.10 
Checking for program 'python'                  : /usr/bin/python 
Checking for python version >= 2.7.0           : 3.8.10 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Debug build                                    : disabled 
Coverage build                                 : disabled 
SITL 32-bit build                              : disabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (0.582s)
{'model': '+', 'waf_target': 'bin/arducopter', 'default_params_filename': 'default_params/copter.parm', 'sitl-port': True}
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/robot/RobotFirmware/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/robot/RobotFirmware/ardupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
Waf: Leaving directory `/home/robot/RobotFirmware/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/robot/RobotFirmware/ardupilot/build/sitl
Target          Text (B)  Data (B)  BSS (B)  Total Flash Used (B)  Free Flash (B)
---------------------------------------------------------------------------------
bin/arducopter   3853743    180909   203976               4034652  Not Applicable

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (1.829s)
SIM_VEHICLE: Using defaults from (../Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/robot/RobotFirmware/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/robot/RobotFirmware/ardupilot/build/sitl/bin/arducopter" "-S" "--model" "+" "--speedup" "1" "--slave" "0" "--defaults" "../Tools/autotest/default_params/copter.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--console"
RiTW: Starting ArduCopter : /home/robot/RobotFirmware/ardupilot/build/sitl/bin/arducopter -S --model + --speedup 1 --slave 0 --defaults ../Tools/autotest/default_params/copter.parm -I0
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
`

rockstarartist, 2023-12-17:
have you tried the following:

docker run -v /dev/shm:/dev/shm --net=host rockstarartist/mavros2:humble-x86 ros2 run mavros mavros_node --ros-args -p fcu_url:=udp://127.0.0.1:14551 -p tgt_system:=1

ZhongmouLi, 2023-12-17:
H, @rockstarartist , thanks wo much for your reply.

But I still got error from your container
---
[INFO] [1702810594.739487662] [mavros]: MAVROS UAS via /uas1 started. MY ID 1.191, TARGET ID 1.1
[INFO] [1702810595.700723875] [mavros_router]: link[1001] detected remote address 1.191
[WARN] [1702810596.713612647] [mavros.sys]: VER: broadcast request timeout, retries left 4
[WARN] [1702810597.707605334] [mavros.sys]: VER: broadcast request timeout, retries left 3
[WARN] [1702810598.714392698] [mavros.sys]: VER: unicast request timeout, retries left 2
[WARN] [1702810599.715993658] [mavros.sys]: VER: unicast request timeout, retries left 1
[WARN] [1702810600.713304930] [mavros.sys]: VER: unicast request timeout, retries left 0
[WARN] [1702810601.713587351] [mavros.sys]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[INFO] [1702810624.453223612] [mavros_router]: link[1000] trying to reconnect...

---

here is output of sim_vehicle.py

```
SIM_VEHICLE: Using defaults from (../Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/robot/RobotFirmware/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/robot/RobotFirmware/ardupilot/build/sitl/bin/arducopter" "-S" "--model" "+" "--speedup" "1" "--slave" "0" "--defaults" "../Tools/autotest/default_params/copter.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--console"
RiTW: Starting ArduCopter : /home/robot/RobotFirmware/ardupilot/build/sitl/bin/arducopter -S --model + --speedup 1 --slave 0 --defaults ../Tools/autotest/default_params/copter.parm -I0
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Detected vehicle 1:1 on link 0
STABILIZE> Received 1310 parameters (ftp)
Saved 1310 parameters to mav.parm

```

rockstarartist, 2023-12-17:
So strange. I never really have issues connecting to the SITL. Let's try connecting it via tcp directly on the mavproxy master port.

So:
docker run -v /dev/shm:/dev/shm --net=host rockstarartist/mavros2:humble-x86 ros2 run mavros mavros_node --ros-args -p fcu_url:=tcp://localhost:5760 -p tgt_system:=1

Or

docker run -v /dev/shm:/dev/shm --net=host rockstarartist/mavros2:humble-x86 ros2 run mavros mavros_node --ros-args -p fcu_url:=tcp://127.0.0.1:5760 -p tgt_system:=1

ZhongmouLi, 2023-12-17:
@rockstarartist , I tried both and got the same error all the time.

I used ArduCopter 4.4.0. Is it possible that this version does not work well with mavros2?

rockstarartist, 2023-12-17:
Ill download Arducopter 4.4.0 on to my dev machine this evening and give it
a try and see where I get.

closed 2026-10-18 by github-actions[bot]