mavlink/mavros #1866: RC-Override in ROS2 Foxy - Equivalent of `rosrun mavros mavparam set SYSID_MYGCS 1`?

Scoeerg, 2023-06-15:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am publishing [RCOVERRIDE](https://mavlink.io/en/messages/common.html#RC_CHANNELS_OVERRIDE) in topic 
`/mavros/rc/override` using some minimal ROS2 Foxy Python Publisher. While the RC works fine, Rcoverride cannot engage the motors. 

What I suspect is, that it's about `SYSID_MYGCS` which is `1` for the Pixhawk, but supposedly must also be `1` for the companion computer. In ROS1 the command would be something along the line of `rosrun mavros mavparam set SYSID_MYGCS 1`. What's the equivalent in ROS2?

### MAVROS version and platform

Mavros: 2.0.0 ROS2 Alpha, installed with `sudo apt install ros-foxy-mavros`
ROS2: Foxy
Ubuntu: 20.04. LTS

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 4.2.3. Ardurover

### Details

```
ros2 run mavros mavros_node --ros-args --params-file ./mavros_param.yaml
```

with `mavros_param.yaml`:

```
# mavros_param.yaml
mavros:
  ros__parameters: {}

mavros_router:
  ros__parameters: {}

mavros_node:
  ros__parameters:
    # Parameters, see: http://wiki.ros.org/mavros
    fcu_url: /dev/ttyUSB0:57600
    system_id: 255
    component_id: 0
    target_system_id: 1
    target_component_id: 1
``` 

```
[INFO] [1686747788.943812714] [mavros_node]: Starting mavros_node container
[INFO] [1686747788.943862778] [mavros_node]: FCU URL: /dev/ttyUSB0:57600
[INFO] [1686747788.943870539] [mavros_node]: GCS URL: 
[INFO] [1686747788.943875523] [mavros_node]: UAS Prefix: /uas1
[INFO] [1686747788.943879979] [mavros_node]: Starting mavros router node
[INFO] [1686747788.945723999] [mavros_router]: Built-in SIMD instructions: SSE, SSE2
[INFO] [1686747788.945737300] [mavros_router]: Built-in MAVLink package version: 2022.12.30
[INFO] [1686747788.945742762] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1686747788.945748080] [mavros_router]: MAVROS Router started
[INFO] [1686747788.945765730] [mavros_router]: Requested to add endpoint: type: 0, url: /dev/ttyUSB0:57600
[INFO] [1686747788.945791745] [mavros_router]: Endpoint link[1000] created
[INFO] [1686747788.946507575] [mavros_router]: link[1000] opened successfully
[INFO] [1686747788.946532163] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1686747788.946547918] [mavros_router]: Endpoint link[1001] created
[INFO] [1686747788.946842226] [mavros_router]: link[1001] opened successfully
[INFO] [1686747788.946861934] [mavros_node]: Starting mavros uas node
[INFO] [1686747788.970065458] [mavros]: UAS Executor started, threads: 16
[INFO] [1686747788.978579203] [mavros]: Plugin actuator_control created
[... 261 lines cut ...]
        self.timer = self.create_timer(timer_period, self.timer_callback)

    def timer_callback(self):
        msg = OverrideRCIn()
        msg.channels = [1800, 0, 1800, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        self.publisher_.publish(msg)
        self.get_logger().info('Type: %s' %(type(msg.channels)))
        self.get_logger().info('Published RC-Override to Pixhawk using MavRos: %s' %(" ".join([str(x) for x in msg.channels])))

def main(args=None):
    rclpy.init(args=args)
    publisher = RCOverride()
    rclpy.spin(publisher)
    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Which might be an issue in itself. Since Python is not a typed language, using MissionPlanner to observe the Ardupilot tells me my
`msg.channels = [1800, 0, 1800, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]` has type array uint16 but the RC-Message has type list(?). Probably nothing, if interested I can attach some screenshots.

vooon, 2023-06-15:
`ros run mavros mav param --help` or `ros param list /mavros/param --help` (a bit unsure about `ros` command).

Scoeerg, 2023-06-15:
Thanks @vooon 

`ros2 run param list /mavros/param` is a dump from all parameters one can set, i.e. a VERY long list: 

```
  ACRO_TURN_RATE
  .
  .
  .
  WRC_ENABLE
```

one of which is actually `SYSID_MYGCS`. Then running [ROS2 Tutorial on Parameters](https://docs.ros.org/en/foxy/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Parameters/Understanding-ROS2-Parameters.html)

```
ros2 param set /mavros/param SYSID_MYGCS 1
```

worked like a charm. Will close the issue.