mavlink/mavros #1867: MAVROS2: `/mavros/mission/push` - Possible Bug: no response from Service

Scoeerg, 2023-06-19:
### Issue details

I would like to send Waypoints in MAVROS2, here's a minimal working example based on [this GIT Mavros Issue](https://github.com/mavlink/mavros/issues/1718):

```
# see https://github.com/mavlink/mavros/issues/1718
import rclpy
from rclpy.node import Node
from mavros_msgs.srv import WaypointPush
from mavros_msgs.msg import Waypoint
from mavros_msgs.msg import WaypointList

class WaypointPushNode(Node):
    def __init__(self):
        super().__init__('waypoint_push_node')
        waypoint_push_client = self.create_client(WaypointPush, '/mavros/mission/push')

        # See /mavros/mavros_msgs/msg/Waypoint.msg and /mavros/mavros_msgs/msg/WaypointList.msg
        waypoint_list = WaypointList()
        waypoint = Waypoint()
        # See https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_WAYPOINT
        waypoint.frame = 0 # frame 6 = Global relative altitude (to home position)
        waypoint.command = 16 # command 16 = MAV_CMD_NAV_WAYPOINT -> go to waypoint
        waypoint.is_current = False
        waypoint.autocontinue = True
        # Hold time (how long does the vehicle stay at the waypoint in seconds):
        waypoint.param1 = float(0)
        # Accept Radius - how close you need to be to mark waypoint as reached in meters:
        waypoint.param2 = float(0)
        # 0 to pass through the WP, if > 0 radius to pass by WP. Positive value for clockwise orbit, negative value for
        # counter-clockwise orbit. Allows trajectory control:
        waypoint.param3 = float(0)
        # Desired yaw angle at waypoint (rotary wing). NaN to use the current system yaw heading mode
        # (e.g. yaw towards next waypoint, yaw to home, etc.):
        waypoint.param4 = float("nan")
        # Latitude:
        waypoint.x_lat = float(49.4255568)
        # Longitude:
        waypoint.y_long = float(7.7538867)
        # Altitude in meters:
        waypoint.z_alt = float(0)
        # Append
        #waypoint_list.waypoints.append(waypoint)
        # Populate your request
        request = WaypointPush.Request()
        # See /mavros/mavros_msgs/srv/WaypointPush.srv
        request.start_index = 0
        request.waypoints.append(waypoint)
        self.get_logger().info('Waypoint sending: %s.' %(str(request)))
        # Make service call and store service response to check for success and number of waypoints transferred
        response = waypoint_push_client.call(request)
        self.get_logger().info(str(response))
        self.get_logger().info('Waypoint sent: %s .' %('successfully' if response else 'not successfully'))
def main(args=None):
    rclpy.init(args=args)
    waypoint_push = WaypointPushNode()
    rclpy.spin(waypoint_push)
    waypoint_push.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

which in terminal outputs:

```
[INFO] [1687170909.321930525] [waypoint_push_node]: Waypoint sent: mavros_msgs.srv.WaypointPush_Request(start_index=0, waypoints=[mavros_msgs.msg.Waypoint(frame=0, command=16, is_current=False, autocontinue=True, param1=0.0, param2=0.0, param3=0.0, param4=nan, x_lat=50.000000, y_long=10.0000, z_alt=0.0)]).
```

[... 629 lines cut ...]
/mavros/time/list_parameters
/mavros/time/set_parameters
/mavros/time/set_parameters_atomically
/mavros/vehicle_info_get
/mavros/wind/describe_parameters
/mavros/wind/get_parameter_types
/mavros/wind/get_parameters
/mavros/wind/list_parameters
/mavros/wind/set_parameters
/mavros/wind/set_parameters_atomically
/mavros_node/describe_parameters
/mavros_node/get_parameter_types
/mavros_node/get_parameters
/mavros_node/list_parameters
/mavros_node/set_parameters
/mavros_node/set_parameters_atomically
/mavros_router/add_endpoint
/mavros_router/del_endpoint
/mavros_router/describe_parameters
/mavros_router/get_parameter_types
/mavros_router/get_parameters
/mavros_router/list_parameters
/mavros_router/set_parameters
/mavros_router/set_parameters_atomically
```

vooon, 2023-06-19:
Sounds familiar to the issue somewhere inside rcl of Foxy that some messages not being received (rpc is affected).

Could you please try to build latest version from source? 2.0.0 quite old.
Also i suggest to upgrade ros distro, Foxy is EOLed.

Have you tried `mav wp pull` or `mav wp push`?
https://github.com/mavlink/mavros/blob/ros2/mavros/mavros/cmd/mission.py#L279

Note it uses python helper client library.

Scoeerg, 2023-06-20:
Thanks @vooon Unfortunately ROS2 Foxy needs to stay, since some sensor manufacturers have not yet published stable ros2 nodes for their hardware, even though it's EOL.

1) I now built from source (current ros2 branch, version 2.5.0)
2) same python-code, see above. Pushing [WaypointList](http://docs.ros.org/en/hydro/api/mavros/html/msg/WaypointList.html) to topic `/mavros/mission/push`

To check if this was successful, looking into MissionPlanner and try to read from Pixhawk.

a) MissionPlanner throws error `Timeout on Read-GetWaypointCount`
b) The ` response = waypoint_push_client.call(request)` still is not received, see above
c) I can finally read the waypoints, by` ros2 topic echo /mavros/mission/waypoints` which responds with:

```
current_seq: 0
waypoints:
- frame: 0
  command: 16
  is_current: false
  autocontinue: true
  param1: 0.0
  param2: 0.0
  param3: 0.0
  param4: .nan
  x_lat: 50.0000
  y_long: 7.0000
  z_alt: 0.0
---
current_seq: 0
waypoints:
- frame: 0
  command: 16
  is_current: false
  autocontinue: true
  param1: 0.0
  param2: 0.0
  param3: 0.0
  param4: 0.0
  x_lat: XXX
  y_long: XXX
  z_alt: XXX
---
```

where `XXX` is roughly the current GPS position (home position?)

Scoeerg, 2023-06-20:
I updated the python-code. It is now a little cleaner and I added `request.start_index = 0`. Here:

```
# see https://github.com/mavlink/mavros/issues/1718
# and https://github.com/mavlink/mavros/issues/1867
import rclpy
from rclpy.node import Node
from mavros_msgs.srv import WaypointPush # http://docs.ros.org/en/noetic/api/mavros_msgs/html/srv/WaypointPush.html
from mavros_msgs.msg import Waypoint # http://docs.ros.org/en/hydro/api/mavros/html/msg/Waypoint.html

class WaypointPushNode(Node):
    def __init__(self):
        super().__init__('waypoint_push_node')
        waypoint_push_client = self.create_client(WaypointPush, '/mavros/mission/push')
        # Waypoint 1:
        waypoint_1 = Waypoint()
        waypoint_1.frame = 0
        waypoint_1.command = 16
        waypoint_1.is_current = False
        waypoint_1.autocontinue = True
        waypoint_1.param1 = float(0)
        waypoint_1.param2 = float(0)
        waypoint_1.param3 = float(0)
        waypoint_1.param4 = float("nan")
        waypoint_1.x_lat = float(50.0000)
        waypoint_1.y_long = float(7.0000)
        waypoint_1.z_alt = float(0)
        # Waypoint 2:
        waypoint_2 = Waypoint()
        waypoint_2.frame = 0
        waypoint_2.command = 16
        waypoint_2.is_current = False
        waypoint_2.autocontinue = True
        waypoint_2.param1 = float(0)
        waypoint_2.param2 = float(0)
        waypoint_2.param3 = float(0)
        waypoint_2.param4 = float("nan")
        waypoint_2.x_lat = float(51.0000)
        waypoint_2.y_long = float(8.0000)
        waypoint_2.z_alt = float(0.0)
        # Populate your request
        request = WaypointPush.Request()
        request.start_index = 0
        request.waypoints.append(waypoint_1)
        request.waypoints.append(waypoint_2)
        self.get_logger().info('Waypoint sending: %s.' %(str(request)))
        response = waypoint_push_client.call(request)
        self.get_logger().info('Waypoint sent: %s.' %('successfully' if response.success else 'not successfully'))
def main(args=None):
    rclpy.init(args=args)
    waypoint_push = WaypointPushNode()
    rclpy.spin(waypoint_push)
    waypoint_push.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

to which the `mavros_node` in terminal responds with

```
[INFO] [1687257439.368765437] [mavros.mission]: WP: item #0  F:0 C: 16 p: 0 0 0 0 x: XXX y: XXX z: XXX
[INFO] [1687257439.684444558] [mavros.mission]: WP: item #1  F:0 C: 16 p: 0 0 0 0 x: 51.0000 y: 8.0000 z: 0
```

where `XXX` is home position. Interestingly, it seems to take the **second** waypoint but not the first. Changing the `request.start_index` to 1 (not overwriting first mission waypoint home position?!) throws inside `mavros_node` a warning `[WARN] [1687257279.555358071] [mavros.mission]: WP: Partial push out of range rejected.`. 

Other errors persist, see above.

Scoeerg, 2023-06-22:
Hi. Hold on tight, I will close this issue with this comment. @vooon I will contact you via Mail, I would like to contribute with tutorials for mavros2. 

A couple things concerning waypoints:

1) waypoints are very badly named, but that is Ardupilot/Mavlink convention, not Mavros. Waypoints CAN be waypoints in the sense that they mean (GPS) points in some reference frame. If so a waypoint will be e.g. 

```
waypoint_2 = Waypoint()
        waypoint_2.frame = 0
        waypoint_2.command = 16
        waypoint_2.is_current = False
        waypoint_2.autocontinue = True
        waypoint_2.param1 = float(0)
        waypoint_2.param2 = float(0)
        waypoint_2.param3 = float(0)
        waypoint_2.param4 = float("nan")
        waypoint_2.x_lat = float(50.0000)
        waypoint_2.y_long = float(7.0000)
        waypoint_2.z_alt = float(0.0)
```

`waypoint_2.command = 16` says _this waypoint concerns actual positions. see `waypoint_2.x_lat = float(50.0000)` ..._ BUT another waypoint:

```
# waypoint 1:
        waypoint_1 = Waypoint()
        waypoint_1.frame = 0
        waypoint_1.command = 178
        waypoint_1.is_current = False
        waypoint_1.autocontinue = True
        waypoint_1.param1 = float(0)
        waypoint_1.param2 = float(0.35) #meters per second
        waypoint_1.param3 = float(0.0) #throttle percentage 0.0 to 1.0 THIS XOR PARAM3
        waypoint_1.param4 = float(0.0)
        waypoint_1.x_lat = float(0.0)
        waypoint_1.y_long = float(0.0)
        waypoint_1.z_alt = float(0.0)
```

where `waypoint_1.command = 178` says _this waypoint concerns velocity rather than position. see `waypoint_1.param2 = float(0.35)` ..._. 
So waypoints would better be described as `missionpoints`, because they will define the way the robot/drone ... behaves during a mission. To understand this, I used a mixture of MissionPlanner and Ardupilot documentation. 

2) as @vooon said: 
> Ardupilot uses first waypoint to store home position (position captured on arming). That's why `mav wp` have some quirks for that.

To "solve" this issue I simply hacked it together in a way that I will send a dummy waypoint `waypoint_0`as the first waypoint to be ignored. And only from the second waypoint `waypoint_1` on I actually send the mission I want. 

3) As usual, the problem was not the code but the idiot (me) trying to use it. This time, I copied some code from [this mavros issue](https://github.com/mavlink/mavros/issues/1718) (see the opening of this issue), and it was not working. Reading the [ros2 tutorials on Services in python again](https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html), combined with the remarks 1) and 2) then comes:

```
# see https://github.com/mavlink/mavros/issues/1718
# and https://github.com/mavlink/mavros/issues/1867
import rclpy
from rclpy.node import Node
from mavros_msgs.srv import WaypointPush # http://docs.ros.org/en/noetic/api/mavros_msgs/html/srv/WaypointPush.html
from mavros_msgs.msg import Waypoint # http://docs.ros.org/en/hydro/api/mavros/html/msg/Waypoint.html

class WaypointPushNode(Node):

    def __init__(self):
        super().__init__('waypointpush_node')
        self.client = self.create_client(WaypointPush, '/mavros/mission/push')
        while not self.client.wait_for_service(timeout_sec=1.0):
            self.get_logger().info('Waypoint Push service is not available. Looking for /mavros/mission/push')
        # waypoints
        # waypoint 0: to be ignored
        waypoint_0 = Waypoint()
        waypoint_0.frame = 0
        waypoint_0.command = 16
        waypoint_0.is_current = False
[... 67 lines cut ...]
    def send_request(self):
        self.get_logger().info('Waypoint sent %s.' %(str(self.waypointlist)))
        self.future = self.client.call_async(self.request)
        rclpy.spin_until_future_complete(self, self.future)
        return self.future.result()

def main(args=None):
    rclpy.init(args=args)

    waypointpush_client = WaypointPushNode()
    response = waypointpush_client.send_request()
    waypointpush_client.get_logger().info('Waypoint sent %s.' %('successfully' if response.success else 'not successfully'))
    waypointpush_client.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

which then solves

**tl;dr:**

**response-error is fixed** - wrong use of ros2 Service
**ignore-first-waypoint-error is fixed** - explained above, Ardupilot does not overwrite first waypoint but says it's home position (first position recorded when arming)

closed 2023-06-22 by Scoeerg