mavlink/mavros #1868: Mavlink output from Cleanflight FC not published

csmotion, 2023-06-27:
I'd like to use MAVROS to publish data coming from the Mavlink output on a Cleanflight fc.  I understand that this isn't a supported controller/typical use case, but was curious if it's possible.  I can see data flowing from the fc as expected using pyMavlink (example below).  I can spin up a mavros_node connected to the serial port, and see that the fc appears to be receiving messages, but nothing is being published.

Thoughts on configuration changes required to make this work?  Interestingly, the output of "ros2 run mavros mav checkid" exploded (see below)...

### sample pyMavlink output

SYS_STATUS {onboard_control_sensors_present : 56359, onboard_control_sensors_enabled : 56359, onboard_control_sensors_health : 39, load : 0, voltage_battery : 20250, current_battery : 389, battery_remaining : 75, drop_rate_comm : 0, errors_comm : 0, errors_count1 : 0, errors_count2 : 0, errors_count3 : 0, errors_count4 : 0}
GPS_RAW_INT {time_usec : 3238052464, fix_type : 1, lat : 0, lon : 0, alt : -17000, eph : 65535, epv : 65535, vel : 0, cog : 0, satellites_visible : 0}
GLOBAL_POSITION_INT {time_boot_ms : 3238052481, lat : 0, lon : 0, alt : -17000, relative_alt : -17000, vx : 0, vy : 0, vz : 0, hdg : 34}
GPS_GLOBAL_ORIGIN {latitude : 0, longitude : 0, altitude : 0}
RC_CHANNELS_RAW {time_boot_ms : 3238132, port : 0, chan1_raw : 1500, chan2_raw : 1499, chan3_raw : 1505, chan4_raw : 999, chan5_raw : 1159, chan6_raw : 1500, chan7_raw : 1500, chan8_raw : 1500, rssi : 0}
ATTITUDE {time_boot_ms : 3238132, roll : 0.02094395086169243, pitch : 0.09948376566171646, yaw : 0.6021385788917542, rollspeed : 0.0, pitchspeed : 0.0, yawspeed : 0.0}
VFR_HUD {airspeed : 0.0, groundspeed : 0.0, heading : 34, throttle : 0, alt : -17.0, climb : 0.0}
HEARTBEAT {type : 2, autopilot : 0, base_mode : 64, custom_mode : 1, system_status : 3, mavlink_version : 3}
ATTITUDE {time_boot_ms : 3238252, roll : 0.02094395086169243, pitch : 0.09948376566171646, yaw : 0.6021385788917542, rollspeed : 0.0, pitchspeed : 0.0, yawspeed : 0.0}
VFR_HUD {airspeed : 0.0, groundspeed : 0.0, heading : 34, throttle : 0, alt : -17.0, climb : 0.0}
HEARTBEAT {type : 2, autopilot : 0, base_mode : 64, custom_mode : 1, system_status : 3, mavlink_version : 3}
RC_CHANNELS_RAW {time_boot_ms : 3238352, port : 0, chan1_raw : 1500, chan2_raw : 1499, chan3_raw : 1505, chan4_raw : 999, chan5_raw : 1159, chan6_raw : 1500, chan7_raw : 1500, chan8_raw : 1500, rssi : 0}
ATTITUDE {time_boot_ms : 3238372, roll : 0.02094395086169243, pitch : 0.09948376566171646, yaw : 0.6021385788917542, rollspeed : 0.0, pitchspeed : 0.0, yawspeed : 0.0}
VFR_HUD {airspeed : 0.0, groundspeed : 0.0, heading : 34, throttle : 0, alt : -17.0, climb : 0.0}
HEARTBEAT {type : 2, autopilot : 0, base_mode : 64, custom_mode : 1, system_status : 3, mavlink_version : 3}
ATTITUDE {time_boot_ms : 3238493, roll : 0.02094395086169243, pitch : 0.09948376566171646, yaw : 0.6021385788917542, rollspeed : 0.0, pitchspeed : 0.0, yawspeed : 0.0}
VFR_HUD {airspeed : 0.0, groundspeed : 0.0, heading : 34, throttle : 0, alt : -17.0, climb : 0.0}
HEARTBEAT {type : 2, autopilot : 0, base_mode : 64, custom_mode : 1, system_status : 3, mavlink_version : 3}
SYS_STATUS {onboard_control_sensors_present : 56359, onboard_control_sensors_enabled : 56359, onboard_control_sensors_health : 39, load : 0, voltage_battery : 20250, current_battery : 372, battery_remaining : 75, drop_rate_comm : 0, errors_comm : 0, errors_count1 : 0, errors_count2 : 0, errors_count3 : 0, errors_count4 : 0}
RC_CHANNELS_RAW {time_boot_ms : 3238573, port : 0, chan1_raw : 1500, chan2_raw : 1499, chan3_raw : 1505, chan4_raw : 999, chan5_raw : 1159, chan6_raw : 1500, chan7_raw : 1500, chan8_raw : 1500, rssi : 0}
GPS_RAW_INT {time_usec : 3238573179, fix_type : 1, lat : 0, lon : 0, alt : -17000, eph : 65535, epv : 65535, vel : 0, cog : 0, satellites_visible : 0}
GLOBAL_POSITION_INT {time_boot_ms : 3238573194, lat : 0, lon : 0, alt : -17000, relative_alt : -17000, vx : 0, vy : 0, vz : 0, hdg : 34}
GPS_GLOBAL_ORIGIN {latitude : 0, longitude : 0, altitude : 0}
ATTITUDE {time_boot_ms : 3238613, roll : 0.02094395086169243, pitch : 0.09948376566171646, yaw : 0.6021385788917542, rollspeed : 0.0, pitchspeed : 0.0, yawspeed : 0.0}
VFR_HUD {airspeed : 0.0, groundspeed : 0.0, heading : 34, throttle : 0, alt : -17.0, climb : 0.0}
HEARTBEAT {type : 2, autopilot : 0, base_mode : 64, custom_mode : 1, system_status : 3, mavlink_version : 3}

### MAVROS version and platform

Mavros: 2.5.0
ROS: Iron
Ubuntu: 22.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4
[x] Cleanflight

### Node logs
csmotion@Ubuntu22:~/Projects/QuadROS$ ros2 launch mavros apm.launch fcu_url:=/dev/ttyS0:115200
[INFO] [launch]: All log files can be found below /home/csmotion/.ros/log/2023-06-27-11-40-00-808087-Ubuntu22-15710
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mavros_node-1]: process started with pid [15713]
[mavros_node-1] [INFO] [1687887601.570050177] [mavros.mavros_node]: Starting mavros_node container
[mavros_node-1] [INFO] [1687887601.570236361] [mavros.mavros_node]: FCU URL: /dev/ttyS0:115200
[mavros_node-1] [INFO] [1687887601.570257512] [mavros.mavros_node]: GCS URL: 
[mavros_node-1] [INFO] [1687887601.570273804] [mavros.mavros_node]: UAS Prefix: /uas1
[mavros_node-1] [INFO] [1687887601.570288885] [mavros.mavros_node]: Starting mavros router node
[mavros_node-1] [INFO] [1687887601.591960837] [mavros.mavros_router]: Built-in SIMD instructions: SSE, SSE2
[mavros_node-1] [INFO] [1687887601.592018308] [mavros.mavros_router]: Built-in MAVLink package version: 2023.6.6
[mavros_node-1] [INFO] [1687887601.592030711] [mavros.mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[mavros_node-1] [INFO] [1687887601.592041406] [mavros.mavros_router]: MAVROS Router started
[mavros_node-1] [INFO] [1687887601.592081433] [mavros.mavros_router]: Requested to add endpoint: type: 0, url: /dev/ttyS0:115200
[mavros_node-1] [INFO] [1687887601.592110482] [mavros.mavros_router]: Endpoint link[1000] created
[mavros_node-1] [INFO] [1687887601.599914641] [mavros.mavros_router]: link[1000] opened successfully
[mavros_node-1] [INFO] [1687887601.600039110] [mavros.mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[mavros_node-1] [INFO] [1687887601.600079221] [mavros.mavros_router]: Endpoint link[1001] created
[mavros_node-1] [INFO] [1687887601.603499606] [mavros.mavros_router]: link[1000] detected remote address 0.200
[mavros_node-1] [INFO] [1687887601.606460475] [mavros.mavros_router]: link[1001] opened successfully
[mavros_node-1] [INFO] [1687887601.606530799] [mavros.mavros_node]: Starting mavros uas node
[mavros_node-1] [INFO] [1687887601.662544807] [mavros.mavros]: Plugin actuator_control ignored
[mavros_node-1] [INFO] [1687887601.662587889] [mavros.mavros]: Plugin altitude ignored
[mavros_node-1] [INFO] [1687887601.664674435] [mavros.mavros]: UAS Executor started, threads: 4
[... 372 lines cut ...]
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 84, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/opt/ros/iron/lib/python3.10/site-packages/mavros/cmd/checkid.py", line 129, in checkid
    checker = Checker(client=client, follow=follow, watch_time=watch_time)
  File "/opt/ros/iron/lib/python3.10/site-packages/mavros/cmd/checkid.py", line 44, in __init__
    self.tgt_ids = client.uas_settings.target_ids
  File "/opt/ros/iron/lib/python3.10/site-packages/mavros/cmd/__init__.py", line 36, in __getattr__
    return getattr(self.cli, key)
  File "/usr/lib/python3.10/functools.py", line 981, in __get__
    val = self.func(instance)
  File "/opt/ros/iron/lib/python3.10/site-packages/mavros/base.py", line 116, in uas_settings
    pd = call_get_parameters(node=self, node_name=self.mavros_ns, names=names)
  File "/opt/ros/iron/lib/python3.10/site-packages/mavros/utils.py", line 75, in call_get_parameters
    wait_for_service(client, lg)
  File "/opt/ros/iron/lib/python3.10/site-packages/mavros/base.py", line 55, in wait_for_service
    raise ServiceWaitTimeout(topic)
mavros.base.ServiceWaitTimeout: mavros/get_parameters
[ros2run]: Process exited with failure 1

vooon, 2023-06-27:
Hi,

Did you run checkid while mavros node running?

Remote ids 0.200 and 51.x looks weird. Is it possible to configure system id on cleanflight?

csmotion, 2023-06-27:
I tried to run checkid, but it fails: 

> csmotion@Ubuntu22:~/Projects/QuadROS$ ros2 run mavros mav checkid
> [ERROR] [1687889272.864801709] [mavpy_szy8]: wait for service time out: {topic}
> Traceback (most recent call last):
> File [...]

There doesn't appear to be a way to configure system id on cleanflight, however, pyMavlink seems to indicate a sysid=0 in this pic:

[image]

vooon, 2023-06-28:
The problem that 0 means a broadcast. Mavlink router might treat it incorrectly.
But maybe worth to try set target_system=0.

csmotion, 2023-06-28:
Setting target_system=0 works, and topics are broadcasting now.  Thank you very much!

output from ros2 topic echo diagnostics:

> header:
>   stamp:
>     sec: 1687969875
>     nanosec: 671766735
>   frame_id: ''
> status:
> - level: "\x02"
>   name: 'mavros: MAVROS UAS'
>   message: disconnected
>   hardware_id: uas:///uas0
>   values: []
> - level: "\x02"
>   name: 'mavros: GPS'
>   message: No satellites
>   hardware_id: uas:///uas0
>   values:
>   - key: Satellites visible
>     value: '0'
>   - key: Fix type
>     value: '1'
>   - key: EPH (m)
>     value: Unknown
>   - key: EPV (m)
>     value: Unknown
> - level: "\x02"
>   name: 'mavros: System'
>   message: Sensor health
>   hardware_id: uas:///uas0
>   values:
>   - key: Sensor present
>     value: '0x0000DC27'
>   - key: Sensor enabled
>     value: '0x0000DC27'
>   - key: Sensor health
>     value: '0x00000027'
>   - key: 3D gyro
>     value: Ok
>   - key: 3D accelerometer
>     value: Ok
>   - key: 3D magnetometer
>     value: Ok
>   - key: GPS
>     value: Ok
>   - key: 3D angular rate control
>     value: Fail
>   - key: attitude stabilization
>     value: Fail
>   - key: yaw position
>     value: Fail
>   - key: x/y position control
>     value: Fail
>   - key: motor outputs / control
>     value: Fail
>   - key: CPU Load (%)
>     value: '0.0'
>   - key: Drop rate (%)
>     value: '0.0'
>   - key: Errors comm
>     value: '0'
>   - key: 'Errors count #1'
>     value: '0'
>   - key: 'Errors count #2'
>     value: '0'
>   - key: 'Errors count #3'
>     value: '0'
>   - key: 'Errors count #4'
[... 329 lines cut ...]
>   values:
>   - key: Voltage
>     value: '20.60'
>   - key: Current
>     value: '3.7'
>   - key: Remaining
>     value: '82.0'
> - level: "\x02"
>   name: 'mavros: Heartbeat'
>   message: No events recorded.
>   hardware_id: uas:///uas0
>   values:
>   - key: Heartbeats since startup
>     value: '0'
>   - key: Frequency (Hz)
>     value: '0.000000'
>   - key: Vehicle type
>     value: Generic micro air vehicle
>   - key: Autopilot type
>     value: Generic autopilot
>   - key: Mode
>     value: ''
>   - key: System status
>     value: UNINIT
> ---

closed 2026-10-18 by github-actions[bot]