mavlink/mavros #1869: Request for Adding Optical Flow Message Support to MAVROS

Vladislavert, 2023-06-28:
### Issue details

This issue aims to highlight a potential solution for adding optical flow message support to MAVROS available in the [feature/optical_flow_msg](https://github.com/Vladislavert/mavros/tree/feature/optical_flow_msg) branch of a forked MAVROS repository.

Currently, MAVROS supports a wide range of MAVLink messages for various sensors and system components. However, [optical flow message](https://mavlink.io/en/messages/common.html#OPTICAL_FLOW) support is missing, limiting the utilization of optical flow sensors within MAVROS-based applications.

Additionally, it would be beneficial to have the ability to integrate custom optical flow sensors with MAVROS. I currently have my own sensor for optical flow, which I successfully use with ArduPilot. Adding support for custom optical flow sensors would allow users like me to seamlessly connect and utilize our own hardware within the MAVROS ecosystem.

### Additional Information:

[MAVLink Optical Flow Message](https://mavlink.io/en/messages/common.html#OPTICAL_FLOW)

vooon, 2023-06-28:
Hi Vlad,
Just make a PR.

BTW it's preferable to use vectors than plain fields with _x/_y.

closed 2023-07-01 by Vladislavert

ZhongmouLi, 2024-03-14:
Yes, I believe this feature is really needed, at least, for me.

BTW, I tried very hard to obtain optical flow data in mavros as I had thought I missed something in settings.