mavlink/mavros #1870: Question: ros2 topic info /mavros/setpoint_velocity/cmd_vel(_unstamped) - Ardupilot Mode?

Scoeerg, 2023-06-29:
### Issue details

I would like to control the robot in MAVROS2 using [geometry_msgs/msg/Twist](http://docs.ros.org/en/melodic/api/geometry_msgs/html/msg/Twist.html), since it is the "standard" interface to control a robot in ROS2, see e.g. 

- [ROS2 Teleop Keyboard](http://wiki.ros.org/teleop_twist_keyboard) 
- [Nav2 Interface](https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2controller_plugin.html) _Method is called when a new velocity command is demanded by the controller server in-order for the robot to follow the global path. This method returns a geometry_msgs::msg::TwistStamped which represents the velocity command for the robot to drive._

## Expected behavior:

After running the Mavros node (Details, see below) and then publishing Twist-Messages into different  topics:

```
ros2 topic pub /mavros/setpoint_velocity/cmd_vel geometry_msgs/msg/TwistStamped '{linear: {x: 1.0}}'
```

which answers with

```
publisher: beginning loop
publishing #1: geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=1.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=0.0)))
```

```
ros2 topic pub /mavros/setpoint_velocity/cmd_vel geometry_msgs/msg/TwistStamped '{ header: {frame_id:  "/base_link"}, twist: { linear: { x: 0.2, y: 0 }, angular: { x: 0, y: 0, z: 2} } }'
```

which answers with

```
publisher: beginning loop
publishing #1: geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id='/base_link'), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=0.2, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=2.0)))
```

```
ros2 topic pub /mavros/setpoint_velocity/cmd_vel_unstamped geometry_msgs/msg/TwistStamped '{twist: {linear: {x: 1.0}}}'
```

which answers with:

```
publishing #1: geometry_msgs.msg.TwistStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id=''), twist=geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=1.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=0.0)))
``` 

The robot should start moving. Question:

**What mode should be Ardupilot be in? GUIDED? MANUAL?** What about the `frame_id`?

### MAVROS version and platform

Mavros: 2.5.0, installed from source
ROS2: Foxy
Ubuntu: 20.04. LTS

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 4.2.3. Ardurover

### Details

```
ros2 run mavros mavros_node --ros-args --params-file ./mavros_param.yaml
```

with `mavros_param.yaml`:

```
# mavros_param.yaml
[... 547 lines cut ...]
/mavros/vfr_hud
/mavros/vibration/raw/vibration
/mavros/vision_pose/pose
/mavros/vision_pose/pose_cov
/mavros/vision_speed/speed_twist
/mavros/vision_speed/speed_twist_cov
/mavros/vision_speed/speed_vector
/mavros/wheel_odometry/odom
/mavros/wind_estimation
/move_base_simple/goal
/parameter_events
/rosout
/tf
/tf_static
/uas1/mavlink_sink
/uas1/mavlink_source
```

`ros2 service list:`

``` 
/mavros/actuator_control/describe_parameters
...
/mavros_router/set_parameters_atomically
```

vooon, 2023-06-29:
Hi,

Since all setpoint_foo uses `SET_*_TARGET_*` i guess that APM need GUIDED mode. PX4 - OFFBOARD.

Scoeerg, 2023-06-30:
Hey @vooon Thanks for the feedback. That is what I expected. Now when calling

```
ros2 topic echo /mavros/state
```

I will get info:

```
header:
  stamp:
    sec: 1688126251
    nanosec: 855269776
  frame_id: ''
connected: true
armed: true
guided: true
manual_input: false
mode: AUTO
system_status: 4
```
which tells me I **am** in guided mode hence ```guided: true```. I set this using the RC-controller and set some APM parameter to set the FlightMode to auto if I press it. Now unfortunately, when trying to set flightmode in MAVROS using

```
ros2 service call /mavros/set_mode mavros_msgs/SetMode '{base_mode: 220}'
``` 

which according to the [message documentation on mavros_msgs/SetMode](http://docs.ros.org/en/noetic/api/mavros_msgs/html/srv/SetMode.html) should be ```MAV_MODE_AUTO_ARMED```, the ```mavros_node``` responds with:

```
[WARN] [1688126325.408404494] [mavros.cmd]: CMD: Unexpected command 11, result 2
```

Any suggestions?

vooon, 2023-06-30:
No, you literally should enable GUIDED mode, not guided-bit flag of base mode. I.e. custom_mode: "GUIDED".

vooon, 2023-07-03:
If i remember correctly plugin doesn't check frame_id. And i'm really unsure if Twist plugin has right type_mask for APM as it was initially developed for PX4.
I think better to do some experimentation with setpoint_raw.

Scoeerg, 2023-07-04:
Additional information can be found here:

1) type_mask discussion: #402 
2) Message Definition [mavros/PositionTarget](http://docs.ros.org/en/noetic/api/mavros_msgs/html/msg/PositionTarget.html)
3) [Mavlink documentation](https://ardupilot.org/dev/docs/mavlink-rover-commands.html#SET_POSITION_TARGET_LOCAL_NED) on `SET_POSITION_TARGET_LOCAL_NED` 

Based on your suggestion I worked on `mavros/setpoint_raw/local` which is a subscriber with `PositionTarget` msg. i.e. it takes 

```
std_msgs/Header header
uint8 coordinate_frame
uint16 type_mask
geometry_msgs/Point position
geometry_msgs/Vector3 velocity
geometry_msgs/Vector3 acceleration_or_force
float32 yaw
float32 yaw_rate
```

messages. With the ```uint16 type_mask``` ignore Bitmask, I will ignore all except for velocity in `x` and `y`:

| 0   | 0   | 0   | 0   | 1       | 1   | 1     | 1   | 1   | 1   | 0  | 0  | 1  | 1  | 1  | 1  |
|-----|-----|-----|-----|---------|-----|-------|-----|-----|-----|----|----|----|----|----|----|
| N/A | N/A | N/A | N/A | YawRate | Yaw | Force | AFZ | AFY | AFX | VZ | VY | VX | PZ | PY | PX |

which I believe will be binary `0000111111001111` and decimal `4047`. I again wrote a minimal example publisher:

```
from mavros_msgs.msg import PositionTarget
from geometry_msgs.msg import Point
from geometry_msgs.msg import Vector3
import rclpy
from rclpy.node import Node

class positiontargetpublisher(Node):

    def __init__(self):
        super().__init__('positiontargetpublisher')
        self.publisher_ = self.create_publisher(PositionTarget, '/mavros/setpoint_raw/local', 10)
        timer_period = 0.1  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.frame_id = 'FRAME_LOCAL_OFFSET_NED'

    def timer_callback(self):
        msg = PositionTarget()
        msg.header.stamp = self.get_clock().now().to_msg()
        msg.header.frame_id = self.frame_id
        msg.coordinate_frame = 7
        msg.type_mask = 4047
        msg.position.x = 0.0
        msg.position.y = 0.0
        msg.position.z = 0.0
        msg.velocity.x = 10.0
        msg.velocity.y = 10.0
        msg.velocity.z = 0.0
        msg.acceleration_or_force.x = 0.0
        msg.acceleration_or_force.y = 0.0
        msg.acceleration_or_force.z = 0.0
        msg.yaw = 0.0
        msg.yaw_rate = 0.0
        self.publisher_.publish(msg)
        self.get_logger().info('Publishing: "%s"' %msg)

def main(args=None):
    rclpy.init(args=args)
    publisher = positiontargetpublisher()
    rclpy.spin(publisher)
    publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

 which does not work. :(

vooon, 2023-07-04:
Hello,

As i remember, you should start publishing setpoints and then switch to GUIDED.
Also even type_mask feeling like you may freely select only needed fields, actual implementation on FC side usually expects only few exact combinations. Usually best way to know them - dive into FCU code...

Scoeerg, 2023-07-04:
Hi,

this is all fun and games, but this: 

> As i remember, you should start publishing setpoints and then switch to GUIDED 

means you cannot establish a velocity "live-stream" the rover (drone) follows as a reference, unless I keep toggling mode. i.e.

```
sampleTime = 50ms
while(True):
  set MANUAL mode
  writeVelocity(getCurrentVelocityIWant())
  set GUIDED mode 
  # robot actually drives with myvelocity
  wait(sampleTime)
``` 

What I've read about, too is [`FOLLOWING` mode](https://ardupilot.org/rover/docs/follow-mode.html) which enables a robot to follow the position of another robot (with an offset). This kind of sounds like one would be able to just "live-stream" a virtual's robot position/velocity. But I did not find this mode supported in MAVROS. 

> Also even type_mask feeling like you may freely select only needed fields, actual implementation on FC side usually expects only few exact combinations. Usually best way to know them - dive into FCU code...

Well this is just disappointing, but thank you for the heads-up.

Scoeerg, 2023-07-13:
Hi, 

as stated earlier, `GUIDED` mode is not made for reference tracking for the Rover/Drone. 

In Control Theory/Robotics terms [read this robotics.stackexchange question on difference path/trajectory planning](https://robotics.stackexchange.com/questions/6683/what-is-the-difference-between-motion-planning-and-trajectory-generation):

- `GUIDED` mode for handing a planned **trajectory** to the drone/rover
- `AUTO` mode for handing over a planned **path** to the drone/rover

but none of them are made for my intention: reference tracking. The closest implementation to reference tracking seems to be the `FOLLOWING` mode. But it seems it only follows position, rather than the entire state space of the system (position & velocity). 

My solution:

The standard way to give commands to a robot in ROS is using a [Twist Message](http://docs.ros.org/en/noetic/api/geometry_msgs/html/msg/Twist.html), as e.g. [Teleop Twist Keyboard](http://wiki.ros.org/teleop_twist_keyboard) in the early turtlebot tutorials or in [NAV2](https://navigation.ros.org).

To now follow the twist message as a reference, I implemented a Node that subscribes any Twist message (topic `\twist` in the code) and publishes a [Mavros OverrideRCIn](http://docs.ros.org/en/hydro/api/mavros/html/msg/OverrideRCIn.html) (topic `\mavros\rc\override` in code). 

**Be aware this code is rather specific to your situation, as it depends on parameters set in Ardupilot and your robot.** Those are:

```
self.steeringMin = 1100 
self.steeringMax = 1900
self.throttleMin = 1100
self.throttleMax = 1900
self.angularVelocityMax = 0.5
self.angularVelocityMin = (-1)*self.angularVelocityMax
self.velocityMax = 2.78 #m/s
self.velocityMin = (-1)*self.velocityMax

msg.channels = [STEERING, 0, THROTTLE, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

mapping(twistMessage)
    ...
    return steering, throttle 
```
The node "translating" Twist to RCOverride:

```
import rclpy
from rclpy.node import Node
from mavros_msgs.msg import OverrideRCIn
from geometry_msgs.msg import Twist

class Twist2RcOverride(Node):

    def __init__(self):
        super().__init__('rcoverride_publisher')
        self.rcoverridepublisher = self.create_publisher(OverrideRCIn, '/mavros/rc/override', 10)
        self.twistsubscription = self.create_subscription(Twist, '/twist', self.twistcallback, 10)
        timer_period = 0.05  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        ### <---- Parameters ---- >
        self.steeringMin = 1100
        self.steeringMax = 1900
        self.throttleMin = 1100
        self.throttleMax = 1900
        self.angularVelocityMax = 0.5
        self.angularVelocityMin = (-1)*self.angularVelocityMax
        self.velocityMax = 2.78
        self.velocityMin = (-1)*self.velocityMax
        ### <---- Parameters ---- >
        # Initializing
        self.steering = 1500
        self.throttle = 1500

    def timer_callback(self):
        msg = OverrideRCIn()
        # msg.channels = [STEERING, 0, THROTTLE, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        msg.channels = [int(self.steering), 0, int(self.throttle), 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
        self.rcoverridepublisher.publish(msg)
        self.get_logger().debug('Published RC-Override to Pixhawk using MavROS: %s' %(" ".join([str(x) for x in msg.channels])))

    def twistcallback(self, twistMessage):
        self.get_logger().debug('I heard: %s' %(str(twistMessage)))
        self.steering, self.throttle = self.mapping(twistMessage)

    def mapping(self, twistMessage):
        steering = 1500 + 400*twistMessage.angular.z/self.angularVelocityMax
        throttle = 1500 + 400*twistMessage.linear.x/self.velocityMax
        if steering < self.steeringMin:
            steering = self.steeringMin
        elif steering > self.steeringMax:
            steering = self.steeringMax
        if throttle < self.throttleMin:
            throttle = self.throttleMin
        elif throttle > self.throttleMax:
            throttle = self.throttleMax
        return steering, throttle

def main(args=None):
    rclpy.init(args=args)
    node = Twist2RcOverride()
    rclpy.spin(node)
    node.destroy_node()
    rclpy.shutdown()
if __name__ == '__main__':
    main()
```

which works just fine, hence I will close the issue. But be aware: the solution is not ideal, my hope is that there is a closed control loop on the robot that somehow digests the RC-channels to references in velocity, otherwise it's open loop. To know, one would need to dig through the Ardupilot code. Thanks @vooon :)

closed 2023-07-13 by Scoeerg