mavlink/mavros #1872: Question about sending external pose to mavros

prashantganesh-episci, 2023-06-29:
Hi,
I am following [PX4's instructions](https://docs.px4.io/main/en/ros/external_position_estimation.html#relaying-pose-data-to-px4) on sending external vison based pose and it requests sending pose estimates to mavros/vision_pose/pose. What is the difference between sending the pose to that topic vs sending it to /mavros/local_position/pose? 
Digging into the setpoint_position.cpp it looks like the /mavros/local_position/pose updates the local position: 
[image]

----

### Issue details

I also noticed that the setpoint_poisition.cpp subscribes to /mavros/local_position/pose while the local_position.cpp plugin subscribes to the same topic which creates a loop. Is this something that needs to be changed? 

[image]

### MAVROS version and platform

Mavros: 1.16.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

### Node logs

```
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
... logging to /home/prashantganesh@int-episci.com/.ros/log/818f93de-16bc-11ee-ab6e-c550d7443979/roslaunch-ironman-1527182.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ironman:43467/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
[... 421 lines cut ...]
    level: 2
    name: "mavros: Time Sync"
    message: "No events recorded."
    hardware_id: "udp://:14540@127.0.0.1:14557"
    values: 
      - 
        key: "Timesyncs since startup"
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---
```

Kian-Wee, 2023-07-04:
If Im not mistakened the /mavros/local_position/pose topic is only for subscribing to the UAV's local position after it runs through the UAV's internal ekf loop, which is what setpoint_position uses to check if its at the desired location. If you want to send in an external pose estimate you need to use the /mavros/vision_pose/pose topic which is an PRE ekf estimation.

vooon, 2023-07-04:
One plugin can subscribe to other plugin. That's fine. But on graph you'll see that mavros both publishes and subscribes some topics.

On your diagnostics there no connection, so you'll never get any response from the FCU.

prashantganesh-episci, 2023-07-06:
@Kian-Wee and @vooon sounds good. I will try publishing to the /vision_pose/pose topic.
Thank you!

Inhan1998, 2023-07-12:
@Kian-Wee @vooon We publish the /vision_pose/pose topic, but /mavros/local_position/pose is not affected. 
We set EKF2_AID_MASK to 280 and SYS_MC_EST_GROUP to 1(ekf2). and We are using melodic.

Inhan1998, 2023-07-12:
* /mavros/vision_pose/tf/child_frame_id: vision_estimate
 * /mavros/vision_pose/tf/frame_id: odom
 * /mavros/vision_pose/tf/listen: False
 * /mavros/vision_pose/tf/rate_limit: 10.0
 * /mavros/vision_speed/listen_twist: True
 * /mavros/vision_speed/twist_cov: True

Should I change these params?

vooon, 2023-07-13:
No, i think you probably need to change something in FC's params.

Inhan1998, 2023-07-13:
> No, i think you probably need to change something in FC's params.

@vooon Could you tell me more detail about that?
What's the FC's params?

Inhan1998, 2023-07-13:
I can't find those params in my qgc
[image]

Kian-Wee, 2023-07-14:
> I can't find those params in my qgc [image]

Those params are for LPE, Not EKF2

What rate are you sending in your vision_pose at? It should be [minimally 30hz(with covariance) or 50hz(without)](https://docs.px4.io/main/en/ros/external_position_estimation.html#px4-mavlink-integration). You should be able to verify if the FC is receiving it by typing listener vehicle_visual_odometry into the mavlink console in QGC if you enable the GCS bridge.

closed 2026-10-18 by github-actions[bot]