mavlink/mavros #1873: mavros/altitude does not publish

gregoryw3, 2023-07-03:
### Issue details

mavros/altitude does not publish altitude from FC. As far as I can see, Ardupilot runs EKF on the FC and calculates the altitude. MAVROS does not seem to properly capture this value which can be seen from both mavproxy.py and Mission Planner.

Does mavros need some service call to enable this? Or need to set home position? I have this in the enable plugins folder and in the mavros log I can see the plugin is initialized. I know for IMU data you need to set a stream rate via (or at least this is how I got IMU data to publish as maybe it just hasn't worked for me and others I'm working with):
`ros2 service call /mavros/set_stream_rate mavros_msgs/StreamRate "{stream_id: 0, message_rate: 10, on_off: 1}"`

I have used this on two boards so far and both have the issue, `Mateksys H743-SLIM V3` and `Flywoo GOKU GN 745 AIO with 40A ESC`

Overall I don't know if this is a configuration issue, or I'm missing something or the plugin doesn't work.

### MAVROS version and platform

Mavros: Unknown, version that is installed with `ros-foxy-mavros`
ROS: Foxy
Ubuntu: 20.04.6 LTS

### Autopilot type and version

[ X ] ArduPilot
[  ] PX4

Version: V4.5.0-dev (9bcc2604)

### Node logs

```bash
ros2 topic echo /mavros/altitude
```
then nothing.

When I made a quick node to output /mavros/altitude it also did not output any data.

```cpp
#include <rclcpp/rclcpp.hpp>
#include <mavros_msgs/msg/altitude.hpp>
using std::placeholders::_1;

class MinimalSubscriber : public rclcpp::Node
{
public:
    MinimalSubscriber()
            : Node("minimal_subscriber")
    {
        subscription_ = this->create_subscription<mavros_msgs::msg::Altitude>(
                "/mavros/altitude", rclcpp::SensorDataQoS(), std::bind(&MinimalSubscriber::topic_callback, this, _1));
    }

private:
    void topic_callback(const mavros_msgs::msg::Altitude::SharedPtr msg) const
    {
        RCLCPP_INFO(this->get_logger(), "I heard: '%d'", msg->relative);
    }
    rclcpp::Subscription<mavros_msgs::msg::Altitude>::SharedPtr subscription_;
};

int main(int argc, char * argv[])
{
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<MinimalSubscriber>());
    rclcpp::shutdown();
    return 0;
}
```

### Diagnostics

ros2 topic echo /diagnostics

```yaml
header:
  stamp:
    sec: 1688412553
    nanosec: 943159319
  frame_id: ''
status:
- level: "\0"
  name: 'mavros: MAVROS UAS'
  message: connected
  hardware_id: uas:///uas1
  values: []
- level: "\x02"
  name: 'mavros: GPS'
  message: No satellites
  hardware_id: uas:///uas1
  values:
  - key: Satellites visible
    value: '0'
  - key: Fix type
    value: '0'
  - key: EPH (m)
    value: Unknown
  - key: EPV (m)
    value: Unknown
- level: "\x01"
  name: 'mavros: Mount'
  message: Can not diagnose in this targeting mode
  hardware_id: uas:///uas1
  values:
  - key: Mode
    value: '255'
- level: "\x02"
  name: 'mavros: System'
  message: Sensor health
  hardware_id: uas:///uas1
  values:
  - key: Sensor present
    value: '0x5770FC0F'
  - key: Sensor enabled
    value: '0x52609C0F'
  - key: Sensor health
    value: '0x47509C0B'
  - key: 3D gyro
    value: Ok
  - key: 3D accelerometer
    value: Ok
  - key: 3D magnetometer
    value: Fail
  - key: absolute pressure
    value: Ok
  - key: 3D angular rate control
    value: Ok
  - key: attitude stabilization
    value: Ok
  - key: yaw position
    value: Ok
  - key: motor outputs / control
    value: Ok
  - key: AHRS subsystem health
    value: Fail
  - key: Terrain subsystem health
    value: Ok
  - key: Battery
    value: Ok
  - key: pre-arm check status. Always healthy when armed
    value: Fail
  - key: propulsion (actuator, esc, motor or propellor)
    value: Ok
  - key: CPU Load (%)
    value: '19.0'
  - key: Drop rate (%)
    value: '0.0'
  - key: Errors comm
    value: '0'
  - key: 'Errors count #1'
    value: '0'
  - key: 'Errors count #2'
    value: '0'
  - key: 'Errors count #3'
    value: '0'
  - key: 'Errors count #4'
    value: '0'
- level: "\x01"
  name: 'mavros: Battery'
  message: Low voltage
  hardware_id: uas:///uas1
  values:
  - key: Voltage
    value: '0.00'
  - key: Current
    value: '-2.0'
  - key: Remaining
    value: '20.0'
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '1899'
  - key: Frequency (Hz)
    value: '1.000207'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: STABILIZE
  - key: System status
    value: STANDBY
---

```

### Check ID

ros2 run mavros mav checkid

```bash
[ERROR] [1688413155.543936369] [mavpy_ipg0]: wait for service time out: {topic}
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/cmd/__init__.py", line 34, in __getattr__
    return object.__getattribute__(self, key)
AttributeError: 'CliClient' object has no attribute 'uas_settings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/mavros/mav", line 14, in <module>
    cli()
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/cmd/checkid.py", line 129, in checkid
    checker = Checker(client=client, follow=follow, watch_time=watch_time)
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/cmd/checkid.py", line 44, in __init__
    self.tgt_ids = client.uas_settings.target_ids
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/cmd/__init__.py", line 36, in __getattr__
    return getattr(self.cli, key)
  File "/usr/lib/python3.8/functools.py", line 967, in __get__
    val = self.func(instance)
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/base.py", line 116, in uas_settings
    pd = call_get_parameters(node=self, node_name=self.mavros_ns, names=names)
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/utils.py", line 75, in call_get_parameters
    wait_for_service(client, lg)
  File "/opt/ros/foxy/lib/python3.8/site-packages/mavros/base.py", line 55, in wait_for_service
    raise ServiceWaitTimeout(topic)
mavros.base.ServiceWaitTimeout: mavros/get_parameters

```

vooon, 2023-07-04:
Please try to build from source. Foxy is EOLed, so perhaps the version is old. Btw you can ask `apt` to tell package version.

gregoryw3, 2023-07-04:
> Please try to build from source. Foxy is EOLed, so perhaps the version is old. Btw you can ask `apt` to tell package version.

I'll try building from source however the project I'm working on we're using a radxa zero which only officially supports Ubuntu 20. Do you know if the latest version can work on Ubuntu 20? Also Foxy just received its last package update, so although it's EoL now it wasn't a week ago.

vooon, 2023-07-04:
You should be able to build latest sources for Foxy.

gregoryw3, 2023-07-05:
> You should be able to build latest sources for Foxy.

I've tried building mavros from source and it still didn't output any altitude.

itskalvik, 2024-07-30:
I've had the same issue, I'm on Humble

vignesht-paar, 2026-01-07:
I am also experiencing the same issue, The distro is **humble** any solution available?

closed 2026-10-18 by github-actions[bot]