mavlink/mavros #1874: The apm.launch file reports udp0: sendto: Invalid argument. Please help!

AyushMaria, 2023-07-14:
----

### Issue details
When I run roslaunch mavros apm.launch I face the problem below.

### MAVROS version and platform

```
Mavros: 2.5.0
ROS: Noetic
Ubuntu: 20.04
```

### Autopilot type and version

```
ArduPilot
Version: Copter-4.3.7
```

### Node logs

```
[ INFO] [1689315821.530053213]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ERROR] [1689315821.624194832]: udp0: sendto: Invalid argument
terminate called after throwing an instance of 'std::system_error'
  what():  Resource deadlock avoided
[mavros-1] process has died [pid 99072, exit code -6, cmd /home/aoi/catkin_ws/devel/lib/mavros/mavros_node __name:=mavros __log:=/home/aoi/.ros/log/2535709a-2206-11ee-9b1a-bd3f2a9f5ee7/mavros-1.log].
log file: /home/aoi/.ros/log/2535709a-2206-11ee-9b1a-bd3f2a9f5ee7/mavros-1*.log
[mavros-1] restarting process
```

### Diagnostics

```
aoi@Aoi:~$ rostopic echo -n1 /diagnostics
header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "Node starting up"
    hardware_id: ''
    values: []
---
```

### Check ID

```
aoi@Aoi:~$ rosrun mavros checkid

ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

closed 2023-07-19 by AyushMaria

sacchinbhg, 2023-09-25:
I were you able to solve this issue? @AyushMaria

AyushMaria, 2023-09-25:
> I were you able to solve this issue? @AyushMaria 

Yes I was. When you run sitl check the udp port mentioned in the --out parameter in mavproxy and see if they're matching with the ports you're connecting MAVROS to.

sacchinbhg, 2023-09-25:
Thank you for this, was able to make it work!

Khalilmeister, 2024-03-27:
can you please elaborate more

Anaskherro, 2024-04-19:
> > I were you able to solve this issue? @AyushMaria
> 
> Yes I was. When you run sitl check the udp port mentioned in the --out parameter in mavproxy and see if they're matching with the ports you're connecting MAVROS to.

Can you please specify the steps that u followed to make this work ?

Ashnidh, 2024-07-05:
Hi everyone! I also encountered the same error and was able to rectify it using the comments by @AyushMaria and referencing [this part of the ArduPilot SITL documentation](https://ardupilot.org/dev/docs/using-sitl-for-ardupilot-testing.html#using-mavproxy-udp-forwarding). Full credit to them.

Here is a detailed walkthrough to fixing the issue - 
1. Observe the MAVROS log you get below the error. It mentions the UDP port MAVROS is trying to connect to. In my case, it said - 
`process[mavros-2]: started with pid [7941]
[ INFO] [17201540.2241190]: FCU URL: udp://127.0.0.1:14551@14555
[ INFO] [17201540.2242534]: udp0: Bind address: 127.0.0.1:14551
[ INFO] [17201540.2270953]: udp0: Remote address: 0.0.56.219:14550
[ INFO] [17201540.2271904]: GCS bridge disabled
`

Here `FCU URL: udp://127.0.0.1:14551@14555` states that MAVROS is trying to connect to a _UDP port_ `127.0.0.1:14551`. It might not be able to find such a port.

2. After starting SITL, use this command in MavProxy terminal - 
`output`
It will tell you how many and which ports are being used to output the data. In my case, it said - 
`STABILIZE> 1 outputs`
`0: 127.0.0.1:14550`

We can clearly see that the output port of SITL (127.0.0.1:14550) and the required connection port of MAVROS(127.0.0.1:14551) do not match. 

3. You may add or remove the required output port to SITL using the `output add <port>` or `output  remove <port>` commands in MavProxy terminal. 
I ran - `output add 127.0.0.1:14551` and the MAVROS error stopped showing up and I was connected. 

However, you may always need to change the output ports or you can change the default connection port of MAVROS in the apm.launch file to your SITL default output port.

Hope it helps. Thanks.

AyushMaria, 2024-07-05:
Perfectly explained