mavlink/mavros #1875: RTK setup - Hardware & Software

Bmoradi93, 2023-07-18:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi friends,
I'm trying to use RTK in my application to increase my localization accuracy.
My hardware:
- Drone Kit: HolyBro X500 PX4 V2
- GPS and RTK: UBLOX FP9 - base station and robot.
- I have radio telemetry on both my ground computer and and on the robot.
- Ground computer is connected to the base station using USB and I'm able to read the base station GPS.

My software:
- QGroundControl is used and I used it to test the robot.
- RTK is also tested in QGroundControl and it works fine!

Now, I'm going to use MAVSDK and MAVROS in my autonomy software and I need to have RTK as well.
When I use MAVROS, I'm not able to get any RTK data for gps/rtk topics. I'm able to read gps/raw though. But no RTK.

Please instruct me how to do my Hardware/Software configuration to enable RTK in MAVROS so I can use it in my application.

### MAVROS version and platform

Mavros: latest
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

I'm using PX4 autopilot.

Version: latest

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

vooon, 2023-07-21:
Hi,

MAVROS just feeds RTCM data to FCU. You should map your base station's feed to gps_rtk plugin.

See also: https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/gps_rtk.cpp#L39

Bmoradi93, 2023-08-09:
My base and rover are UBLOX F9P.
I used UCenter to configure RTCM3 output from the base station.

The base station is connected to the ground computer through the ttyACM0 USB port.

The [ROS driver](https://github.com/ros-agriculture/ublox_f9p#ros-driver-for-ublox-zed-f9p-receiver) does not read the RTCM message.

Can you please explain how to read the RTCM data from this GPS?
Is there an API for this?

gitsr-sys, 2023-10-02:
@vooon 
I am having the same setting and same issue.
In short, I can confirmed RTK GPS streaming, in mavlink `gps status`, it showed `GPS_FIX_TYPE = 6 (RTK Fixed, 3D position)`.
But, when I run `mavros`, no outputs from gps/rtk topics: `mavros/gps_rtk/rtk_baseline`, `mavros/gps_rtk/send_rtcm`, and `/mavros/gpsstatus/gps1/rtk`.

I do not understand `map your base station's feed to gps_rtk plugin`....
In the `px4_config.yaml` and `px4_pluginlists.yaml`, I cannot find any `rtk` setting.
Could you please help me to get `/mavros/gpsstatus/gps1/rtk` data?

Thank you in advance.

diablo1281, 2023-10-02:
I would also appreciate some clarification. 

I am working on injecting Mavlink messages from RTK station which is streaming RTCM to the network.

Is it also possible to stream from UAS to multiple GCSes?

gitsr-sys, 2023-10-10:
@vooon any thought?

vooon, 2023-10-16:
@diablo1281 https://mavlink.io/en/messages/common.html#GPS_RTCM_DATA as you can see, message doesn't have target, so it's broadcast one. Just make other nodes to hear it with something like mavlink-router (or built-in router on ros2).

@gitsr-sys mavros is not a driver for your GPS device. It doesn't understand serial protocol of your receiver, like QGC does. Instead there must be a ros node which talks to the device and provide gps fix and rtcm stream. If it's not already done by existing driver i suggest that you should white your own. That's must be easy as there plenty of libraries to talk to Ublox, i almost sure that it'll be about 100 LOC of python.

Bmoradi93, 2024-05-24:
Yes. MAVROS RTK plugin is able to send RTCM data to FCU.
The problem here is getting the correct RTCM3 message from UBLOX F9P Base Station.
UBLOX has a ROS package and I tested it. However, it does not provide RTCM3 data.

closed 2026-10-18 by github-actions[bot]