mavlink/mavros #1881: /mavros/local_position/odom data wrong

Kasho-2001, 2023-08-01:
----
***Is the odometry in gazebo hallucinating?!***
We use the iris_vision drone for a university project, where we give out some velocity commands to "/mavros/setpoint_velocity/cmd_vel"
In the first 2 minutes the drone behaves exactly as we want, however after this time, the drone suddenly begins to drift in the z direction. After printing out the z-value of  /mavros/local_position/odom we noticed, that it has a huge difference to what we can see in the gazebo simulation.  
The problem just occurs if we set a horizontal and yaw velocity together. It is so strange, because it works perfectly in the first minutes.

I would appreciate any help!

### MAVROS version and platform.

Mavros: ?1.16.0?
ROS: ?Noetic?
Ubuntu: ?20.04?

rishabhdevyadav, 2024-07-10:
Hi, anyone find the solution for this. I am also facing same issue. 
I launched:
cd PX4-Autopilot
source ~/catkin_ws/devel/setup.bash  
source Tools/simulation/gazebo-classic/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic
roslaunch px4 mavros_posix_sitl.launch

And I can see the drone in gazebo, till here everything is completely normal. I am printing mavros/local_position/pose and after 4-5 seconds as soon as I am getting wrong z: position. 
Quadrotor is on ground and it is printing 
    x: 0.011687700636684895
    y: 0.004684464540332556
    z: -1.1767728328704834

Slowly z is drifting slowly. However 'base_link' position is correct. Any solution ?
I am using docker. Ubuntu 20, Noetic.

erichjl, 2024-12-12:
> Hi, anyone find the solution for this. I am also facing same issue. I launched: cd PX4-Autopilot source ~/catkin_ws/devel/setup.bash source Tools/simulation/gazebo-classic/setup_gazebo.bash ( p w d [...]

same problem, have you solve it? also Ubuntu20.04 Noetic PX4 1.15

3AMengineer, 2026-09-20:
please upload your Simulation files. did you make any changes in gazebo-iris 
?