mavlink/mavros #1882: PX4 support in MAVROS for ROS2?

DiTo97, 2023-08-05:
Hi all,

As the title goes, I would like to know what is the state of PX4 support in MAVROS for ROS2 (which is the default branch) considering that the [official PX4 docs related to MAVROS](https://docs.px4.io/main/en/ros/mavros_installation.html) still report that it is a ROS1-only package.

mzahana, 2023-09-04:
I am also interested to know about this state.

I can't find documentation on how to use mavros with px4 in ros2.

Vicidel, 2023-11-27:
From what I see in my usage, MAVROS2 supports PX4 perfectly fine. I do not have any information missing by using MAVROS2. From what I know the currently recommended interface between PX4 and ROS2 is the [uXRCE-DDS bridge](https://docs.px4.io/main/en/middleware/uxrce_dds.html)