mavlink/mavros #1886: Only Heartbeat seen over UDP Connection

hrrymtthews, 2023-08-11:
### Issue details

When running Mavros onboard, I don't receive any Mavlink messages except a heartbeat at 1Hz. My current configuration consists of a PI connected to a FCU over serial and transmitting those messages to my main machine on the same network via UDP.

When running Mavproxy on the same board over the same connection (`mavproxy.py --master=/dev/ttyS0 --out=udp:192.168.0.112:14550`), there is no issue; all Mavlink messages are detected by QGC.

- All Mavros topics can be seen on the ROS network from my main machine.
- Tested with a TCP connection with the same outcome.
- Tested on different hardware and fresh ROS2 + Mavros install.

Occasionally during initialising mavros_node, many instances of the error message below were posted. 

```
Error: mavconn: 1: DROPPED Message-Id 0 [0 bytes] IDs: 0.0 Seq: 0: MAVConnUDP::send_message: TX queue overflow at line 165 in ./src/interface.cpp
```

I couldn't find any way to consistently repeat this error.

### MAVROS version and platform

Mavros: 2.5.0-1jammy.20230722.001659 (ros-humble-mavros package)
ROS: Humble
Ubuntu: 22.04

### Autopilot type and version

[ ] ArduPilot
[ x ] PX4

(Problem persists on Ardupilot also)

Version: 1.13.3

### Node logs

Result of `ros2 run mavros mavros_node --ros-args --params-file params.yaml`:

```
[INFO] [1691755241.552538830] [mavros_node]: Starting mavros_node container
[INFO] [1691755241.553274403] [mavros_node]: FCU URL: /dev/ttyS0:921600
[INFO] [1691755241.553411017] [mavros_node]: GCS URL: udp://:14550@192.168.0.131:14550
[INFO] [1691755241.553477580] [mavros_node]: UAS Prefix: /uas1
[INFO] [1691755241.553538309] [mavros_node]: Starting mavros router node
[INFO] [1691755241.646831226] [mavros_router]: Built-in SIMD instructions: ARM NEON
[INFO] [1691755241.647054663] [mavros_router]: Built-in MAVLink package version: 2023.6.6
[INFO] [1691755241.647139246] [mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all csAirLink cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[INFO] [1691755241.647217007] [mavros_router]: MAVROS Router started
[INFO] [1691755241.647479663] [mavros_router]: Requested to add endpoint: type: 0, url: /dev/ttyS0:921600
[INFO] [1691755241.647683361] [mavros_router]: Endpoint link[1000] created
[INFO] [1691755241.649890913] [mavros_router]: link[1000] opened successfully
[INFO] [1691755241.654353101] [mavros_router]: Requested to add endpoint: type: 1, url: udp://:14550@192.168.0.131:14550
[INFO] [1691755241.654614819] [mavros_router]: Endpoint link[1001] created
[INFO] [1691755241.656425601] [mavros_router]: link[1001] opened successfully
[INFO] [1691755241.656898465] [mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[INFO] [1691755241.657264715] [mavros_router]: Endpoint link[1002] created
[INFO] [1691755241.762190496] [mavros_router]: link[1002] opened successfully
[INFO] [1691755241.762595184] [mavros_node]: Starting mavros uas node
[WARN] [1691755242.799077319] [mavros]: diagnostic_updater: No HW_ID was set. This is probably a bug. Please report it. For devices that do not have a HW_ID, set this value to 'none'. This warning only occurs once all diagnostics are OK. It is okay to wait until the device is open before calling setHardwareID.
[INFO] [1691755242.808682319] [mavros]: UAS Executor started, threads: 4
[INFO] [1691755243.139785808] [mavros]: Plugin actuator_control created
[INFO] [1691755243.140156069] [mavros]: Plugin actuator_control initialized
[INFO] [1691755243.493093516] [mavros]: Plugin adsb created
[INFO] [1691755243.493506850] [mavros]: Plugin adsb initialized
[INFO] [1691755243.537439454] [mavros]: Plugin altitude created
[INFO] [1691755243.538387162] [mavros]: Plugin altitude initialized
[INFO] [1691755243.580565912] [mavros]: Plugin cam_imu_sync created
[INFO] [1691755243.580853152] [mavros]: Plugin cam_imu_sync initialized
[INFO] [1691755243.632943152] [mavros]: Plugin camera created
[INFO] [1691755243.633811641] [mavros]: Plugin camera initialized
[INFO] [1691755243.683713152] [mavros]: Plugin cellular_status created
[... 309 lines cut ...]
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '0'
  - key: Frequency (Hz)
    value: '0.000000'
  - key: Vehicle type
    value: Generic micro air vehicle
  - key: Autopilot type
    value: Generic autopilot
  - key: Mode
    value: ''
  - key: System status
    value: UNINIT
---
```
(Repeated)

### Check ID

Result of `ros2 run mavros checkid`:

```
No executable found
```

vooon, 2023-08-11:
Hello,

I'm sure that you're getting HEARTBEATs from the MAVROS itself, and not from FCU. Please check in the inspector source ids and type (should be ONBOARD COMPUTER's one's).

Are you sure that port is ttyS0? Typically rpi have ttyAMA. Also are you sure that baud rate is correctly set on FC's side? Is it 921600?

Look at router diagnostic for ep 1000, they all zero.

hrrymtthews, 2023-08-11:
Hi Vooon, thanks for the fast reply and helpful suggestions.

I currently don't have access to the setup so I'll test more on Monday. I believe I set the baudrate of the FCU to 921600 when flashing the firmware. I'll check the port and router diagnostics when I can and send another update. Thanks again.

hrrymtthews, 2023-08-14:
You're correct in assuming the heartbeat was from Mavros. The type of the message is ONBOARD_CONTROLLER and the source is ONBOARD_COMPUTER. 

Switching port to AMA0 seems to have resolved the issue (after first disabling blutooth with `dtoverlay=pi3-disable-bt` in config.txt). Now all Mavlink messages are detected by the autopilot and the source is the FCU. Thank you for your help👍

closed 2023-08-14 by hrrymtthews