mavlink/mavros #1893: Not able to takeoff my drone using Mavros

Nikhiljohnson21, 2023-08-31:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

Pushkar-05, 2023-09-01:
#Question - Does anyone know mavros and fcu connection? My Mavros is publishing the setpoints yet it is not being taken by fcu because of which gazebo drone is not moving. Can anyone help with this?
#ROS #Neotic #Mavros #PX4 #Drone #Linux

vooon, 2023-09-04:
Hmm, i do not see any problems except battery voltage. Are you sure, that there wasn't 'prearm' messages from the FCU?
Also please verify arming state in state topic.

Nikhiljohnson21, 2023-09-22:
The problem was that for takeoff the mode automatically changed into guided mode inside ardupilot firmware which needed gps fix and i was trying it indoor , when i tried it outside it worked and now i am looking for alternatives for gps and i tired with PX4flow optical sensor and i wasnt able to connect it properly.
I followed the steps said in Link: https://ardupilot.org/copter/docs/common-px4flow-overview.html
The green, blue, and orange LED lights on PX4Flow were flashing when connected.
However, I do not see any optical flow data on mission planner and there in no tick mark on Initial Setup>>optional hardware>>Optical Flow section where it should be shown. does anyone know about this

vooon, 2023-09-24:
Sorry, i haven't had flow module.

closed 2026-10-17 by github-actions[bot]