mavlink/mavros #1895: Is there a working example of ROS 2 launch file for px4?

mzahana, 2023-09-05:
I am not able to find a `px4.launch.py` script to launch mavros to communicate with PX4.

Also, I could not find instructions on how to use mavros in ROS 2.

@vooon Can you please give some hints?

Thanks.

vooon, 2023-09-06:
https://github.com/mavlink/mavros/blob/ros2/mavros/launch/px4.launch should be working, isn't it?

mzahana, 2023-09-06:
Thanks @vooon for your reply.

Yes, it is working.
I wanted a Python kind of launch file so I can include many launch files into one.
So, I wrote a launch.py file, and it worked, at least for PX4.

Thanks.

mzahana, 2023-09-06:
Here is the script I wrote.

```python
#!/usr/bin/env python3

import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch_ros.actions import Node
from ament_index_python import get_package_share_directory
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    ld = LaunchDescription()

    fcu_url = LaunchConfiguration('fcu_url')
    gcs_url = LaunchConfiguration('gcs_url')
    tgt_system = LaunchConfiguration('tgt_system')
    tgt_component = LaunchConfiguration('tgt_component')
    log_output = LaunchConfiguration('log_output')
    pluginlists_yaml = LaunchConfiguration('pluginlists_yaml')
    config_yaml = LaunchConfiguration('config_yaml')
    fcu_protocol = LaunchConfiguration('fcu_protocol')
    namespace = LaunchConfiguration('namespace')

    fcu_url_arg = DeclareLaunchArgument(
        'fcu_url',
        default_value='/dev/ttyACM0:57600',
        description="FCU interface Example /dev/ttyACM0:57600"
    )
    gcs_url_arg = DeclareLaunchArgument(
        'gcs_url',
        default_value='',
        description="GCS bridge interface"
    )
    tgt_system_arg = DeclareLaunchArgument(
        'tgt_system',
        default_value='1',
        description="MAVLink ID of the target system"
    )
    tgt_component_arg = DeclareLaunchArgument(
        'tgt_component',
        default_value='1',
        description="MAVLink component ID inside the target system"
    )
    log_output_arg = DeclareLaunchArgument(
        'log_output',
        default_value='screen',
        description="Type of log output"
    )
    package_name = 'mavros'
    file_name = 'launch/px4_pluginlists.yaml'
    package_share_directory = get_package_share_directory(package_name)
    file_path = os.path.join(package_share_directory, file_name)
    pluginlists_yaml_arg = DeclareLaunchArgument(
        'pluginlists_yaml',
        default_value=file_path,
        description="File path of the pluginlist_yaml file"
    )
    package_name = 'mavros'
    file_name = 'launch/px4_config.yaml'
    package_share_directory = get_package_share_directory(package_name)
    file_path = os.path.join(package_share_directory, file_name)
    config_yaml_arg = DeclareLaunchArgument(
        'config_yaml',
        default_value=file_path,
        description="File path of the mavros config_yaml file"
    )
    fcu_protocol_arg = DeclareLaunchArgument(
        'fcu_protocol',
        default_value='v2.0',
        description="MAVLink version"
    )
    namespace_arg = DeclareLaunchArgument(
        'namespace',
        default_value='mavros',
        description="Node namespace"
    )

    # MAVROS node
    mavros_node = Node(
        namespace=namespace,
        package='mavros',
        executable='mavros_node',
        parameters=[{
                'fcu_url': fcu_url,
                'gcs_url': gcs_url,
                'tgt_system': tgt_system,
                'tgt_component': tgt_component,
                'fcu_protocol': fcu_protocol,
                'config_yaml': config_yaml,
                'pluginlists_yaml': pluginlists_yaml, 
        }]
    )

    ld.add_action(fcu_url_arg)
    ld.add_action(gcs_url_arg)
    ld.add_action(tgt_component_arg)
    ld.add_action(tgt_system_arg)
    ld.add_action(log_output_arg)
    ld.add_action(fcu_protocol_arg)
    ld.add_action(namespace_arg)
    ld.add_action(pluginlists_yaml_arg)
    ld.add_action(config_yaml_arg)
    ld.add_action(mavros_node)

    return ld
```

YazdanshensAmin, 2023-09-19:
Could you please explain more about what you did and how did you manage the ROS 2 launch file for px4

> Here is the script I wrote.
>
> ```python
> #!/usr/bin/env python3
>
> import os
> from launch import LaunchDescription
> from launch.actions import DeclareLaunchArgument
> from launch_ros.actions import Node [...]

Hello, I have the same problem. Could you please explain more about how you fixed the issue by running the Python script?
tnx

mzahana, 2023-09-19:
Hi @YazdanshensAmin 
I am not sure what problem you refer to, but I just used the 
https://github.com/mavlink/mavros/blob/ros2/mavros/launch/px4.launch
launch file and it worked. You can use the following command
```
ros2 launch mavros px4.launch
```

For me, I also wanted to use a python launch file, so I created one and I shared it in my previous reply. That launch file can be run using the following command
```
ros2 launch pkg_name px4.launch.py
```

YazdanshensAmin, 2023-09-19:
When I try the

`ros2 launch mavros px4.launch`

I get the following error:
/
/opt/ros/foxy/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
[INFO] [launch]: All log files can be found below /home/aminys/.ros/log/2023-09-19-12-02-40-071396-Amin-27932
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: The launch file may have a syntax error, or its format is unknown
/

mzahana, 2023-09-19:
I don't use ros foxy, so I can't tell.
I am using ros humble.

mzahana, 2023-09-22:
This is the updated python launch files. The previous ones did not set the config and pluginlists params correctly.

```python
#!/usr/bin/env python3

import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch_ros.actions import Node
from ament_index_python import get_package_share_directory
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    ld = LaunchDescription()

    fcu_url = LaunchConfiguration('fcu_url')
    gcs_url = LaunchConfiguration('gcs_url')
    tgt_system = LaunchConfiguration('tgt_system')
    tgt_component = LaunchConfiguration('tgt_component')
    log_output = LaunchConfiguration('log_output')
    pluginlists_yaml = LaunchConfiguration('pluginlists_yaml')
    config_yaml = LaunchConfiguration('config_yaml')
    fcu_protocol = LaunchConfiguration('fcu_protocol')
    mavros_namespace = LaunchConfiguration('mavros_namespace')

    fcu_url_arg = DeclareLaunchArgument(
        'fcu_url',
        default_value='udp://:14540@127.0.0.1:14557',
        description="FCU interface Example /dev/ttyACM0:57600 OR udp://:14540@127.0.0.1:14557"
    )
    gcs_url_arg = DeclareLaunchArgument(
        'gcs_url',
        default_value='',
        description="GCS bridge interface"
    )
    tgt_system_arg = DeclareLaunchArgument(
        'tgt_system',
        default_value='1',
        description="MAVLink ID of the target system"
    )
    tgt_component_arg = DeclareLaunchArgument(
        'tgt_component',
        default_value='1',
        description="MAVLink component ID inside the target system"
    )
    log_output_arg = DeclareLaunchArgument(
        'log_output',
        default_value='screen',
        description="Type of log output"
    )
    package_name = 'mavros'
    file_name = 'launch/px4_pluginlists.yaml'
    package_share_directory = get_package_share_directory(package_name)
    file_path = os.path.join(package_share_directory, file_name)
    pluginlists_yaml_arg = DeclareLaunchArgument(
        'pluginlists_yaml',
        default_value=file_path,
        description="File path of the pluginlist_yaml file"
    )
    package_name = 'mavros'
    file_name = 'launch/px4_config.yaml'
    package_share_directory = get_package_share_directory(package_name)
    file_path = os.path.join(package_share_directory, file_name)
    config_yaml_arg = DeclareLaunchArgument(
        'config_yaml',
        default_value=file_path,
        description="File path of the mavros config_yaml file"
    )
    fcu_protocol_arg = DeclareLaunchArgument(
        'fcu_protocol',
        default_value='v2.0',
        description="MAVLink version"
    )
    namespace_arg = DeclareLaunchArgument(
        'mavros_namespace',
        default_value='mavros',
        description="Node namespace"
    )

    # MAVROS node
    mavros_node = Node(
        namespace=mavros_namespace,
        package='mavros',
        executable='mavros_node',
        parameters=[
                {'fcu_url': fcu_url},
                {'gcs_url': gcs_url},
                {'tgt_system': tgt_system},
                {'tgt_component': tgt_component},
                {'fcu_protocol': fcu_protocol},
                LaunchConfiguration('pluginlists_yaml'),
                LaunchConfiguration('config_yaml')
        ]
    )

    ld.add_action(fcu_url_arg)
    ld.add_action(gcs_url_arg)
    ld.add_action(tgt_component_arg)
    ld.add_action(tgt_system_arg)
    ld.add_action(log_output_arg)
    ld.add_action(fcu_protocol_arg)
    ld.add_action(namespace_arg)
    ld.add_action(pluginlists_yaml_arg)
    ld.add_action(config_yaml_arg)
    ld.add_action(mavros_node)

    return ld
```

vooon, 2023-09-24:
I actually unsure why python script is better, as in my opinion better to have some declarative dsl.
But you probably can make a PR, so i'll add that version too.

AjayShanker-geek, 2023-12-20:
> @vooon When I try the
> 
> `ros2 launch mavros px4.launch`
> 
> I get the following error: / /opt/ros/foxy/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html from pkg_resources import load_entry_point [INFO] [launch]: All log files can be found below /home/aminys/.ros/log/2023-09-19-12-02-40-071396-Amin-27932 [INFO] [launch]: Default logging verbosity is set to INFO Task exception was never retrieved future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('')> Traceback (most recent call last): File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event await self.__process_event(next_event) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event visit_all_entities_and_collect_futures(entity, self.__context)) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures sub_entities = entity.visit(context) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit return self.execute(context) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute launch_description = self.__launch_description_source.get_launch_description(context) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description self._get_launch_description(self.__expanded_location) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description return get_launch_description_from_any_launch_file(location) File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file raise InvalidLaunchFileError(extension, likely_errors=exceptions) launch.invalid_launch_file_error.InvalidLaunchFileError: The launch file may have a syntax error, or its format is unknown /

Any solution to this issue?

closed 2026-10-16 by github-actions[bot]