mavlink/mavros #1898: mavros - problem with the colcon build

naorwaiss, 2023-09-25:
hi , new at mavros 
try to colcon build and get this error 

```
naor@naor-laptop:~/mav_ros$ colcon build 
Starting >>> mavlink 
Starting >>> mavros_msgs
Finished <<< mavlink [0.18s]                                  
Starting >>> libmavconn
Finished <<< libmavconn [0.13s]                                      
Finished <<< mavros_msgs [1.76s]                     
Starting >>> mavros
Finished <<< mavros [0.40s]                    
Starting >>> mavros_extras
--- stderr: mavros_extras                             
/home/naor/mav_ros/src/mavros/mavros_extras/src/plugins/gps_status.cpp: In member function ‘void mavros::extra_plugins::GpsStatusPlugin::handle_gps2_raw(const mavlink_message_t*, mavlink::common::msg::GPS2_RAW&, mavros::plugin::filter::SystemAndOk)’:
/home/naor/mav_ros/src/mavros/mavros_extras/src/plugins/gps_status.cpp:138:37: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘alt_ellipsoid’
  138 |     ros_msg.alt_ellipsoid = mav_msg.alt_ellipsoid;
      |                                     ^~~~~~~~~~~~~
/home/naor/mav_ros/src/mavros/mavros_extras/src/plugins/gps_status.cpp:139:29: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘h_acc’
  139 |     ros_msg.h_acc = mav_msg.h_acc;
      |                             ^~~~~
/home/naor/mav_ros/src/mavros/mavros_extras/src/plugins/gps_status.cpp:140:29: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘v_acc’
  140 |     ros_msg.v_acc = mav_msg.v_acc;
      |                             ^~~~~
/home/naor/mav_ros/src/mavros/mavros_extras/src/plugins/gps_status.cpp:141:31: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘vel_acc’
  141 |     ros_msg.vel_acc = mav_msg.vel_acc;
      |                               ^~~~~~~
/home/naor/mav_ros/src/mavros/mavros_extras/src/plugins/gps_status.cpp:142:31: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘hdg_acc’
  142 |     ros_msg.hdg_acc = mav_msg.hdg_acc;
      |                               ^~~~~~~
make[2]: *** [CMakeFiles/mavros_extras_plugins.dir/build.make:232: CMakeFiles/mavros_extras_plugins.dir/src/plugins/gps_status.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:136: CMakeFiles/mavros_extras_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< mavros_extras [3.13s, exited with code 2]

Summary: 4 packages finished [5.42s]
  1 package failed: mavros_extras
  1 package had stderr output: mavros_extras
```

thanks for the help

vooon, 2023-09-26:
That's quite strange because you should have compatible version of mavros_msgs within a workspace. It's even listed in colcon log.

What version of ROS do do you use? Can you try to update mavros and mavlink clones, clean build dir and retry?

naorwaiss, 2023-09-27:
i am use ros2 with foxy 
the installation finish and the mavros_node is run

naorwaiss, 2023-09-29:
```
naor@naor-laptop:~/git_mav$ colcon build 
Starting >>> libmavconn
Starting >>> mavros_msgs
Finished <<< mavros_msgs [4.15s]                                   
Finished <<< libmavconn [5.99s]                     
Starting >>> mavros
[Processing: mavros]                             
--- stderr: mavros                               
/home/naor/git_mav/src/mavros/mavros/src/plugins/setpoint_accel.cpp:19:10: fatal error: tf2_eigen/tf2_eigen.hpp: No such file or directory
   19 | #include "tf2_eigen/tf2_eigen.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/mavros_plugins.dir/build.make:271: CMakeFiles/mavros_plugins.dir/src/plugins/setpoint_accel.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:189: CMakeFiles/mavros_plugins.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< mavros [31.5s, exited with code 2]

Summary: 2 packages finished [38.5s]
  1 package failed: mavros
  1 package had stderr output: mavros
  1 package not processed

```

now the problem change little bit 
can help ?

vooon, 2023-10-01:
That's compatibility problem with old releases. There was `.h` suffix, in more recent releases `.hpp`.
So you either can try newer release, or find and replace tf2 imports to use `.h`.

Since Foxy is EOLed, in my opinion better to upgrade.

duguguang, 2024-08-02:
> That's compatibility problem with old releases. There was `.h` suffix, in more recent releases `.hpp`. So you either can try newer release, or find and replace tf2 imports to use `.h`.
> 
> Since Foxy is EOLed, in my opinion better to upgrade.

`nokov@ubuntu:~/ros2_ws$ find ./src/mavros/ -type f -exec sed -i 's/tf2_eigen\.hpp/tf2_eigen\.h/g' {} +`

yes, you can replace all tf2 imports in the whole directory, It's useful!

DZ-yxz, 2024-09-10:
> That's compatibility problem with old releases. There was `.h` suffix, in more recent releases `.hpp`. So you either can try newer release, or find and replace tf2 imports to use `.h`.
> 
> Since Foxy is EOLed, in my opinion better to upgrade.

I ues mavros2.8.0 but meet this problem,how should i do to solve this problem

closed 2026-10-16 by github-actions[bot]