mavlink/mavros #1900: /mavros/rc/override Topic Not Working for PX4

mescar-nmt, 2023-10-02:
----

### Issue details
Hello, I am setting up a custom UAV with a Pixhawk 2.4.8 flight computer for an experiment. This experiment requires custom RC inputs. After investigating the best method of controlling the RC inputs, I decided to use the /mavros/rc/override topic. However, once I publish my command, nothing happens. The topic surely publishes on the /mavros/rc/override topic, but it does not translate to the /mavros/rc/in or /mavros/rc/out topics. Perhaps my setup is incorrect, or I have to change mavros source code to get the override topic to work... Any help is appreciated.

### MAVROS version and platform

Mavros: 1.17.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.13.3 

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 372 lines cut ...]
        key: "Frequency (Hz)"
        value: "10.071442"
      - 
        key: "Last RTT (ms)"
        value: "124.737890"
      - 
        key: "Mean RTT (ms)"
        value: "248.875189"
      - 
        key: "Last remote time (s)"
        value: "7836.285960000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"

```

### Check ID

```
Received 445 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 65, 1, 36, 4, 230, 8, 74, 141, 77, 111, 83, 245, 22, 281, 411, 285, 30

```

lei-zihao, 2024-06-04:
你好，请问这是什么原因呢，我又有遇到这种情况。

3AMengineer, 2026-09-20:
find parameter MAV_SYSID_GCS