mavlink/mavros #1902: basic ros2 topic run

naorwaiss, 2023-10-03:
try to run basic topic with ros2 

```
#!/usr/bin/env python

import rclpy
from rclpy.node import Node
from sensor_msgs.msg import NavSatFix

class DronePositionMonitor(Node):
    def __init__(self):
        super().__init__('drone_position_monitor')
        self.current_position = None

        self.pos_sub = self.create_subscription(NavSatFix, '/mavros/global_position/global', self.position_callback, 10)

    def position_callback(self, data):
        self.current_position = data
        self.get_logger().info("Drone Position (Lat, Long, Alt): %.7f, %.7f, %.2f", data.latitude, data.longitude, data.altitude)

def main(args=None):
    rclpy.init(args=args)
    drone_monitor = DronePositionMonitor()
    rclpy.spin(drone_monitor)
    rclpy.shutdown()

if __name__ == '__main__':
    main()
    ```
    
    try to get the position of the drone and on the way learn how to use nodes
    
    this is the problem:
    ```
    naor@naor-laptop:~$ ros2 run drone_position_monitor position_monitor 
/opt/ros/foxy/bin/ros2:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
```

thanks for the help

vooon, 2023-10-05:
First you should configure same QoS as mavros uses. Then try doing the same with ros2 tool.

naorwaiss, 2023-10-06:
sory for  the annoying i am fresh here 
how can i configure the Qos?
have some tutorials?

vooon, 2023-10-07:
https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html

closed 2026-10-16 by github-actions[bot]