mavlink/mavros #1903: mavros apm not publishing topics

JayanthAmmapalli, 2023-10-04:
### Issue details
I am new to pichawk and mavros please guide me though this error of not getting messges in mavros topics.
The FCU is pixhawk cube orange with latest firmware i.e, chibi OS 4.4.1. The FCU is connected to onbord computer with micro USB. After starting mavros with roslaunch mavros apm.launch fcu_url:=/dev/ttyACM1:115200, the mavros is getting stuck at line [ INFO] [1696379422.304927156]: MAVROS started. MY ID 1.240, TARGET ID 1.1. During this time i can see the rostopic list and all the topics in it. After sometime getting error : 
[ERROR] [1696379963.787417128]: 0: DROPPED Message TIMESYNC: MAVConnSerial::send_message: TX queue overflow

But when echoing they are all empty except for /mavros/state. I can echo all the topics when running mavros with telemetry antennas. 

### MAVROS version and platform

Mavros installed with sudo apt install ros-noetic-mavros
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

[ *] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_mav_type: ONBOARD_CONTROLLER
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_id: 4
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/horiz_accuracy: 0.5
 * /mavros/fake_gps/mocap_transform: False
 * /mavros/fake_gps/mocap_withcovariance: False
 * /mavros/fake_gps/satellites_visible: 6
 * /mavros/fake_gps/speed_accuracy: 0.0
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_hil_gps: True
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fake_gps/vert_accuracy: 0.5
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: /dev/ttyACM1:115200
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
[... 405 lines cut ...]
        value: "0"
      - 
        key: "Frequency (Hz)"
        value: "0.000000"
      - 
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.000000"
      - 
        key: "Last remote time (s)"
        value: "0.000000000"
      - 
        key: "Estimated time offset (s)"
        value: "0.000000000"
---

### Check ID

ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

vooon, 2023-10-05:
Check your connection url, as with current one nothing came at all, see "Rx total bytes: 0".

mbr4477, 2024-01-10:
Might vary by AP configuration, but make sure you have told the AP to start sending messages. For example, to get global position data @ 1 Hz you might need call the `/mavros/set_message_interval` service with `{ message_id: 33, message_rate: 1.0 }`. Then try echoing (for example) `/mavros/global_position/rel_alt` to see relative altitude.

JayanthAmmapalli, 2024-02-02:
I am facing a similar issue of not seeing any data published in ros topics after starting mavros. I was able to solve this problem by establishing a connecting with GCS software like Mission Planner or Qgroundcontrol first and then switching to mavros will start publishing data in all these topics.

Can @mbr4477 please elaborate on how to manually start sending messages for the topics in mavros.

mbr4477, 2024-02-04:
*Disclaimer: I've only been using the ArduCopter simulator, so not sure if the PX4 has any big differences.*

@JayanthAmmapalli MAVROS uses MAVLink to talk to the AP. The MAVROS documentation isn't super clear, but that means plugins depend on certain messages being sent from the AP over MAVLink. 

I'll use global position as an example, but this all applies for other topics/plugins as well. For the global position plugin, we can see [here](https://github.com/mavlink/mavros/blob/2e20c1cbe62fe2bdec51ae6f63bdd740dee39cb7/mavros/src/plugins/global_position.cpp#L147) that it has what appear to be handlers for GPS_RAW_INT, GLOBAL_POSITION_INT, GPS_GLOBAL_ORIGIN, and LOCAL_POSITION_NED_SYSTEM_GLOBAL_OFFSET ***MAVLink*** messages. These MAVLink messages have definitions [here](https://mavlink.io/en/messages/common.html). 

Looking further down the `global_position.cpp` plugin code, we see that the [GLOBAL_POSITION_INT handler](https://github.com/mavlink/mavros/blob/2e20c1cbe62fe2bdec51ae6f63bdd740dee39cb7/mavros/src/plugins/global_position.cpp#L312) unpacks the MAVLink data and [publishes to various topics](https://github.com/mavlink/mavros/blob/2e20c1cbe62fe2bdec51ae6f63bdd740dee39cb7/mavros/src/plugins/global_position.cpp#L434). The trick then is to get the AP to send back the GLOBAL_POSITION_INT message to kick all this off on the MAVROS side. My guess is that it works when you connect the GCS because that GCS sends messages to kick off telemetry. Without it, you need to do it yourself. 

To start the GLOBAL_POSITION_INT stream, we need to send [MAV_CMD_SET_MESSAGE_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_MESSAGE_INTERVAL) via MAVROS. From the MAVLink docs, we can see that the GLOBAL_POSITION_INT message [has a message id of 33](https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT). Thus, to send this with a rate of 40 Hz, we can do this:

```python
from mavros.srv import MessageInterval
from rclpy.node import Node

def main():
    rclpy.init()
    node = Node("mavros_client")
    logger = node.get_logger()

    client = node.create_client(
        MessageInterval, 
        "mavros/set_message_interval"
    )
    
    logger.info("Waiting for MessageInterval service . . .")
    while not client.service_is_ready():
        rclpy.spin_once(node)
    logger.info("MessageInterval service is ready")
    
    req = MessageInterval.Request()
    req.message_id = 33
    req.message_rate = 40.0

    logger.info(
        "Calling MessageInterval with" 
        + f" msg id = {req.message_id}, rate = {req.message_rate}"
    )
    future = client.call_async(req)
    while not future.done():
        rclpy.spin_once(node)
    res = future.result()
    logger.info("Call completed with success = {res.success}")
    
    node.destroy_node()
    rclpy.shutdown()

 if __name__ == "__main__":
    main()
```

This is a short-lived node that just spins up, calls the service, and shuts down. More likely you'd want to include these calls in the initialization steps of your existing MAVROS client node. Untested, but hopefully it's enough to point you in the right direction.

This is just for global position data. To get things like IMU data, you'll have to call the MessageInterval service more times to start streaming the other required MAVLink message(s), e.g., MAVLink message id 27 for raw imu data and message id 31 for the attitude quaternion.

closed 2026-10-16 by github-actions[bot]