mavlink/mavros #1905: ROS2 RTK setup - replacement for gcs_bridge

diablo1281, 2023-10-09:
### Issue details
Hi, I am trying to connect RTK node to mavros node.
I have a device on the network that sends Mavlink messages with RTCM for all UAVs in the network (it is seen as a vehicle with its own ID).
On ROS1 I used `gcs_bridge` and it worked fine. Now as I updated to ROS2 I can't get it to work. I tried to modify `mavros_node` to accept additional GCS in the router setup, but messages are not visible in the Mavlink Inspector on QGCS and the drone doesn't receive any RTCM messages.

I would appreciate any help on this topic.

### MAVROS version and platform

Mavros: ros2 branch
ROS: Humble (compiled from source)
Ubuntu: Debian Bullseye RT

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

Version: 1.13.1

vooon, 2023-10-09:
You don't need `gcs_bridge` anymore. Just call router to add additional link: https://github.com/mavlink/mavros/blob/ros2/mavros_msgs/srv/EndpointAdd.srv

Note that router doesn't pass messages from the same endpoint type. So message coming from GCS will be routed to FCU, but not to other GCS (i.e. QGC).

diablo1281, 2023-11-10:
@vooon thanks for anwers.

I still didn't get it to work. After trying to create an endpoint for the RTK station there are no incoming messages to MAVROS from it.
I did try it through C++ module code and service. Also, I disabled the block for exchanging messages between the same type of endpoint to find out it using QGC. As a result, in Mavlink Inspector I can see UAV and GCS messages, but no RTK.
Also, I did try to connect 2 QGC with different IDs and there is some strange behavior - sometimes it connects, sometimes not. Messages are getting laggy also.

I think I tried everything but without luck. Do you have any tips?

vooon, 2023-11-10:
That's odd. Probably you can try to put mavlink-router in front of mavros router as a workaround.

XXLiu-HNU, 2024-09-19:
have you solves this problem？ We meet the same error when use ros
2 and RTK

diablo1281, 2024-09-19:
Unfortunately not

XXLiu-HNU, 2024-09-20:
> Unfortunately not, I am out of ideas. And not comfortable using another software as a workaround.

Well, at first we didn't use WiFi and mavros to connect, but used additional hardware (radio，CUAV P9 Radio )for communication, after our test is OK!

But when use mavros2 and qgc, there is something wrong

closed 2026-10-15 by github-actions[bot]