mavlink/mavros #1907: base_link and base_link_frd are hardcoded - Better to define them using parameters

mzahana, 2023-10-11:
### Issue details
Can we make the `base_link`  frame_id a parameter instead of having it fixed in [this static transform](https://github.com/mavlink/mavros/blob/e0be266e675db5b3d615404d910c164430d46da9/mavros/src/lib/mavros_uas.cpp#L168) ?

The `odom.cpp` plugin looks up a transform with respect to the`base_link_frd` [here](https://github.com/mavlink/mavros/blob/ros2/mavros_extras/src/plugins/odom.cpp#L148)

The issues is, when doing multi vehicle simulation, all the vehicles will have the same `base_link` and `base_link_frd` with the current implementation. However, they should be unique for each vehicle to have a proper TF tree. That is why I propose to make the `base_link` a parameter. Then, the FRD frame can be defined by adding `_frd` post-fix, so it becomes `base_link_frame_id+"_frd"`.

I don't know where to define this parameter so I can propose a PR.

### MAVROS version and platform

Mavros: 2.6.0
ROS: humble
Ubuntu: 22.04

### Autopilot type and version

[  ] ArduPilot
[ x ] PX4

Version: 1.14

mzahana, 2023-10-12:
@vooon here is my PR https://github.com/mavlink/mavros/pull/1909 to solve issue.

vooon, 2023-10-16:
Should be fixed now.

closed 2023-10-16 by vooon