mavlink/mavros #1911: SensorQoS necessary for gp_origin?

haavarpb, 2023-10-18:
### Issue details

Hi,

I'm currently working on a ros package and here I subscribe to `gp_origin` from the mavros global position plugin. At least in my setup using Ardupilot SITL this is published once when `GPS_GLOBAL_ORIGIN` is emitted. This means my nodes requiring this information must be subscribed before this event happens because of the Sensor Quality of Service. Is this really necessary? In my mind it would be nice to allow late-joining of subscribers using another durability setting i.e. transient_local. I am not an expert at ROS, so I'm not sure of all the implications, but I would like to ask anyways. 

### MAVROS version and platform

Mavros: 2.6.0
ROS: Humble
Ubuntu: 22.04

### Autopilot type and version

[x] ArduPilot
[  ] PX4

Version: 4.4.1

vooon, 2023-10-19:
Well, i'm unsure what's best QoS here. Let's try that PR.

haavarpb, 2023-10-20:
Sorry for linking to the PR without more context, I was a bit short on time. The PR updates the guided target subscription side. I believe the author @kribe48 can best explain why, but it's a necessary update for the subscription callback to fire (by having the same QoS as the publisher). By linking the PR I hoped he maybe could chime in on the discussion. 

I'm raising the question of possibly reviewing the QoS publisher side on the `gp_origin` publisher, and perhaps on others aswell.

I can atleast list one pro with having a durability setting since it allows late-joining of subscribers that need this information. The situation right now requires the subscriber to be ready before `gp_origin` is published or perhaps requiring to use MavLink messages to directly fetch it by itself.

kribe48, 2023-10-26:
I updated the subscriber QoS since it did not subscribe to data with the current setting. I am not sure about what QoS to use, but they need to be the same on both sides :) Your suggestion sounds reasonable to me!

vooon, 2025-09-14:
Fixed by #2065

closed 2025-09-14 by vooon