mavlink/mavros #1917: Some plugins still load even if blacklisted

mzahana, 2023-10-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I want to disable a set of plugins that I don't need. So, here is what I do.
* I list all the plugins(nodes) that I don't need in the `px4_pluginlists.yaml`. Then I run mavros launch file.
* Many of these plugins don't run as expected. However, there are still some plugins that are in the `plugin_denylist` which seems to run (I can see their topics using `ros2 topic list`).
* Here is my `px4_pluginlists.yaml`

```yaml
/mavros/**:
  ros__parameters:
    plugin_denylist:
      # common

      # extras
      - image_pub
      - vibration
      - distance_sensor
      - rangefinder
      - wheel_odometry
      - mount_control
      - log_transfer
      - landing_target
      - gps_rtk
      - manual_control
      - hil
      - adsb
      - optical_flow
      - terrain
      - ftp
      - tdr_radio
      - rallypoint
      - obstacle
      - cellular_status
      - actuator_control
      - fake_gps
      - esc_telemetry
      - mag_calibration
      - px4flow
      - esc_status
      - gps_input
      - debug_value
      - geofence
      - nav_controller_output
      - wind
      - play_tune
      - camera
      - guided_target
      - mocap

    # plugin_allowlist:
    #   - 'sys_*'
```
* When I execute `ros2 node list` I get,
```bash
/mavros/altitude
/mavros/cam_imu_sync
/mavros/cmd
/mavros/companion_process
/mavros/global_position
/mavros/gpsstatus
/mavros/home_position
/mavros/imu
/mavros/local_position
/mavros/mag_calibration
/mavros/mavros
/mavros/mavros_node
/mavros/mavros_router
/mavros/mission
/mavros/mocap
/mavros/obstacle
/mavros/odometry
/mavros/onboard_computer
/mavros/param
/mavros/rc
/mavros/setpoint_accel
/mavros/setpoint_attitude
/mavros/setpoint_position
/mavros/setpoint_raw
/mavros/setpoint_trajectory
/mavros/setpoint_velocity
/mavros/sys
/mavros/time
/mavros/trajectory
/mavros/transform_listener_impl_56316dba66d0
/mavros/tunnel
/mavros/vfr_hud
/mavros/vision_pose
/mavros/vision_speed
/mavros/wheel_odometry
/mavros/wind
```
* So you can see that blacklisted `/mavros/wheel_odometry` and `/interceptor/mavros/mag_calibration`, but they still run!

@vooon Is this a bug, or I am missing something?

Thanks in advance.

### MAVROS version and platform

Mavros: 2.6.0
ROS: humble
Ubuntu: 22.0

### Autopilot type and version

[  ] ArduPilot
[ x] PX4

Version: 1.14.0

vooon, 2023-10-26:
Could you please check what lists node really have? I mean run the node and then get allow/deny lists parameters.

mzahana, 2023-10-26:
@vooon here is the `plugin_denylist` after I run the node
```bash
String values are: ['image_pub', 'vibration', 'distance_sensor', 'rangefinder', 'wheel_odometry', 'mount_control', 'log_transfer', 'landing_target', 'gps_rtk', 'manual_control', 'hil', 'adsb', 'optical_flow', 'terrain', 'ftp', 'tdr_radio', 'rallypoint', 'obstacle', 'cellular_status', 'actuator_control', 'fake_gps', 'esc_telemetry', 'mag_calibration', 'px4flow', 'esc_status', 'gps_input', 'debug_value', 'geofence', 'nav_controller_output', 'wind', 'play_tune', 'camera', 'guided_target', 'mocap']
```

You can see that `wheel_odometry` is there in the list. However, I can see the node when I execute `ros2 node list`

vooon, 2023-10-27:
Ok, then need to run the node with debug logging enabled. Then need to copy log lines here.

gregoryw3, 2023-11-10:
I'm having a simular issue, ROS2 Foxy

```yaml
/mavros/**:
  ros__parameters:
    plugin_denylist:
      - actuator_control
      - adsb
      # altitude
      - cam_imu_sync
      - camera
      - cellular_status
      # command
      # companion_process_status
      # common
      - debug_value
      # distance_sensor
      # esc_status
      # esc_telemetry
      # extras
      - fake_gps
      # ftp
      - geofence
      - global_position
      - gps_input
      - gps_rtk
      - gps_status
      - guided_target
      - hil
      - home_position
      - image_pub
      # imu
      - landing_target
      - local_position
      # log_transfer
      # mag_calibration_status
      # manual_control
      - mocap_pose_estimate
      - mount_control
      - nav_controller_output
      - obstacle
      - obstacle_distance
      - optical_flow
      # odometry
      # onboard_computer_status
      # param
      - play_tune
      - px4flow
      - rallypoint
      # rangefinder
      # rc_io
      - setpoint_accel
      - setpoint_attitude
      - setpoint_position
      - setpoint_raw
      - setpoint_trajectory
      - setpoint_velocity
      # sys_status
      # sys_time
      - tdr_radio
      - terrain
      - trajectory
      - tunnel
      - vfr_hud
      # vibration
      - vision_pose
      - vision_speed
      - vision_speed_estimate
      - waypoint
      - wheel_odometry
[... 131 lines cut ...]
[mavros_node-1] [INFO] [1699578809.218730814] [mavros]: Plugin wind_estimation created
[mavros_node-1] [INFO] [1699578809.219096229] [mavros]: Plugin wind_estimation initialized
```

```python
mavros_node = Node(
            package='mavros',
            executable='mavros_node',
            parameters=[{
                'pluginlists_yaml': "./apm_pluginlists.yaml",
                'config_yaml': './apm_config.yaml',
                'fcu_url': fcu_url_arg_str,
                'gcs_url': '',
                'tgt_system': 1,
                'tgt_component': 1,
                'fcu_protocol': "v2.0",
                'respawn_mavros': "false",
                'namespace': "mavros",
            }],
            emulate_tty=True,
            output='screen',
        )
```

What do you mean by debug logging enabled? Is it what I sent or is there a documentation link you can send?

vooon, 2023-11-10:
Can you please add a debug print here: https://github.com/mavlink/mavros/blob/ros2/mavros/src/lib/mavros_uas.cpp#L238

My suspicion that on loading time lists are empty.

gregoryw3, 2023-11-10:
Here is what I added, let me know if this isn't what you meant:
```cpp
bool UAS::is_plugin_allowed(
  const std::string & pl_name)
{
  RCLCPP_INFO(get_logger(), "is_plugin_allowed pl_name input: %s", pl_name.c_str());
  for (auto & bl_pattern : plugin_denylist) {
    if (pattern_match(bl_pattern, pl_name)) {
      for (auto & wl_pattern : plugin_allowlist) {
        if (pattern_match(wl_pattern, pl_name)) {
          return true;
        }
      }

      return false;
    }
  }

  return true;
}
```

```python
[mavros_node-1] [INFO] [1699633077.099793636] [mavros]: UAS Executor started, threads: 4
[mavros_node-1] [INFO] [1699633077.100350383] [mavros]: is_plugin_allowed pl_name input: actuator_control
[mavros_node-1] [INFO] [1699633077.128672502] [mavros]: Plugin actuator_control created
[mavros_node-1] [INFO] [1699633077.129517997] [mavros]: Plugin actuator_control initialized
[mavros_node-1] [INFO] [1699633077.129542163] [mavros]: is_plugin_allowed pl_name input: altitude
[mavros_node-1] [INFO] [1699633077.132688019] [mavros]: Plugin altitude created
[mavros_node-1] [INFO] [1699633077.132857685] [mavros]: Plugin altitude initialized
[mavros_node-1] [INFO] [1699633077.132872518] [mavros]: is_plugin_allowed pl_name input: command
[mavros_node-1] [INFO] [1699633077.138693399] [mavros]: Plugin command created
[mavros_node-1] [INFO] [1699633077.138859106] [mavros]: Plugin command initialized
[mavros_node-1] [INFO] [1699633077.138870981] [mavros]: is_plugin_allowed pl_name input: ftp
[mavros_node-1] [INFO] [1699633077.145700023] [mavros]: Plugin ftp created
[mavros_node-1] [INFO] [1699633077.145858772] [mavros]: Plugin ftp initialized
[mavros_node-1] [INFO] [1699633077.145870189] [mavros]: is_plugin_allowed pl_name input: geofence
[mavros_node-1] [INFO] [1699633077.151748944] [mavros]: Plugin geofence created
[mavros_node-1] [INFO] [1699633077.152146859] [mavros]: Plugin geofence initialized
[mavros_node-1] [INFO] [1699633077.152162859] [mavros]: is_plugin_allowed pl_name input: global_position
[mavros_node-1] [INFO] [1699633077.162381921] [mavros]: Plugin global_position created
[mavros_node-1] [INFO] [1699633077.162747419] [mavros]: Plugin global_position initialized
[mavros_node-1] [INFO] [1699633077.162761461] [mavros]: is_plugin_allowed pl_name input: home_position
[mavros_node-1] [INFO] [1699633077.166757020] [mavros]: Plugin home_position created
[mavros_node-1] [INFO] [1699633077.166891477] [mavros]: Plugin home_position initialized
[mavros_node-1] [INFO] [1699633077.166901977] [mavros]: is_plugin_allowed pl_name input: imu
[mavros_node-1] [INFO] [1699633077.171515616] [mavros]: Plugin imu created
[mavros_node-1] [INFO] [1699633077.171950405] [mavros]: Plugin imu initialized
[mavros_node-1] [INFO] [1699633077.171966946] [mavros]: is_plugin_allowed pl_name input: local_position
[mavros_node-1] [INFO] [1699633077.176939333] [mavros]: Plugin local_position created
[mavros_node-1] [INFO] [1699633077.177115707] [mavros]: Plugin local_position initialized
[mavros_node-1] [INFO] [1699633077.177126748] [mavros]: is_plugin_allowed pl_name input: manual_control
[mavros_node-1] [INFO] [1699633077.181840303] [mavros]: Plugin manual_control created
[mavros_node-1] [INFO] [1699633077.182095551] [mavros]: Plugin manual_control initialized
[mavros_node-1] [INFO] [1699633077.182109801] [mavros]: is_plugin_allowed pl_name input: nav_controller_output
[mavros_node-1] [INFO] [1699633077.185786862] [mavros]: Plugin nav_controller_output created
[mavros_node-1] [INFO] [1699633077.185934403] [mavros]: Plugin nav_controller_output initialized
[mavros_node-1] [INFO] [1699633077.185945611] [mavros]: is_plugin_allowed pl_name input: param
[mavros_node-1] [INFO] [1699633077.191155621] [mavros]: Plugin param created
[mavros_node-1] [INFO] [1699633077.191300704] [mavros]: Plugin param initialized
[mavros_node-1] [INFO] [1699633077.191342453] [mavros]: is_plugin_allowed pl_name input: rallypoint
[mavros_node-1] [INFO] [1699633077.195875926] [mavros]: Plugin rallypoint created
[mavros_node-1] [INFO] [1699633077.196036341] [mavros]: Plugin rallypoint initialized
[mavros_node-1] [INFO] [1699633077.196047550] [mavros]: is_plugin_allowed pl_name input: rc_io
[mavros_node-1] [INFO] [1699633077.201640057] [mavros]: Plugin rc_io created
[mavros_node-1] [INFO] [1699633077.201861223] [mavros]: Plugin rc_io initialized
[mavros_node-1] [INFO] [1699633077.201874222] [mavros]: is_plugin_allowed pl_name input: setpoint_accel
[mavros_node-1] [INFO] [1699633077.205894156] [mavros]: Plugin setpoint_accel created
[mavros_node-1] [INFO] [1699633077.205940989] [mavros]: Plugin setpoint_accel initialized
[mavros_node-1] [INFO] [1699633077.205948281] [mavros]: is_plugin_allowed pl_name input: setpoint_attitude
[... 3 lines cut ...]
[mavros_node-1] [INFO] [1699633077.218609870] [mavros]: Plugin setpoint_position created
[mavros_node-1] [INFO] [1699633077.218663578] [mavros]: Plugin setpoint_position initialized
[mavros_node-1] [INFO] [1699633077.218672995] [mavros]: is_plugin_allowed pl_name input: setpoint_raw
[mavros_node-1] [INFO] [1699633077.225499453] [mavros]: Plugin setpoint_raw created
[mavros_node-1] [INFO] [1699633077.225721160] [mavros]: Plugin setpoint_raw initialized
[mavros_node-1] [INFO] [1699633077.225733494] [mavros]: is_plugin_allowed pl_name input: setpoint_trajectory
[mavros_node-1] [INFO] [1699633077.233082824] [mavros]: Plugin setpoint_trajectory created
[... cut ...]
[mavros_node-1] [INFO] [1699633077.260626198] [mavros]: is_plugin_allowed pl_name input: waypoint
[mavros_node-1] [INFO] [1699633077.271618173] [mavros]: Plugin waypoint created
[mavros_node-1] [INFO] [1699633077.272021045] [mavros]: Plugin waypoint initialized
[mavros_node-1] [INFO] [1699633077.272044712] [mavros]: is_plugin_allowed pl_name input: wind_estimation
[mavros_node-1] [INFO] [1699633077.278958794] [mavros]: Plugin wind_estimation created
[mavros_node-1] [INFO] [1699633077.279173376] [mavros]: Plugin wind_estimation initialized
```

vooon, 2023-11-10:
@gregoryw3 it's almost what we need, just you've checked what's came in to a function, while we need to see a result.

I think you need to add similar print to `pattern_match()`.

gregoryw3, 2023-11-10:
So the issue is that plugin_denylist is empty:

```python
[mavros_node-1] [INFO] [1699638403.604880084] [mavros_node]: Starting mavros uas node
[mavros_node-1] [INFO] [1699638403.685164378] [mavros]: UAS Executor started, threads: 4
[mavros_node-1] [INFO] [1699638403.685348918] [mavros]: is_plugin_allowed pl_name input: actuator_control
[mavros_node-1] [INFO] [1699638403.685371918] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.723797121] [mavros]: Plugin actuator_control created
[mavros_node-1] [INFO] [1699638403.724146035] [mavros]: Plugin actuator_control initialized
[mavros_node-1] [INFO] [1699638403.724171701] [mavros]: is_plugin_allowed pl_name input: altitude
[mavros_node-1] [INFO] [1699638403.724185410] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.728227582] [mavros]: Plugin altitude created
[mavros_node-1] [INFO] [1699638403.728395706] [mavros]: Plugin altitude initialized
[mavros_node-1] [INFO] [1699638403.728408289] [mavros]: is_plugin_allowed pl_name input: command
[mavros_node-1] [INFO] [1699638403.728412414] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.737765540] [mavros]: Plugin command created
[mavros_node-1] [INFO] [1699638403.737920705] [mavros]: Plugin command initialized
[mavros_node-1] [INFO] [1699638403.737933455] [mavros]: is_plugin_allowed pl_name input: ftp
[mavros_node-1] [INFO] [1699638403.737937830] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.748283948] [mavros]: Plugin ftp created
[mavros_node-1] [INFO] [1699638403.748487071] [mavros]: Plugin ftp initialized
[mavros_node-1] [INFO] [1699638403.748499862] [mavros]: is_plugin_allowed pl_name input: geofence
[mavros_node-1] [INFO] [1699638403.748507154] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.756669791] [mavros]: Plugin geofence created
[mavros_node-1] [INFO] [1699638403.757438242] [mavros]: Plugin geofence initialized
[mavros_node-1] [INFO] [1699638403.757461242] [mavros]: is_plugin_allowed pl_name input: global_position
[mavros_node-1] [INFO] [1699638403.757469783] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.770632709] [mavros]: Plugin global_position created
[mavros_node-1] [INFO] [1699638403.770952915] [mavros]: Plugin global_position initialized
[mavros_node-1] [INFO] [1699638403.770969831] [mavros]: is_plugin_allowed pl_name input: home_position
[mavros_node-1] [INFO] [1699638403.770976415] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.776322368] [mavros]: Plugin home_position created
[mavros_node-1] [INFO] [1699638403.776488325] [mavros]: Plugin home_position initialized
[mavros_node-1] [INFO] [1699638403.776499283] [mavros]: is_plugin_allowed pl_name input: imu
[mavros_node-1] [INFO] [1699638403.776505783] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.782854477] [mavros]: Plugin imu created
[mavros_node-1] [INFO] [1699638403.783569971] [mavros]: Plugin imu initialized
[mavros_node-1] [INFO] [1699638403.783595012] [mavros]: is_plugin_allowed pl_name input: local_position
[mavros_node-1] [INFO] [1699638403.783656512] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.795815905] [mavros]: Plugin local_position created
[mavros_node-1] [INFO] [1699638403.796174068] [mavros]: Plugin local_position initialized
[mavros_node-1] [INFO] [1699638403.796191860] [mavros]: is_plugin_allowed pl_name input: manual_control
[mavros_node-1] [INFO] [1699638403.796224068] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.801805019] [mavros]: Plugin manual_control created
[mavros_node-1] [INFO] [1699638403.802128099] [mavros]: Plugin manual_control initialized
[mavros_node-1] [INFO] [1699638403.802149682] [mavros]: is_plugin_allowed pl_name input: nav_controller_output
[mavros_node-1] [INFO] [1699638403.802158182] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.806950015] [mavros]: Plugin nav_controller_output created
[mavros_node-1] [INFO] [1699638403.807151930] [mavros]: Plugin nav_controller_output initialized
[mavros_node-1] [INFO] [1699638403.807161763] [mavros]: is_plugin_allowed pl_name input: param
[mavros_node-1] [INFO] [1699638403.807165888] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.813432750] [mavros]: Plugin param created
[mavros_node-1] [INFO] [1699638403.813597706] [mavros]: Plugin param initialized
[mavros_node-1] [INFO] [1699638403.813609790] [mavros]: is_plugin_allowed pl_name input: rallypoint
[mavros_node-1] [INFO] [1699638403.813613915] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.819724986] [mavros]: Plugin rallypoint created
[mavros_node-1] [INFO] [1699638403.819894984] [mavros]: Plugin rallypoint initialized
[mavros_node-1] [INFO] [1699638403.819904193] [mavros]: is_plugin_allowed pl_name input: rc_io
[mavros_node-1] [INFO] [1699638403.819908318] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.825606643] [mavros]: Plugin rc_io created
[mavros_node-1] [INFO] [1699638403.825827391] [mavros]: Plugin rc_io initialized
[mavros_node-1] [INFO] [1699638403.825839849] [mavros]: is_plugin_allowed pl_name input: setpoint_accel
[mavros_node-1] [INFO] [1699638403.825846015] [mavros]: Size of plugin_denylist: 0
[mavros_node-1] [INFO] [1699638403.831866296] [mavros]: Plugin setpoint_accel created
[... cut ...]
    if (pattern_match(bl_pattern, pl_name)) {
      RCLCPP_INFO(get_logger(), "pattern_match bl_pattern: true");
      for (auto & wl_pattern : plugin_allowlist) {
        if (pattern_match(wl_pattern, pl_name)) {
          RCLCPP_INFO(get_logger(), "pattern_match wl_pattern: true");
          return true;
        }
        RCLCPP_INFO(get_logger(), "pattern_match wl_pattern: false");
      }

      return false;
    }
    RCLCPP_INFO(get_logger(), "pattern_match bl_pattern: false");
  }

  return true;
}
```

vooon, 2023-11-13:
It's done async to constructor:

https://github.com/mavlink/mavros/blob/ros2/mavros/src/lib/mavros_uas.cpp#L101

vooon, 2023-11-13:
Please try to change params.yaml that way:
```
/mavros:
  ros__parameters:
    plugin_denylist:
      - '*'
```

gregoryw3, 2023-11-13:
```yaml
/mavros/**:
  ros__parameters:
    plugin_blacklist:
      - '*'
    # plugin_allowlist:
    #   - 'sys_*'
```
and

```yaml
/mavros:
  ros__parameters:
    plugin_denylist:
      - '*'
    # plugin_allowlist:
    #   - 'sys_*'
```
Did not change anything, it had the same output as https://github.com/mavlink/mavros/issues/1917#issuecomment-1806165243

vooon, 2023-11-14:
@gregoryw3 could you please change 10ms to 1000ms at startup timer? Suspicion that it runs earlier than parameter server does something.

mgiurato, 2023-11-17:
This problem should have been fixed by this commit:
https://github.com/mavlink/mavros/commit/97cf3c606197d2e5ee83d1b8896e135db6bd649a

Which commit are you working on?

mzahana, 2023-11-18:
@mgiurato would that commit affect the parameters loading in multi-uav simulation,  where there are multiple uavs with different namespaces?

gregoryw3, 2023-11-19:
> This problem should have been fixed by this commit: [97cf3c6](https://github.com/mavlink/mavros/commit/97cf3c606197d2e5ee83d1b8896e135db6bd649a)
> 
> Which commit are you working on?

I have no idea anymore. I can't seem to build anymore and the folder got corrupted. After git cloneing the latest commit, I get this error when building:

```shell
Starting >>> libmavconn
Starting >>> mavros_msgs
Finished <<< libmavconn [0.98s]                                                             
--- stderr: mavros_msgs                             
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/rosidl_typesupport_fastrtps_cpp/rosidl_typesupport_fastrtps_cpp", line 23, in <module>
    sys.exit(main())
  File "/opt/ros/foxy/lib/rosidl_typesupport_fastrtps_cpp/rosidl_typesupport_fastrtps_cpp", line 19, in main
    return generate_cpp(args.generator_arguments_file)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_typesupport_fastrtps_cpp/__init__.py", line 24, in generate_cpp
    generate_files(generator_arguments_file, mapping)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 55, in generate_files
    args = read_generator_arguments(generator_arguments_file)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 39, in read_generator_arguments
    return json.load(h)
  File "/usr/lib/python3.8/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 208 column 1 (char 18507)
make[2]: *** [CMakeFiles/mavros_msgs__rosidl_typesupport_fastrtps_cpp.dir/build.make:301: rosidl_typesupport_fastrtps_cpp/mavros_msgs/msg/detail/dds_fastrtps/adsb_vehicle__type_support.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:412: CMakeFiles/mavros_msgs__rosidl_typesupport_fastrtps_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/rosidl_typesupport_c/rosidl_typesupport_c", line 28, in <module>
    sys.exit(main())
  File "/opt/ros/foxy/lib/rosidl_typesupport_c/rosidl_typesupport_c", line 24, in main
    return generate_c(args.generator_arguments_file, args.typesupports)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_typesupport_c/__init__.py", line 28, in generate_c
    generate_files(
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 55, in generate_files
    args = read_generator_arguments(generator_arguments_file)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_cmake/__init__.py", line 39, in read_generator_arguments
    return json.load(h)
  File "/usr/lib/python3.8/json/__init__.py", line 293, in load
    return loads(fp.read(),
  File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Invalid control character at: line 94 column 3 (char 9511)
make[2]: *** [CMakeFiles/mavros_msgs__rosidl_typesupport_c.dir/build.make:299: rosidl_typesupport_c/mavros_msgs/msg/adsb_vehicle__type_support.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:439: CMakeFiles/mavros_msgs__rosidl_typesupport_c.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< mavros_msgs [2.71s, exited with code 2]

Summary: 1 package finished [3.93s]
  1 package failed: mavros_msgs
  1 package had stderr output: mavros_msgs
  2 packages not processed
```

I belive I was on 2.4.0-1 ROS Foxy release. It was selected as a tag, however I don't see the tag on github anymore.

21coolkarni, 2024-01-15:
did anyone find solution for this problem?

limshoonkit, 2024-03-13:
i can see the plugins being ignored

px4.launch.py
```
import os

from ament_index_python import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    px4_config_path = os.path.join(
        get_package_share_directory('mavros'),
        'launch', 'px4_config.yaml'
    )
    px4_pluginlists_path = os.path.join(
        get_package_share_directory('mavros'),
        'launch', 'px4_pluginlists.yaml'
    )

    return LaunchDescription([

        Node(
            package='mavros',
            executable='mavros_node',
            parameters=[
                px4_pluginlists_path,
                px4_config_path,
            {    
                'fcu_url': '/dev/ttyACM0:921600',
                'gcs_url': '',
                'tgt_system': 1,
                'tgt_component': 1,
                'fcu_protocol': "v2.0",
                'respawn_mavros': "false",
                'namespace': "mavros",
            }],
        ),
    ])
```

px4_pluginlists.yaml
```
/**:
  ros__parameters:
    plugin_denylist:
      - '*'

      
    # plugin_allowlist:
    #   - 'sys_*'

```
mavros_node_output
```
[mavros_node-1] [INFO] [1710311570.165550198] [mavros]: Plugin actuator_control ignored
[mavros_node-1] [INFO] [1710311570.165688965] [mavros]: Plugin adsb ignored
[mavros_node-1] [INFO] [1710311570.165709479] [mavros]: Plugin altitude ignored
[mavros_node-1] [INFO] [1710311570.165725033] [mavros]: Plugin cam_imu_sync ignored
[mavros_node-1] [INFO] [1710311570.165738795] [mavros]: Plugin camera ignored
[mavros_node-1] [INFO] [1710311570.165751244] [mavros]: Plugin cellular_status ignored
[mavros_node-1] [INFO] [1710311570.165762861] [mavros]: Plugin command ignored
[mavros_node-1] [INFO] [1710311570.165775055] [mavros]: Plugin companion_process_status ignored
[mavros_node-1] [INFO] [1710311570.165786672] [mavros]: Plugin debug_value ignored
[mavros_node-1] [INFO] [1710311570.165797201] [mavros]: Plugin distance_sensor ignored
[mavros_node-1] [INFO] [1710311570.165807730] [mavros]: Plugin esc_status ignored
[mavros_node-1] [INFO] [1710311570.165818996] [mavros]: Plugin esc_telemetry ignored
[mavros_node-1] [INFO] [1710311570.165829589] [mavros]: Plugin fake_gps ignored
[mavros_node-1] [INFO] [1710311570.165841046] [mavros]: Plugin ftp ignored
[mavros_node-1] [INFO] [1710311570.165853047] [mavros]: Plugin geofence ignored
[... 22 lines cut ...]
[mavros_node-1] [INFO] [1710311570.166217215] [mavros]: Plugin px4flow ignored
[mavros_node-1] [INFO] [1710311570.166227841] [mavros]: Plugin rallypoint ignored
[mavros_node-1] [INFO] [1710311570.166238402] [mavros]: Plugin rangefinder ignored
[mavros_node-1] [INFO] [1710311570.166248643] [mavros]: Plugin rc_io ignored
[mavros_node-1] [INFO] [1710311570.166259428] [mavros]: Plugin setpoint_accel ignored
[mavros_node-1] [INFO] [1710311570.166269637] [mavros]: Plugin setpoint_attitude ignored
[mavros_node-1] [INFO] [1710311570.166280198] [mavros]: Plugin setpoint_position ignored
[mavros_node-1] [INFO] [1710311570.166290504] [mavros]: Plugin setpoint_raw ignored
[mavros_node-1] [INFO] [1710311570.166300905] [mavros]: Plugin setpoint_trajectory ignored
[mavros_node-1] [INFO] [1710311570.166311626] [mavros]: Plugin setpoint_velocity ignored
[mavros_node-1] [INFO] [1710311570.166322027] [mavros]: Plugin sys_status ignored
[mavros_node-1] [INFO] [1710311570.166332268] [mavros]: Plugin sys_time ignored
[mavros_node-1] [INFO] [1710311570.166343053] [mavros]: Plugin tdr_radio ignored
[mavros_node-1] [INFO] [1710311570.166353390] [mavros]: Plugin terrain ignored
[mavros_node-1] [INFO] [1710311570.166363984] [mavros]: Plugin trajectory ignored
[mavros_node-1] [INFO] [1710311570.166374417] [mavros]: Plugin tunnel ignored
[mavros_node-1] [INFO] [1710311570.166384754] [mavros]: Plugin vfr_hud ignored
[mavros_node-1] [INFO] [1710311570.166395315] [mavros]: Plugin vibration ignored
[mavros_node-1] [INFO] [1710311570.166406004] [mavros]: Plugin vision_pose ignored
[mavros_node-1] [INFO] [1710311570.166416309] [mavros]: Plugin vision_speed ignored
[mavros_node-1] [INFO] [1710311570.166427095] [mavros]: Plugin waypoint ignored
[mavros_node-1] [INFO] [1710311570.166437592] [mavros]: Plugin wheel_odomotry ignored
[mavros_node-1] [INFO] [1710311570.166448121] [mavros]: Plugin wind_estimation ignored

```

gregoryw3, 2024-03-28:
I was able to figure it some things out.
For some reason mavros ignores any config passed to it.

If you sudo edit "/opt/ros/iron/share/mavros/launch/px4_pluginlists.yaml" then launch it, it shows the same output that [limshoonkit](https://github.com/limshoonkit) has. Once mavros exits or maybe upon launching mavros again, the file is rewritten to be in it's default state.

gregoryw3, 2024-03-28:
I was able to get it (at least in iron and python package):

- package
   - config
     - px4_config.yaml
     - px4_pluginlists.yaml
   - package
   - launch
      - px4.launch.py 
   - resource
   - setup.py
   
px4_config.yaml
`default`

px4_pluginlists.yaml
```
/**:
  ros__parameters:
    plugin_denylist:
      - '*'

      
    # plugin_allowlist:
    #   - 'sys_*'
```

px4.launch.py
```
px4.launch.py
import os

from ament_index_python import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    px4_config_path = os.path.join(
        get_package_share_directory('controls'),
        'config', 'px4_config.yaml'
    )
    px4_pluginlists_path = os.path.join(
        get_package_share_directory('controls'),
        'config', 'px4_pluginlists.yaml'
    )

    return LaunchDescription([

        Node(
            package='mavros',
            executable='mavros_node',
            parameters=[
                px4_pluginlists_path,
                px4_config_path,
            {    
                'fcu_url': '/dev/ttyACM0:921600',
                'gcs_url': '',
                'tgt_system': 1,
                'tgt_component': 1,
                'fcu_protocol': "v2.0",
                'respawn_mavros': "false",
                'namespace': "mavros",
            }],
        ),
    ])
```

setup.py
```
import os
from glob import glob
from setuptools import find_packages, setup

package_name = 'controls'

setup(
    name=package_name,
    version='0.0.1',
    packages=find_packages(),
    py_modules=[],
    install_requires=[
        'setuptools',
        ],
    data_files=[
        ('share/ament_index/resource_index/packages',
            ['resource/' + package_name]),
        ('share/' + package_name, ['package.xml']),
        (os.path.join('share', package_name, 'launch'), glob(os.path.join('launch', '*.launch*'))),
        (os.path.join('share', package_name, 'config'), glob(os.path.join('config', '*'))),
        

    ],
    entry_points={
        'console_scripts': [
            'mavros_coms = controls.mavros_coms:main'
        ],
    },
)

```

joseVolair, 2024-07-03:
For me it finally worked when I removed `\mavros` from the apm_pluginlist file:
```
/mavros/**:
  ros__parameters:
    plugin_denylist:
      # common
      - actuator_control
      - ftp
      - hil
      - extras
      - altitude
      - debug_value
      - image_pub
      - px4flow
      - vibration
      - distance_sensor
      # - vision_speed_estimate
      - wheel_odometry

    # plugin_allowlist:
    #   - 'sys_*'
```

to 

```
/**:
  ros__parameters:
    plugin_denylist:
      # common
      - actuator_control
      - ftp
      - hil
      - extras
      - altitude
      - debug_value
      - image_pub
      - px4flow
      - vibration
      - distance_sensor
      # - vision_speed_estimate
      - wheel_odometry

    # plugin_allowlist:
    #   - 'sys_*'
```

mnumanuyar, 2025-01-20:
to use it in cli use it like: `ros2 run mavros mavros_node --ros-args -p mavros:plugin_denylist:=[wheel_odometry]`

closed 2026-10-15 by github-actions[bot]