mavlink/mavros #1926: log_output option ignored in launch files

tropappar, 2023-11-27:
### Issue details

I don't want to see the output of the MAVROS node on the terminal screen. The `log_output` option is ignored in the `apm.launch` and `node.launch` launch files.

### MAVROS version and platform

Mavros: 2.6.0
ROS: Humble
Ubuntu: 22.04

### Autopilot type and version

[X] ArduPilot
[X] PX4

ilicmilannis, 2024-10-24:
Just following up on this issue — has there been any progress on addressing the ignored log_output option in the launch files (`apm.launch` and `node.launch`)?

As a temporary workaround, I've been using `grep -v mavros_node` to filter out the MAVROS node output from the terminal. It works for now, but would appreciate any updates on a more permanent fix.

Thanks!

robert-conservaition, 2024-11-21:
Having this same issue. I cannot find anywhere in the repo where it references log_output.

closed 2026-10-15 by github-actions[bot]