mavlink/mavros #1931: Errors when building with Clang

mishmetall, 2024-02-03:
When building with Clang, I'm getting these errors:
```
> colcon build --cmake-args -DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++

...
In file included from /home/rosdev/ros2_ws/src/mavros/mavros/src/lib/mavros_router.cpp:20:
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_router.hpp:72:32: error: allocating an object of abstract class type 'mavros::router::Endpoint'
  RCLCPP_SMART_PTR_DEFINITIONS(Endpoint)
                               ^
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_router.hpp:103:16: note: unimplemented pure virtual method 'is_open' in 'Endpoint'
  virtual bool is_open() = 0;
               ^
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_router.hpp:104:40: note: unimplemented pure virtual method 'open' in 'Endpoint'
  virtual std::pair<bool, std::string> open() = 0;
                                       ^
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_router.hpp:105:16: note: unimplemented pure virtual method 'close' in 'Endpoint'
  virtual void close() = 0;
               ^
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_router.hpp:107:16: note: unimplemented pure virtual method 'send_message' in 'Endpoint'
  virtual void send_message(
               ^
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_router.hpp:113:16: note: unimplemented pure virtual method 'diag_run' in 'Endpoint'
  virtual void diag_run(diagnostic_updater::DiagnosticStatusWrapper & stat) = 0;
               ^
In file included from /home/rosdev/ros2_ws/src/mavros/mavros/src/lib/uas_data.cpp:19:
In file included from /home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/mavros_uas.hpp:45:
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/plugin.hpp:68:32: error: allocating an object of abstract class type 'mavros::plugin::Plugin'
  RCLCPP_SMART_PTR_DEFINITIONS(Plugin)
                               ^
/home/rosdev/ros2_ws/src/mavros/mavros/include/mavros/plugin.hpp:95:25: note: unimplemented pure virtual method 'get_subscriptions' in 'Plugin'
  virtual Subscriptions get_subscriptions() = 0;
```
### MAVROS version and platform

Mavros: [2e20c1c](https://github.com/mavlink/mavros/commit/2e20c1cbe62fe2bdec51ae6f63bdd740dee39cb7)
ROS: Humble, Foxy
Ubuntu: 22.04, 20.04

vooon, 2024-02-05:
Clang, what for? Never tried to build rclcpp with it.

mishmetall, 2024-02-05:
Clang has various features for analysis, e.g. [Quality guide: ensuring code quality](https://docs.ros.org/en/foxy/The-ROS2-Project/Contributing/Quality-Guide.html), but it's also a good thing to not depend on a single WoW, gcc and clang are the most widely used compilers

vooon, 2024-02-05:
So, does rclcpp and other components compile fine? I'm not quite interested in jumping and changing rcl's macros across whole codebase of mavros.

mishmetall, 2024-02-05:
Not sure about rclcpp, never tried, but other components that I tried to compile just fine. I guess Mavros will compile just fine as well if this problem is fixed, I have seen only this issue so far

mishmetall, 2024-02-05:
It's mentioned in the documentation that it is possible to do [Ros2 Building with clang ](https://docs.ros.org/en/eloquent/Installation/Linux-Development-Setup.html#clang)

mishmetall, 2025-05-06:
The problem is that RCLCPP_SMART_PTR_DEFINITIONS expands to among others to RCLCPP_UNIQUE_PTR_DEFINITIONS(__VA_ARGS__) which expands to
  __RCLCPP_UNIQUE_PTR_ALIAS(__VA_ARGS__) \
  __RCLCPP_MAKE_UNIQUE_DEFINITION(__VA_ARGS__)

and __RCLCPP_MAKE_UNIQUE_DEFINITION is
template<typename ... Args> \
  static std::unique_ptr<__VA_ARGS__> \
  make_unique(Args && ... args) \
  { \
    return std::unique_ptr<__VA_ARGS__>(new __VA_ARGS__(std::forward<Args>(args) ...)); \
  }

which is not possible for abstract classes, the alternative is to use RCLCPP_SMART_PTR_DEFINITIONS_NOT_COPYABLE which is same as RCLCPP_SMART_PTR_DEFINITIONS but without this make_unique function

Created a pull request https://github.com/mavlink/mavros/pull/2037

closed 2026-08-25 by Felaadar