mavlink/mavros #1933: Command_Int:UNSUPPORTED

nraines2002, 2024-02-14:
### Issue details

Sending MAVLink messages via Command_Int is returning COMMAND_ACK:[COMMAND_NAME]:UNSUPPORTED. I am using MAVROS built using ROS2 Foxy to pack and send these commands, using the Command_Long MAVROS service to send a Set_Mode message works but returns COMMAND_ACK:DO_SET_MODE:DENIED. I have sent the message with frames, MAV_FRAME_GLOBAL, MAV_FRAME_LOCAL_NED, MAV_FRAME_MISSION, MAV_FRAME_GLOBAL_RELATIVE_ALT, MAV_FRAME_GLOBAL_RELATIVE_ALT_INT to no avail.

Current usage:

`auto request = std::make_shared<mavros_msgs::srv::CommandInt::Request>();`
`                request->frame = mavros_msgs::msg::Waypoint::FRAME_GLOBAL_REL_ALT;`
`                request->command = 113;`
`                request->current = 0;`
`                request->autocontinue = 0;`
`                request->param1 = 0;`                        
`                request->param2 = 0;`                        
`                request->param3 = 0;`                        
`                request->param4 = 0;`                        
`                request->x = 0;`                        
`                request->y = 0;`                        
`                request->z = alt; (altitude in meters)`
`                                                                      `
`                auto result = comm_client_->async_send_request(request);`

### MAVROS version and platform

Mavros: 2.6.0
ROS: Foxy(ROS2)
Ubuntu: 20.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 4.4/4.3/4.2

### Node logs
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mavros_node-1]: process started with pid [31941]
[mavros_node-1] [INFO] [1707933903.665168676] [mavros.mavros_node]: Starting mavros_node container
[mavros_node-1] [INFO] [1707933903.665209627] [mavros.mavros_node]: FCU URL: udp://127.0.0.1:14551@14555
[mavros_node-1] [INFO] [1707933903.665214103] [mavros.mavros_node]: GCS URL: 
[mavros_node-1] [INFO] [1707933903.665217100] [mavros.mavros_node]: UAS Prefix: /uas1
[mavros_node-1] [INFO] [1707933903.665219714] [mavros.mavros_node]: Starting mavros router node
[mavros_node-1] [INFO] [1707933903.666820160] [mavros.mavros_router]: Built-in SIMD instructions: SSE, SSE2
[mavros_node-1] [INFO] [1707933903.666834159] [mavros.mavros_router]: Built-in MAVLink package version: 2022.12.30
[mavros_node-1] [INFO] [1707933903.666837857] [mavros.mavros_router]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all cubepilot development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
[mavros_node-1] [INFO] [1707933903.666841316] [mavros.mavros_router]: MAVROS Router started
[mavros_node-1] [INFO] [1707933903.666910852] [mavros.mavros_router]: Requested to add endpoint: type: 0, url: udp://127.0.0.1:14551@14555
[mavros_node-1] [INFO] [1707933903.667066687] [mavros.mavros_router]: Endpoint link[1000] created
[mavros_node-1] [INFO] [1707933903.667800774] [mavros.mavros_router]: link[1000] opened successfully
[mavros_node-1] [INFO] [1707933903.667853348] [mavros.mavros_router]: Requested to add endpoint: type: 2, url: /uas1
[mavros_node-1] [INFO] [1707933903.667909999] [mavros.mavros_router]: Endpoint link[1001] created
[mavros_node-1] [INFO] [1707933903.668478830] [mavros.mavros_router]: link[1001] opened successfully
[mavros_node-1] [INFO] [1707933903.668499282] [mavros.mavros_node]: Starting mavros uas node
[mavros_node-1] [INFO] [1707933903.692806472] [mavros.mavros]: UAS Executor started, threads: 16
[mavros_node-1] [INFO] [1707933903.702229853] [mavros.mavros]: Plugin actuator_control created
[mavros_node-1] [INFO] [1707933903.702450869] [mavros.mavros]: Plugin actuator_control initialized
[mavros_node-1] [INFO] [1707933903.712210043] [mavros.mavros]: Plugin adsb created
[mavros_node-1] [INFO] [1707933903.712367654] [mavros.mavros]: Plugin adsb initialized
[mavros_node-1] [INFO] [1707933903.714001194] [mavros.mavros]: Plugin altitude created
[mavros_node-1] [INFO] [1707933903.714112516] [mavros.mavros]: Plugin altitude initialized
[mavros_node-1] [INFO] [1707933903.715526701] [mavros.mavros]: Plugin cam_imu_sync created
[mavros_node-1] [INFO] [1707933903.715620668] [mavros.mavros]: Plugin cam_imu_sync initialized
[mavros_node-1] [INFO] [1707933903.717062842] [mavros.mavros]: Plugin camera created
[mavros_node-1] [INFO] [1707933903.717154404] [mavros.mavros]: Plugin camera initialized
[mavros_node-1] [INFO] [1707933903.718840078] [mavros.mavros]: Plugin cellular_status created
[mavros_node-1] [INFO] [1707933903.718858793] [mavros.mavros]: Plugin cellular_status initialized
[mavros_node-1] [INFO] [1707933903.722791091] [mavros.mavros]: Plugin command created
[mavros_node-1] [INFO] [1707933903.722904703] [mavros.mavros]: Plugin command initialized
[mavros_node-1] [INFO] [1707933903.724987898] [mavros.mavros]: Plugin companion_process_status created
[mavros_node-1] [INFO] [1707933903.725006676] [mavros.mavros]: Plugin companion_process_status initialized
[... 248 lines cut ...]
  - key: Current
    value: '-28.1'
  - key: Remaining
    value: '0.0'
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '2691'
  - key: Frequency (Hz)
    value: '0.999989'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: GUIDED
  - key: System status
    value: ACTIVE

### Check ID

(unsure of ROS2 command to check id)

vooon, 2024-02-15:
I'm unsure how you got unsupported, since command int do not expect to get COMMAND_ACK after COMMAND_INT:

https://github.com/mavlink/mavros/blob/227bb05c34a75188f1d6fccff6f767d0cc9129b0/mavros/src/plugins/command.cpp#L292-L302

Bthw: `ros2 run mavros mav checkid`

nraines2002, 2024-02-15:
> I'm unsure how you got unsupported, since command int do not expect to get COMMAND_ACK after COMMAND_INT:
> 
> https://github.com/mavlink/mavros/blob/227bb05c34a75188f1d6fccff6f767d0cc9129b0/mavros/src/plugins/command.cpp#L292-L302
> 
> Bthw: `ros2 run mavros mav checkid`

The COMMAND_ACK I see is from the specific commands that I'm sending using COMMAND_INT in the mavproxy window. So when I send a CONDITION_CHANGE_ALT command using command int i recieve "Got COMMAND_ACK:CONDITION_CHANGE_ALT:UNSUPPORTED"

vooon, 2024-02-15:
Ah, i see. MAVProxy catches COMMAND_ACK and prints it on that console.
The same message captured by mavros: `CMD: Unexpected command 113, result 3` (unexpected because as i said, mavros do not wait and do not expect to get ACK).

But that's simply on a firmware side. Have you tried to send that command via mavproxy?

vooon, 2024-02-15:
See also: https://mavlink.io/en/messages/common.html#MAV_RESULT

nraines2002, 2024-02-16:
I've found the issue, only very specific commands can be sent using COMMAND_INT. I will close the issue and update it to help any future lost souls.

Thanks for your help @vooon.

closed 2024-02-16 by nraines2002