mavlink/mavros #1936: linear velocities in odometry/in and local_position/odom are different for the same child frame

parakhm95, 2024-03-07:
----

### Issue details

I have been very confused by several topics on the mavros reporting similar yet slightly different data. We are flying using a companion/off-board computer with our PX4 flightcontroller. We noticed that despite having the same `frame_id`/`child_frame_id`,  the topics `odometry/in` and `local_position/odom` report different linear velocities as shown in the plot. The plot also contains `local_position/velocity_body` and `local_position/velocity_local`  and they are also different from both the `odometry/in` and `local_position/odom`. We have **three** different velocities being reported, none of which are the true derivatives of the position. This  
1. Only the `local_position/velocity_local` topic seems to do that correctly and I understand the others might be in the body frame but all three are different. So what is going on here and which is correct? I can provide  you the rosbag for this. 
2. Also, what are the purposes for these 4-5 different topics and which should we use for high-rate data for agile flights. There is also `global_position/global` and `global_position/local` so that is also confusing.

### MAVROS version and platform

Mavros: 1.13.0
ROS: Noetic
Ubuntu: 20.04

### Autopilot type and version

Pixhawk Orange CubePilotPlus on 1.14.0 and Holybro H-RTK F9P Helical

### Node logs

Will update tomorrow from the actual vehicle

### Diagnostics

Will update tomorrow from the actual vehicle

### Check ID

Will update tomorrow from the actual vehicle

[image]

parakhm95, 2024-03-07:
@penickar

parakhm95, 2024-03-21:
Is someone going to resolve this?

Yayantaff, 2024-07-11:
Hi @parakhm95 , late, i know, but were you able to resolve this ambiguity ?

parakhm95, 2024-07-11:
No, it was never fixed and so I moved my subscriber to `local_pose/odom`

closed 2026-10-14 by github-actions[bot]