mavlink/mavros #1938: Catkin Build : Ubuntu 20.04 Ros Neotic Mavros isntall

Pardeepanesar, 2024-03-15:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
When trying to install mavros I am having issues with catkin build in the catkin_ws

More specifically the issue is with the mavros_extras package relating to the gps_status.cpp

### MAVROS version and platform

Mavros: 1.18.0
ROS: Noetic
Ubuntu: 20.04

Catkin build logs below:

Errors     << mavros_extras:make /home/admin/catkin_ws/logs/mavros_extras/build.make.006.log
/home/admin/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp: In member function ‘void mavros::extra_plugins::GpsStatusPlugin::handle_gps2_raw(const mavlink_message_t*, mavlink::common::msg::GPS2_RAW&)’:
/home/admin/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:107:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘alt_ellipsoid’
  107 |   ros_msg->alt_ellipsoid     = mav_msg.alt_ellipsoid;
      |                                        ^~~~~~~~~~~~~
/home/admin/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:108:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘h_acc’
  108 |   ros_msg->h_acc             = mav_msg.h_acc;
      |                                        ^~~~~
/home/admin/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:109:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘v_acc’
  109 |   ros_msg->v_acc             = mav_msg.v_acc;
      |                                        ^~~~~
/home/admin/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:110:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘vel_acc’
  110 |   ros_msg->vel_acc           = mav_msg.vel_acc;
      |                                        ^~~~~~~
/home/admin/catkin_ws/src/mavros/mavros_extras/src/plugins/gps_status.cpp:111:40: error: ‘struct mavlink::common::msg::GPS2_RAW’ has no member named ‘hdg_acc’
  111 |   ros_msg->hdg_acc           = mav_msg.hdg_acc;
      |                                        ^~~~~~~
make[2]: *** [CMakeFiles/mavros_extras.dir/build.make:232: CMakeFiles/mavros_extras.dir/src/plugins/gps_status.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:735: CMakeFiles/mavros_extras.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/admin/catkin_ws/build/mavros_extras; catkin build --get-env mavros_extras | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

could anyone please help

installation I have followed: https://github.com/mavlink/mavros/tree/master/mavros#installation

bwh1270, 2024-04-01:
I encountered the same error while building during MAVROS source installation. 
Upon inspecting the gps_status.cpp file, it seemed that GPS_RAW and GPS_RAW2 had the same fields. 
So, I added the missing field to GPS2_RAW in the mavlink/message_definitions/v1.0/common.xml file, and the issue was resolved.

The code I added to GPS2_RAW with id=124 in the common.xml file is as follows:
```
<message id="124" name="GPS2_RAW">
      <description>Second GPS data.</description>
      ...
      <extensions/>
      <field type="int32_t" name="alt_ellipsoid" units="mm">Altitude (above WGS84, EGM96 ellipsoid). Positive for up.</field>
      <field type="uint32_t" name="h_acc" units="mm">Position uncertainty.</field>
      <field type="uint32_t" name="v_acc" units="mm">Altitude uncertainty.</field>
      <field type="uint32_t" name="vel_acc" units="mm">Speed uncertainty.</field>
      <field type="uint32_t" name="hdg_acc" units="degE5">Heading / track uncertainty</field>
      <field type="uint16_t" name="yaw" units="cdeg">Yaw in earth frame from north. Use 0 if this GPS does not provide yaw. Use 65535 if this GPS is configured to provide yaw and is currently unable to provide it. Use 36000 for north.</field>
</message>
...
```

guiaugustoga987, 2024-07-04:
> I encountered the same error while building during MAVROS source installation. Upon inspecting the gps_status.cpp file, it seemed that GPS_RAW and GPS_RAW2 had the same fields. So, I added the [...]

After I did this change I got the following error : 
```

make[2]: *** [CMakeFiles/paparazzi.xml-v2.0.dir/build.make:63: paparazzi-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/paparazzi.xml-v2.0.dir/all] Erro 2
make[1]: ** Esperando que outros processos terminem.
make[2]: *** [CMakeFiles/ualberta.xml-v2.0.dir/build.make:63: ualberta-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:289: CMakeFiles/ualberta.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/standard.xml-v2.0.dir/build.make:63: standard-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:118: CMakeFiles/standard.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/ASLUAV.xml-v2.0.dir/build.make:63: ASLUAV-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:640: CMakeFiles/ASLUAV.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/uAvionix.xml-v2.0.dir/build.make:63: uAvionix-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:343: CMakeFiles/uAvionix.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/matrixpilot.xml-v2.0.dir/build.make:63: matrixpilot-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:397: CMakeFiles/matrixpilot.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/autoquad.xml-v2.0.dir/build.make:63: autoquad-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:235: CMakeFiles/autoquad.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/ardupilotmega.xml-v2.0.dir/build.make:63: ardupilotmega-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:262: CMakeFiles/ardupilotmega.xml-v2.0.dir/all] Erro 2
make[2]: *** [CMakeFiles/all.xml-v2.0.dir/build.make:63: all-v2.0-cxx-stamp] Erro 1
make[1]: *** [CMakeFiles/Makefile2:694: CMakeFiles/all.xml-v2.0.dir/all] Erro 2
make: *** [Makefile:118: all] Erro 2
```

How can I solve it ?

vooon, 2024-07-04:
Try to remove build dir.

GabrielWellander, 2024-07-16:
> After I did this change I got the following error :
>
> ```
>
> make[2]: *** [CMakeFiles/paparazzi.xml-v2.0.dir/build.make:63: paparazzi-v2.0-cxx-stamp] Erro 1
> make[1]: *** [CMakeFiles/Makefile2:145: [...]

I've got the same issue and a 
'catkin clean -y
'
 did not work to solve my issue.

The error message I got
`make[2]: *** [CMakeFiles/standard.xml-v2.0.dir/build.make:63: standard-v2.0-cxx-stamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:118: CMakeFiles/standard.xml-v2.0.dir/all] Error 2
make: *** [Makefile:118: all] Error 2
`

Wangben1019, 2024-08-09:
> > > 我在 MAVROS 源安装过程中构建时遇到了同样的错误。检查 gps_status.cpp 文件后发现，GPS_RAW 和 GPS_RAW2 似乎具有相同的字段。因此，我在 mavlink/message_definitions/v1.0/common.xml 文件中将缺失的字段添加到 GPS2_RAW，问题得到解决。
> > > 我在 common.xml 文件中 id=124 的 GPS2_RAW 中添加的代码如下：
> > > ```
> > > <message id="124" name="GPS2_RAW">
> > >       <description>Second GPS data.</description>
> > >       ...
> > >       <extensions/>
> > >       <field type="int32_t" name="alt_ellipsoid" units="mm">Altitude (above WGS84, EGM96 ellipsoid). Positive for up.</field>
> > >       <field type="uint32_t" name="h_acc" units="mm">Position uncertainty.</field>
> > >       <field type="uint32_t" name="v_acc" units="mm">Altitude uncertainty.</field>
> > >       <field type="uint32_t" name="vel_acc" units="mm">Speed uncertainty.</field>
> > >       <field type="uint32_t" name="hdg_acc" units="degE5">Heading / track uncertainty</field>
> > >       <field type="uint16_t" name="yaw" units="cdeg">Yaw in earth frame from north. Use 0 if this GPS does not provide yaw. Use 65535 if this GPS is configured to provide yaw and is currently unable to provide it. Use 36000 for north.</field>
> > > </message>
> > > ...
> > > ```
> > 
> > 
> > 完成此更改后，出现以下错误：
> > ```
> > 
> > make[2]: *** [CMakeFiles/paparazzi.xml-v2.0.dir/build.make:63: paparazzi-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/paparazzi.xml-v2.0.dir/all] Erro 2
> > make[1]: ** Esperando que outros processos terminem.
> > make[2]: *** [CMakeFiles/ualberta.xml-v2.0.dir/build.make:63: ualberta-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:289: CMakeFiles/ualberta.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/standard.xml-v2.0.dir/build.make:63: standard-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:118: CMakeFiles/standard.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/ASLUAV.xml-v2.0.dir/build.make:63: ASLUAV-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:640: CMakeFiles/ASLUAV.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/uAvionix.xml-v2.0.dir/build.make:63: uAvionix-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:343: CMakeFiles/uAvionix.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/matrixpilot.xml-v2.0.dir/build.make:63: matrixpilot-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:397: CMakeFiles/matrixpilot.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/autoquad.xml-v2.0.dir/build.make:63: autoquad-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:235: CMakeFiles/autoquad.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/ardupilotmega.xml-v2.0.dir/build.make:63: ardupilotmega-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:262: CMakeFiles/ardupilotmega.xml-v2.0.dir/all] Erro 2
> > make[2]: *** [CMakeFiles/all.xml-v2.0.dir/build.make:63: all-v2.0-cxx-stamp] Erro 1
> > make[1]: *** [CMakeFiles/Makefile2:694: CMakeFiles/all.xml-v2.0.dir/all] Erro 2
> > make: *** [Makefile:118: all] Erro 2
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 我该如何解决它？
> 
> 我遇到了同样的问题，但 `catkin clean -y` 没有解决我的问题。
> 
> 我收到的错误消息 `make[2]: *** [CMakeFiles/standard.xml-v2.0.dir/build.make:63: standard-v2.0-cxx-stamp] Error 1 make[1]: *** [CMakeFiles/Makefile2:118: CMakeFiles/standard.xml-v2.0.dir/all] Error 2 make: *** [Makefile:118: all] Error 2 `

I think you should recheck your common.xml because the solution provided by [bwh1270](https://github.com/mavlink/mavros/issues/1938#issuecomment-2029293665) cannot be directly copied and pasted. He provided context to facilitate locating and changing the location

MSZ98, 2024-09-10:
You're trying to build MAVROS, but it won't build because it uses a MAVLink submodule and requires an older version of MAVLink. You'll need to swap out that submodule.

``` bash
cd .../catkin_ws/src
rm -rdf mavlink
git clone https://github.com/mavlink/mavlink.git -b 1.0.12
```

Also, you may have to remove the old build of mavros, mavros_extras and mavros_msgs.
```
cd .../catkin_ws/build
rm -rdf mavros mavros_extras mavros_msgs
```

Now build again and it should be working.

prodrone-liu, 2024-09-12:
cakin build successfully with mavros: 1.18.0 and mavlink: 1.0.12 in ROS Noetic.

vooon, 2024-09-12:
@MSZ98 you're wrong. https://github.com/mavlink/mavlink-gbp-release (or https://github.com/ros2-gbp/mavlink-gbp-release) must be used, and rosinstall_generator should provide it.

closed 2026-10-14 by github-actions[bot]