mavlink/mavros #1939: [mavros_node-1] [INFO] [1710587963.274799146] [mavros.sys]: FCU: EVENT 377561 with args

yst-one, 2024-03-16:
-
[mavros_node-1] [WARN] [1710587961.227784585] [mavros.sys]: FCU: EVENT 453929 with args -255-255-255-255-7-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [INFO] [1710587961.228064984] [mavros.sys]: FCU: EVENT 6801787 with args -255-255-255-255-12-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [WARN] [1710587961.228304964] [mavros.sys]: FCU: UNK(8): EVENT 1914663 with args -0-2-0-2-40-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [WARN] [1710587963.274236291] [mavros.sys]: FCU: UNK(8): EVENT 11047904 with args -0-0-0-16-0-128-0-0-0-192-59-129-255-192-59-129-255-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [ERROR] [1710587963.274499803] [mavros.sys]: FCU: EVENT 13835193 with args -60-0-126-0-20-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [ERROR] [1710587963.274665854] [mavros.sys]: FCU: EVENT 10011251 with args -56-0-0-0-20-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [INFO] [1710587963.274799146] [mavros.sys]: FCU: EVENT 377561 with args -8-0-0-0-20-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [ERROR] [1710587963.274996881] [mavros.sys]: FCU: EVENT 3087815 with args -0-64-0-0-20-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [INFO] [1710587963.275248356] [mavros.sys]: FCU: EVENT 79408 with args -32-0-0-0-20-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [WARN] [1710587963.275453510] [mavros.sys]: FCU: EVENT 453929 with args -255-255-255-255-7-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [INFO] [1710587963.275588535] [mavros.sys]: FCU: EVENT 6801787 with args -255-255-255-255-12-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-
[mavros_node-1] [WARN] [1710587963.275710582] [mavros.sys]: FCU: UNK(8): EVENT 1914663 with args -0-2-0-2-40-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-0-

vooon, 2024-03-18:
And what's the problem?

yst-one, 2024-04-19:
humble problem

yst-one, 2024-04-19:
I am currently This situation was also encountered in the simulation of ROS2 Humble PX4

yst-one, 2024-04-19:
[image]

vooon, 2024-04-19:
I do not see any error here. Your FCU sends those lines (STATUSTEXT).

Yyote, 2024-06-26:
For me this messages also sometimes result in px4 losing offboard signal. There are quite a lot of them. I use ROS2, gazebo garden and PX4 Autopilot on ubuntu 22.04

vooon, 2024-06-26:
Closing as no activity.

closed 2024-06-26 by vooon