mavlink/mavros #1941: Issues with building MAVROS with ROS2 humble in Docker Container

maximecapelle, 2024-03-27:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Hi, 
I'm sorry if this has been mentioned before but I am consistently running into the same problem when trying to colcon build the MAVROS package. I am running ROS2 humble in a Docker container and everytime I try to colcon build the packages in my src folder the MAVROS package fails with the following error: 

-------------------------------------------------------------------------------------------------------------------
Starting >>> mavros
--- stderr: mavros                           
CMake Warning at /usr/src/gtest/CMakeLists.txt:54 (project):
  VERSION keyword not followed by a value or was followed by a value that
  expanded to nothing.

CMake Deprecation Warning at /usr/src/gtest/CMakeLists.txt:56 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Warning at /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_cmake_gmock-extras.cmake:118 (message):
  'gmock' not found, C++ tests using 'Google Mock' can not be built.  Please
  install the 'Google Mock' headers globally in your system to enable these
  tests (e.g.  on Ubuntu/Debian install the package 'google-mock') or get the
  ament package 'gmock_vendor'
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_gmock/cmake/ament_add_gmock.cmake:50 (_ament_cmake_gmock_find_gmock)
  CMakeLists.txt:254 (ament_add_gmock)

CMake Error at CMakeLists.txt:255 (target_link_libraries):
  Cannot specify link libraries for target "mavros-router-test" which is not
  built by this project.

---
Failed   <<< mavros [1.23s, exited with code 1]

------------------------------------------------------------------------------------------------------------------------------------------------------------------

Maybe I am clearly just missing something simple but I would really appreciate some help. The link library "mavros-router-test" keeps causing problems for me. Note, I have managed to install this in the past with ROS1 noetic without issues, but the install for ROS2 humble is more of a struggle. Are there install instructions that are updated for ROS2 humble, the current ones seem to only use catkin. 

My current install process is:
sudo apt install -y python3-vcstool python3-rosinstall-generator python3-osrf-pycommon
rosinstall_generator --format repos mavlink | tee /tmp/mavlink.repos
rosinstall_generator --format repos --upstream mavros | tee -a /tmp/mavros.repos
vcs import src < /tmp/mavlink.repos
vcs import src < /tmp/mavros.repos
rosdep install --from-paths src --ignore-src -y
sudo ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh
colcon build --packages-select mavros mavlink

Then the error occurs...

Thank you in advance! 

### MAVROS version and platform

Mavros: latest ros2 branch
ROS: Humble
Ubuntu: 20.04

### Autopilot type and version

[ X] ArduPilot
[  ] PX4

maximecapelle, 2024-03-27:
Hey, 
thank you for the fast reply. I have tried the line you suggested and the same error seems to appear. This is what I am currently running:

# Build ROS dependency
echo "source /opt/ros/${ROS_DISTRO}/setup.bash" >> ~/.bashrc
source /opt/ros/${ROS_DISTRO}/setup.bash

sudo apt-get update
rosdep update

# MAVROS
rosdep update && rosdep install -y -i --from-paths src
rosdep install --from-paths ${ROS_WORKSPACE} --ignore-src --rosdistro=${ROSDISTRO}
sudo apt install -y python3-vcstool python3-rosinstall-generator python3-osrf-pycommon
rosinstall_generator --format repos mavlink | tee /tmp/mavlink.repos
rosinstall_generator --format repos --upstream mavros | tee -a /tmp/mavros.repos
vcs import src < /tmp/mavlink.repos
vcs import src < /tmp/mavros.repos
sudo ./src/mavros/mavros/scripts/install_geographiclib_datasets.sh
colcon build --packages-select mavros mavlink

I have tried many things now so sorry if the commands are a bit cluttered :)

vooon, 2024-03-27:
Try to complete this instructins in a new workspace: https://github.com/mavlink/mavros/tree/ros2/mavros#source-installation

Do not forget to `source /opt/ros/humble/setup.bash` before you're first call to `colcon`.

maximecapelle, 2024-03-28:
I should also add, I had to build ROS2 Humble on Ubuntu 20.04 for the Isaac ros pkgs. This might have some effect.

vooon, 2024-03-28:
Just do all the steps manually first. Simply `docker exec -ti ros:humble` and run the commands.

That's my "ros-in-docker.txt", which works pretty fine:
```
docker run -it --name ros2-iron -v $HOME/ros2:/ws -v $HOME/.gitconfig:/root/.gitconfig -v /root/.ssh:/root/.ssh -v $HOME/.gnupg:/root/.gnupg -v $HOME/.config:/root/.config -p 14540:14540/udp -p 14545:14545/udp -p 5761:5761 ros:iron

# apt update && apt install ranger tig vim openssh-client python3-bloom python3-future
# rosdep update && rosdep install -y -i --from-paths src
```