mavlink/mavros #1948: mavros_msgs/srv/CommandLong service not working

kursatkomurcu, 2024-05-07:
### Issue details

I am trying to send commands from laptop to drone using telemetry module. I could takeoff the drone but I couldn't move it. I tried **ros2 service call /mavros/cmd/command mavros_msgs/srv/CommandLong "{command: 0, confirmation: 0, param1: 1.0, param2: 0.0, param3: 0.0, param4: 0.0, param5: 0.0, param6: 0.0, param7: 1.0}"** command but the response is **mavros_msgs.srv.CommandLong_Response(success=False, result=3)**

How can I move the drone from terminal?

### MAVROS version and platform

Mavros: 2.7
ROS: Humble
Ubuntu: 22.04

### Autopilot type and version

[  ] ArduPilot

### Node logs

```
(base) kursat@kursat:~/tracking-drone-0.0.1$ ros2 service call /mavros/cmd/arming mavros_msgs/srv/CommandBool "{value: true}"
requester: making request: mavros_msgs.srv.CommandBool_Request(value=True)

response:
mavros_msgs.srv.CommandBool_Response(success=True, result=0)

(base) kursat@kursat:~/tracking-drone-0.0.1$ ros2 service call /mavros/cmd/takeoff mavros_msgs/srv/CommandTOL "{min_pitch: 0.0, yaw: 0.0, latitude: 0.0, longitude: 0.0, altitude: 1.0}"
requester: making request: mavros_msgs.srv.CommandTOL_Request(min_pitch=0.0, yaw=0.0, latitude=0.0, longitude=0.0, altitude=1.0)

response:
mavros_msgs.srv.CommandTOL_Response(success=True, result=0)

(base) kursat@kursat:~/tracking-drone-0.0.1$ ros2 service call /mavros/cmd/command mavros_msgs/srv/CommandLong "{command: 0, param1: 1.0, param2: 0.0, param3: 0.0, param4: 0.0, param5: 0.0, param6: 0.0, param7: 0.0}"
requester: making request: mavros_msgs.srv.CommandLong_Request(broadcast=False, command=0, confirmation=0, param1=1.0, param2=0.0, param3=0.0, param4=0.0, param5=0.0, param6=0.0, param7=0.0)

response:
mavros_msgs.srv.CommandLong_Response(success=False, result=3)

```

### Diagnostics

```
(base) kursat@kursat:~/tracking-drone-0.0.1$ ros2 topic echo /diagnostics
header:
  stamp:
    sec: 1715085353
    nanosec: 690737283
  frame_id: ''
status:
- level: "\0"
  name: 'mavros_router: MAVROS Router'
  message: ok
  hardware_id: none
  values:
  - key: Endpoints
    value: '2'
  - key: Messages routed
    value: '33869'
  - key: Messages sent
    value: '33869'
  - key: Messages dropped
    value: '0'
- level: "\0"
  name: 'mavros_router: endpoint 1000: /dev/ttyUSB0'
  message: ok
  hardware_id: none
  values:
  - key: Received packets
    value: '19440'
  - key: Dropped packets
    value: '0'
  - key: Buffer overruns
    value: '0'
  - key: Parse errors
    value: '0'
  - key: Rx sequence number
    value: '204'
  - key: Tx sequence number
    value: '0'
  - key: Rx total bytes
    value: '511182'
  - key: Tx total bytes
    value: '384798'
  - key: Rx speed
    value: '434.000000'
  - key: Tx speed
    value: '320.000000'
  - key: Remotes count
    value: '12'
  - key: Remote [0]
    value: '0.0'
  - key: Remote [1]
    value: '1.0'
  - key: Remote [2]
    value: '1.1'
  - key: Remote [3]
    value: '10.0'
  - key: Remote [4]
    value: '10.250'
  - key: Remote [5]
    value: '42.0'
  - key: Remote [6]
    value: '42.100'
  - key: Remote [7]
    value: '42.250'
  - key: Remote [8]
    value: '51.0'
  - key: Remote [9]
    value: '51.68'
  - key: Remote [10]
    value: '255.0'
  - key: Remote [11]
    value: '255.190'
- level: "\0"
  name: 'mavros_router: endpoint 1001: /uas1'
  message: ok
  hardware_id: none
  values:
  - key: Remotes count
    value: '3'
  - key: Remote [0]
    value: '0.0'
  - key: Remote [1]
    value: '1.0'
  - key: Remote [2]
    value: '1.191'
---
header:
  stamp:
    sec: 1715085353
    nanosec: 700292359
  frame_id: ''
status:
- level: "\0"
  name: 'mavros: MAVROS UAS'
  message: connected
  hardware_id: uas:///uas1
  values: []
- level: "\x02"
  name: 'mavros: GPS'
  message: No satellites
  hardware_id: uas:///uas1
  values:
  - key: Satellites visible
    value: '0'
  - key: Fix type
    value: '0'
  - key: EPH (m)
    value: Unknown
  - key: EPV (m)
    value: Unknown
- level: "\x01"
  name: 'mavros: Mount'
  message: Can not diagnose in this targeting mode
  hardware_id: uas:///uas1
  values:
  - key: Mode
    value: '255'
- level: "\0"
  name: 'mavros: System'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Sensor present
    value: '0x00000000'
  - key: Sensor enabled
    value: '0x00000000'
  - key: Sensor health
    value: '0x00000000'
  - key: CPU Load (%)
    value: '0.0'
  - key: Drop rate (%)
    value: '0.0'
  - key: Errors comm
    value: '0'
  - key: 'Errors count #1'
    value: '0'
  - key: 'Errors count #2'
    value: '0'
  - key: 'Errors count #3'
    value: '0'
  - key: 'Errors count #4'
    value: '0'
- level: "\x02"
  name: 'mavros: Battery'
  message: No data
  hardware_id: uas:///uas1
  values:
  - key: Voltage
    value: '-1.00'
  - key: Current
    value: '0.0'
  - key: Remaining
    value: '0.0'
- level: "\0"
  name: 'mavros: Heartbeat'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Heartbeats since startup
    value: '1197'
  - key: Frequency (Hz)
    value: '0.999998'
  - key: Vehicle type
    value: Quadrotor
  - key: Autopilot type
    value: ArduPilot
  - key: Mode
    value: LAND
  - key: System status
    value: STANDBY
- level: "\0"
  name: 'mavros: Time Sync'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: Timesyncs since startup
    value: '11902'
  - key: Frequency (Hz)
    value: '10.099976'
  - key: Last RTT (ms)
    value: '98.245383'
  - key: Mean RTT (ms)
    value: '160.806135'
  - key: Last remote time (s)
    value: '1366.796581000'
  - key: Estimated time offset (s)
    value: '0.000000000'
- level: "\0"
  name: 'mavros: 3DR Radio'
  message: Normal
  hardware_id: uas:///uas1
  values:
  - key: RSSI
    value: '199'
  - key: RSSI (dBm)
    value: '-22.3'
  - key: Remote RSSI
    value: '196'
  - key: Remote RSSI (dBm)
    value: '-23.8'
  - key: Tx buffer (%)
    value: '100'
  - key: Noice level
    value: '59'
  - key: Remote noice level
    value: '42'
  - key: Rx errors
    value: '1'
  - key: Fixed
    value: '0'
---

```

vooon, 2024-05-07:
You should provide correct command code: https://mavlink.io/en/messages/common.html#mav_commands
But you receiving correct response from the FCU: https://mavlink.io/en/messages/common.html#MAV_RESULT

XXLiu-HNU, 2024-09-10:
> You should provide correct command code: https://mavlink.io/en/messages/common.html#mav_commands But you receiving correct response from the FCU: https://mavlink.io/en/messages/common.html#MAV_RESULT

hello,  i use SET_GPS_GLOBAL_ORIGIN (48) to set the origin of the uav, and i use `ros2 service call /mavros/cmd/command_int mavros_msgs/srv/CommandInt "{command: 48, param1: 1, param2: 503977420, param3: 85455940, param4: 500000}" `for test,
 and i get, it looks like good :
[image]
but in the terminal of running `ros2 run mavros mavros_node --ros-args --param fcu_url:=udp://:14540@` i get :
[image]
And the Local_position is not change, some mistake maybe, the version of PX4 is v1.15 .

vooon, 2024-09-12:
http://docs.ros.org/en/noetic/api/mavros_msgs/html/srv/CommandInt.html & [MAV_RESULT_UNSUPPORTED](https://mavlink.io/en/messages/common.html#MAV_RESULT_UNSUPPORTED)

closed 2026-10-13 by github-actions[bot]