mavlink/mavros #1949: installing issue

moyubailan, 2024-05-09:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
can i use ROS2 rolling to install mavros?the terminal failed to acquire "ros-rolling-mavros" package.

### MAVROS version and platform

Mavros: 2.7.0
ROS: rolling
Ubuntu: 22.04

### Node logs

no.because i haven't come to this yet. ```
copy output of mavros_node. Usually console where you run roslaunch
```

vooon, 2024-05-12:
https://build.ros2.org/job/Rdev__mavros__ubuntu_noble_amd64/1/console

Looks like some problem. Anyway rolling not the best idea to use. Consider building from source.

TheMightiestCarrot, 2024-06-02:
I am experiencing the same issue. Have you managed to resolve it?

moyubailan, 2024-06-07:
> 发送日期：2024-05-12 17:54:55
> 抄送人："Fan Kaiwen" ***@***.***>, Author ***@***.***>
> 主 题：Re: [mavlink/mavros] installing issue (Issue #1949)
>
> https://build.ros2.org/job/Rdev__mavros__ubuntu_noble_amd64/1/console

Compris.Merci beaucoup

closed 2026-10-13 by github-actions[bot]