mavlink/mavros #1951: mavros apm not runnig just restaring the process again and again

AAK121, 2024-05-17:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I'm trying to launch mavros but node is just restarting the process and gets killed automatically

### MAVROS version and platform

Mavros: 1.18.0
ROS: noetic
Ubuntu: 20.04 LTS

### Autopilot type and version

[ * ] ArduPilot
[  ] PX4

Version:  dont know

### Node logs

```
process[mavros-1]: started with pid [79751]
[ INFO] [1715936445.393076041]: FCU URL: udp://127.0.0.1:14551@14555
[ INFO] [1715936445.407841453]: udp0: Bind address: 127.0.0.1:14551
[ INFO] [1715936445.409003070]: udp0: Remote address: 0.0.56.219:14550
[ INFO] [1715936445.409721192]: GCS bridge disabled
[ INFO] [1715936445.466258433, 268.701000000]: Plugin 3dr_radio loaded
[ INFO] [1715936445.472299311, 268.706000000]: Plugin 3dr_radio initialized
[ INFO] [1715936445.472375812, 268.706000000]: Plugin actuator_control blacklisted
[ INFO] [1715936445.495181037, 268.722000000]: Plugin adsb loaded
[ INFO] [1715936445.505682629, 268.730000000]: Plugin adsb initialized
[ INFO] [1715936445.505924529, 268.731000000]: Plugin altitude blacklisted
[ INFO] [1715936445.506675391, 268.731000000]: Plugin cam_imu_sync loaded
[ INFO] [1715936445.509399808, 268.733000000]: Plugin cam_imu_sync initialized
[ INFO] [1715936445.509680248, 268.733000000]: Plugin camera loaded
[ INFO] [1715936445.512062538, 268.735000000]: Plugin camera initialized
[ INFO] [1715936445.512312845, 268.735000000]: Plugin cellular_status loaded
[ INFO] [1715936445.521033233, 268.742000000]: Plugin cellular_status initialized
[ INFO] [1715936445.521623565, 268.743000000]: Plugin command loaded
[ INFO] [1715936445.552828207, 268.758000000]: Plugin command initialized
[ INFO] [1715936445.553228597, 268.758000000]: Plugin companion_process_status loaded
[ INFO] [1715936445.562803082, 268.758000000]: Plugin companion_process_status initialized
[ INFO] [1715936445.562881200, 268.758000000]: Plugin debug_value blacklisted
[ INFO] [1715936445.563655184, 268.758000000]: Plugin distance_sensor loaded
[ INFO] [1715936445.600137998, 268.792000000]: Plugin distance_sensor initialized
[ INFO] [1715936445.601148935, 268.792000000]: Plugin esc_status loaded
[ INFO] [1715936445.606079470, 268.799000000]: Plugin esc_status initialized
[ INFO] [1715936445.606693754, 268.799000000]: Plugin esc_telemetry loaded
[ INFO] [1715936445.611294625, 268.801000000]: Plugin esc_telemetry initialized
[ INFO] [1715936445.611932513, 268.801000000]: Plugin fake_gps loaded
[ INFO] [1715936445.677899986, 268.850000000]: Plugin fake_gps initialized
[ INFO] [1715936445.678009972, 268.850000000]: Plugin ftp blacklisted
[ INFO] [1715936445.678370924, 268.851000000]: Plugin geofence loaded
[ INFO] [1715936445.692888704, 268.859000000]: Plugin geofence initialized
[ INFO] [1715936445.695335874, 268.863000000]: Plugin global_position loaded
[ INFO] [1715936445.766200303, 268.914000000]: Plugin global_position initialized
[ INFO] [1715936445.766563244, 268.914000000]: Plugin gps_input loaded
[ INFO] [1715936445.776105416, 268.920000000]: Plugin gps_input initialized
[ INFO] [1715936445.776695244, 268.921000000]: Plugin gps_rtk loaded
[ INFO] [1715936445.787176074, 268.926000000]: Plugin gps_rtk initialized
[ INFO] [1715936445.787510440, 268.926000000]: Plugin gps_status loaded
[ INFO] [1715936445.797843448, 268.931000000]: Plugin gps_status initialized
[ INFO] [1715936445.798388502, 268.931000000]: Plugin guided_target loaded
[ INFO] [1715936445.827231095, 268.953000000]: Plugin guided_target initialized
[ INFO] [1715936445.827361844, 268.953000000]: Plugin hil blacklisted
[ INFO] [1715936445.827816543, 268.954000000]: Plugin home_position loaded
[... 94 lines cut ...]
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: "mavros: FCU connection"
    message: "Node starting up"
    hardware_id: ''
    values: []
---

```

### Check ID

```
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

vooon, 2024-05-17:
Looks like there some incompatibility in boost.asio. Have you tried other connection methods?

AAK121, 2024-05-17:
> Looks like there some incompatibility in boost.asio. Have you tried other connection methods?

other connections methods means?

closed 2026-10-13 by github-actions[bot]