mavlink/mavros #1960: mavros_extras plugins not loaded with ROS2 Jazzy on Ubuntu 24.04

igorsgcampos, 2024-06-15:
### Issue details

I build mavros from source according to the Jazzy instructions (commit b0da849a06eb1a215b9205b92b8cf39c6d7cf88f (HEAD, tag: 2.8.0, origin/ros2, origin/HEAD, ros2)), but upon running the mavros node, it fails to load all plugins in the mavros_extras, although the std_plugins are loaded successfully.

### MAVROS version and platform

Mavros: 2.8.0
ROS: Jazzy
Ubuntu: 24.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: N/A

### Node logs

```
[INFO] [1718454301.043908190] [mavros]: Plugin actuator_control created
[INFO] [1718454301.045180490] [mavros]: Plugin actuator_control initialized
[ERROR] [1718454301.076783145] [mavros]: Plugin adsb load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[INFO] [1718454301.102932325] [mavros]: Plugin altitude created
[INFO] [1718454301.103591558] [mavros]: Plugin altitude initialized
[ERROR] [1718454301.121035135] [mavros]: Plugin cam_imu_sync load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.138585266] [mavros]: Plugin camera load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.155941899] [mavros]: Plugin cellular_status load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[INFO] [1718454301.235567926] [mavros]: Plugin command created
[INFO] [1718454301.236171270] [mavros]: Plugin command initialized
[ERROR] [1718454301.253585588] [mavros]: Plugin companion_process_status load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.270976054] [mavros]: Plugin debug_value load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.288362169] [mavros]: Plugin distance_sensor load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.305715728] [mavros]: Plugin esc_status load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.323401284] [mavros]: Plugin esc_telemetry load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.340907082] [mavros]: Plugin fake_gps load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[INFO] [1718454301.450236581] [mavros]: Plugin ftp created
[INFO] [1718454301.450928851] [mavros]: Plugin ftp initialized
[INFO] [1718454301.497113244] [mavros]: Plugin geofence created
[INFO] [1718454301.498707004] [mavros]: Plugin geofence initialized
[ERROR] [1718454301.516375318] [mavros]: Plugin gimbal_control load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[INFO] [1718454301.606536668] [mavros]: Plugin global_position created
[INFO] [1718454301.607774765] [mavros]: Plugin global_position initialized
[ERROR] [1718454301.625774354] [mavros]: Plugin gps_input load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.643209171] [mavros]: Plugin gps_rtk load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[ERROR] [1718454301.660592397] [mavros]: Plugin gps_status load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in t [... cut ...]
[ERROR] [1718454303.222686299] [mavros]: Plugin wheel_odometry load exception: Failed to load library /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: /home/igor.campos/ros2_ws/install/mavros_extras/lib/libmavros_extras_plugins.so: undefined symbol: _ZTIN4YAML13BadConversionE, at ./src/shared_library.c:99
[INFO] [1718454303.277355799] [mavros]: Plugin wind_estimation created
[INFO] [1718454303.278290806] [mavros]: Plugin wind_estimation initialized
```

sathak93, 2024-06-18:
this problem arises everywhere where yaml_cpp_vendor used. it seems it moved from using ament target dependencies  and requires to explicitly include yaml-cpp in target_link_libraries
 https://github.com/ros2/yaml_cpp_vendor/issues/51

MarekC96, 2024-08-09:
> this problem arises everywhere where yaml_cpp_vendor used. it seems it moved from using ament target dependencies and requires to explicitly include yaml-cpp in target_link_libraries [ros2/yaml_cpp_vendor#51](https://github.com/ros2/yaml_cpp_vendor/issues/51)

Had the same problem, added:
find_package(yaml-cpp REQUIRED)
target_link_libraries(mavros_extras_plugins
  yaml-cpp
) 
target_link_libraries(mavros_extras
  yaml-cpp
)

to the mavros_extras CMakeLists.txt package and everything works  now, all plugins initialized.

evan-palmer, 2024-09-25:
This should be resolved now by #1994

PR #1996 (2024-09-27): Update build list in README
As promised, here is the updated README, which includes the build statuses for the latest Jazzy and Rolling releases. It doesn't appear as though the badges are available for these releases yet, but folks should still be able to determine that the binaries are available.

(marking these two issues as fixed - I can remove these if you'd prefer to manually close them)
- Fixes #1960
- Fixes #1958
closed 2024-09-27 by vooon