mavlink/mavros #1969: ROS MAVROS build Error

peakyquest, 2024-07-23:
I am currently trying to build the mavros on ros noetic, I am getting this error.
Yesterday it was working fine, but when I added a custom message in mavros package, it failed to build after that and now even if I build the mavros from scratch I am still getting this error? what I am doing wrong ? 

```
Traceback (most recent call last):
  File "/opt/ros/noetic/share/catkin/cmake/interrogate_setup_dot_py.py", line 49, in <module>
    import setuptools
  File "/home/haroon/.local/lib/python3.8/site-packages/setuptools/__init__.py", line 21, in <module>
    from .dist import Distribution
  File "/home/haroon/.local/lib/python3.8/site-packages/setuptools/dist.py", line 30, in <module>
    from . import _entry_points
  File "/home/haroon/.local/lib/python3.8/site-packages/setuptools/_entry_points.py", line 44, in <module>
    def validate(eps: metadata.EntryPoints):
AttributeError: module 'importlib_metadata' has no attribute 'EntryPoints'
CMake Error at /opt/ros/noetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):

  execute_process(/home/haroon/catkin_ws/build/mavros/catkin_generated/env_cached.sh
  "/usr/bin/python3"
  "/opt/ros/noetic/share/catkin/cmake/interrogate_setup_dot_py.py" "mavros"
  "/home/haroon/catkin_ws/src/mavros/mavros/setup.py"
  "/home/haroon/catkin_ws/build/mavros/catkin_generated/setup_py_interrogation.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_python_setup.cmake:46 (safe_execute_process)
  CMakeLists.txt:65 (catkin_python_setup)

```

vooon, 2024-07-23:
Try to clean build first.

peakyquest, 2024-07-23:
I have tried the below command, still getting the same error. 
```
catkin build
```

peakyquest, 2024-07-23:
is there anything else that I can do @vooon

peakyquest, 2024-07-23:
```
[build] Summary: 3 of 7 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 3 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 12.2 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

peakyquest, 2024-07-23:
nevermind I update the packages, and it works

```
pip install --upgrade importlib-metadata setuptools
```

closed 2024-07-23 by peakyquest