mavlink/mavros #1974: Unable to switch to `Arm` and `Offboard` via `offb_node.cpp` example

HuaYuXiao, 2024-08-07:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am running a simulation experienment with `Gazebo11`, adn ran into some problems, please offer me some help or hint, ThankU so much!

With mavros official example `offb_node.cpp`, I cannot switch to neither `Arm` mode nor `Offboard` mode, although it says `[INFO] Offboard enabled.`

Additional info:

1. When I check node graph with `rqt_graph`, I find `offb_node` ONLY subs to `/mavros/state`, BUT the pubed topic `/mavros/set_position/local` is NOT subed by `mavros`.
2. output of `rostopic echo /mavros/state`:
```
connected: True
armed: False
guided: True
manual_input: False
mode: "AUTO.LOITER"
system_status: 3
```
3. I cannot switch mode via `rosservice call ...` neither, although its feedback tells me that response is properly sent.
4. I cannot switch mode to `Offboard` via `QGroundControl`, but `Arm` is OK.

### MAVROS version and platform

Mavros: 1.19.0
ROS: noetic
Ubuntu: 20.04

### Autopilot type and version

[  ] ArduPilot
[ yes ] PX4

Version: v1.12.3

### Node logs

```
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[INFO] [1723015493.847271, 0.000000]: Loading model XML from file /home/hyx020222/EasonDrone/Simulator/EasonDrone_Gazebo/gazebo_models/uav_models/p450_Eason/p450_Eason.sdf
[INFO] [1723015493.849924, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1723015493.852662, 10.403000]: Calling service /gazebo/spawn_sdf_model
[INFO] [1723015497.169747, 10.538000]: Spawn status: SpawnModel: Successfully spawned entity
[p450_Eason_spawn-2] process has finished cleanly
log file: /home/hyx020222/.ros/log/1cb95b32-548e-11ef-b54c-21755298ea9c/p450_Eason_spawn-2*.log
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-08-07/07_24_59.ulg
INFO  [logger] Opened full log file: ./log/2024-08-07/07_24_59.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
[... 235 lines cut ...]
        key: "Last RTT (ms)"
        value: "0.000000"
      - 
        key: "Mean RTT (ms)"
        value: "0.178949"
      - 
        key: "Last remote time (s)"
        value: "434.712000000"
      - 
        key: "Estimated time offset (s)"
        value: "10.541446606"
```

### Check ID

```
hyx020222@thinkbook-14:~$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 6171 messages, from 2 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 4, 8, 140, 141, 147, 24, 30, 31, 32, 33, 36, 74, 331, 77, 83, 340, 85, 230, 105, 111, 241, 242, 245
 255:190   0
```

vooon, 2024-08-07:
Can you call arming manually? What FC reports?

Because it looks like `INFO  [tone_alarm] notify negative` + ` "rc receiver": "Fail"` makes me thinking that you haven't pass prearm checks in the firmware.

HuaYuXiao, 2024-08-07:
> Can you call arming manually? What FC reports?
> 
> Because `INFO [tone_alarm] notify negative` + ` "rc receiver": "Fail"` makes me thinking that you haven't pass prearm checks in the firmware.

ThankU for your response! Unfortunately I have no physical RC controller with me, so I try to `Arm` quadrotor from scratch via `QGroundControl`.

I successfully `Arm` the vehicle. However, besides the previous `[tone_alarm] notify negative`, I get two more warnings:

1st, before I Arm, 

```
WARN  [mc_pos_control] Cruise speed has been constrained by max speed
WARN  [mc_pos_control] Manual speed has been constrained by max speed
```

2nd, after I Arm,

```
[ WARN] [1723020331.689537117, 25.750000000]: CMD: Unexpected command 400, result 0
```

At the same time, I am trying to solve these 2 problems as U mention:

```
key: "rc receiver"
value: "Fail"

key: "pre-arm check status. Always healthy when armed"
value: "Ok"
```

Full output is like:

```
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_x.main.mix on /dev/pwm_output0
  PWM_AUX_OUT: curr: 0 -> new: 1234
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
WARN  [mc_pos_control] Cruise speed has been constrained by max speed
WARN  [mc_pos_control] Manual speed has been constrained by max speed
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-08-07/08_45_03.ulg
INFO  [logger] Opened full log file: ./log/2024-08-07/08_45_03.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
[ INFO] [1723020303.906988619, 0.318000000]: udp0: Remote address: 127.0.0.1:14580
[ INFO] [1723020303.907061456, 0.318000000]: IMU: High resolution IMU detected!
INFO  [px4] Startup script returned successfully
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known positionINFO  [ecl/EKF] reset position to last known position

INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] GPS checks passed
INFO  [ecl/EKF] GPS checks passed
INFO  [ecl/EKF] GPS checks passed
INFO  [ecl/EKF] GPS checks passed
INFO  [ecl/EKF] GPS checks passed
INFO  [ecl/EKF] GPS checks passed
[ INFO] [1723020304.988333782, 1.290000000]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1723020304.988844979, 1.290000000]: IMU: High resolution IMU detected!
[ INFO] [1723020305.995137846, 2.294000000]: GF: Using MISSION_ITEM_INT
[ INFO] [1723020305.995185029, 2.294000000]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1723020305.995208745, 2.294000000]: VER: 1.1: Flight software:     010c0300 (2e8918da66000000)
[ INFO] [1723020305.995233840, 2.294000000]: VER: 1.1: Middleware software: 010c0300 (2e8918da66000000)
[ INFO] [1723020305.995257982, 2.294000000]: VER: 1.1: OS software:         050f00ff (bf660cba2af81f05)
[ INFO] [1723020305.995279872, 2.294000000]: VER: 1.1: Board hardware:      00000001
[ INFO] [1723020305.995302390, 2.294000000]: VER: 1.1: VID/PID:             0000:0000
[ INFO] [1723020305.995325247, 2.294000000]: VER: 1.1: UID:                 4954414c44494e4f
INFO  [ecl/EKF] 6820000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
[ INFO] [1723020310.785990942, 7.078000000]: IMU: Attitude quaternion IMU detected!
INFO  [ecl/EKF] 6876000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6880000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6884000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6880000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 6884000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
[ INFO] [1723020321.104654254, 16.291000000]: GF: mission received
INFO  [commander] Armed by external command
[ WARN] [1723020331.689537117, 25.750000000]: CMD: Unexpected command 400, result 0
```

HuaYuXiao, 2024-08-07:
Update: Now I am able to `Arm` with my program, but still have no idea about the reson why `Offboard` fails.

Here is part of my `px4ctrl_node.cpp`:
(variables are declared in .h files, initialized in .cpp file):

```
int main(int argc, char **argv){
    ros::init(argc, argv, "px4ctrl_node");
    ros::NodeHandle nh("~");

    state_sub = nh.subscribe<mavros_msgs::State>
            ("/mavros/state", 10, state_cb);

    local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("/mavros/setpoint_position/local", 10);

    arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("/mavros/cmd/arming");
    set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("/mavros/set_mode");

    //the setpoint publishing rate MUST be faster than 2Hz
    ros::Rate rate(50.0);

    // wait for FCU connection
    while(ros::ok() && !current_state.connected){
        ros::spinOnce();
        rate.sleep();
    }

    pose.header.frame_id = "map";
    pose.pose.position.x = 0;
    pose.pose.position.y = 0;
    pose.pose.position.z = 0;

    //send a few setpoints before starting
    for(int i = 100; ros::ok() && i > 0; --i){
        local_pos_pub.publish(pose);
        ros::spinOnce();
        rate.sleep();
    }

    offb_set_mode.request.custom_mode = "AUTO.LOITER";

    arm_cmd.request.value = true;

while(ros::ok()){
        switch (Command_Now.Mode){
            case easondrone_msgs::ControlCommand::Arm:{
                cout << "------------------------" << endl;
                cout_color("FSM_EXEC_STATE: Arm", BLUE_COLOR);

                if (!current_state.armed) {
                    arm_cmd.request.value = true;

                    if (arming_client.call(arm_cmd) &&
                        arm_cmd.response.success) {
                        cout_color("Arm response success", YELLOW_COLOR);
                    }
                    else{
                        cout_color("Vehicle Arm failed", RED_COLOR);
                    }
                }
                else{
                    cout_color("Vehicle already Armed", GREEN_COLOR);
                }

                break;
            }

            case easondrone_msgs::ControlCommand::Offboard:{
                cout << "------------------------" << endl;
                cout_color("FSM_EXEC_STATE: Offboard", BLUE_COLOR);

                if (current_state.mode != "OFFBOARD") {
                    pose.pose.position.x = 0;
                    pose.pose.position.y = 0;
                    pose.pose.position.z = 0;

                    //send a few setpoints before starting
                    for(int i = 100; ros::ok() && i > 0; --i){
                        pose.header.stamp = ros::Time::now();
                        local_pos_pub.publish(pose);
                        ros::spinOnce();
                        rate.sleep();
                    }

                    offb_set_mode.request.custom_mode = "OFFBOARD";

                    if (set_mode_client.call(offb_set_mode) &&
                        offb_set_mode.response.mode_sent) {
                        cout_color("Offboard response sent", YELLOW_COLOR);
                    }
                    else{
                        cout_color("Offboard enable failed", RED_COLOR);
                    }
                }
                else{
                    cout_color("Offboard already enabled", GREEN_COLOR);
                }

                break;
            }
}

        ros::spinOnce();
        rate.sleep();
    }

    return 0;
}
```

I first `Arm`, result is sueecss. Then `Offboard`, but it stuck in `cout_color("Offboard response sent", YELLOW_COLOR);` part, never enter `cout_color("Offboard already enabled", GREEN_COLOR);`.

vooon, 2024-08-07:
You can disable some of pre-arm checks via firmware parameters (check manual).
Ensure that you have continuous stream of offboard commands before you switch to OFFBOARD.

HuaYuXiao, 2024-08-07:
(For my own purpose, save some useful blogs here)

- https://github.com/PX4/PX4-Autopilot/issues/20667
- https://discuss.px4.io/t/offboard-mode-in-sitl/25727
- https://github.com/PX4/PX4-Autopilot/issues/19349
- https://discuss.px4.io/t/offboard-mode-not-working-in-sitl/31675
- https://docs.px4.io/v1.12/en/advanced_config/parameter_reference.html
- https://github.com/PX4/px4_ros_com/issues/111

HuaYuXiao, 2024-08-07:
Finally, I have already SOLVED this issue! 

It turns out that, since I have no RC controller, I have to alternatively CONTINUOUSLY publish something to `/mavros/setpoint_raw/local` or `/mavros/setpoint_raw/global`.

I find out cause because of following warnings when I switch to `Takeoff` under `Arm` mode. 

```
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
[ INFO] [1723040965.948630253, 16.679000000]: GF: mission received
[ WARN] [1723040972.904043053, 23.334000000]: CMD: Unexpected command 22, result 0
INFO  [commander] Armed by external command
[ WARN] [1723040972.915859524, 23.346000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.022730198, 23.446000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.123663606, 23.546000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.225413643, 23.646000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.428666766, 23.746000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.533654839, 23.846000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.740070792, 23.946000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.841237773, 24.046000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040973.938943105, 24.146000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040974.044466365, 24.246000000]: PositionTargetGlobal failed because no origin
[ WARN] [1723040974.142026869, 24.346000000]: PositionTargetGlobal failed because no origin
INFO  [commander] Takeoff detected
```

With previous infomation, this hints me that, setpoition command SHOULD be sent to `/mavros/setpoint_raw/global` RATHER THAN `/mavros/set_position/local`.

closed 2024-08-07 by HuaYuXiao