mavlink/mavros #1977: Cant create composable node for mavros UAS

TitikD, 2024-08-21:
### Issue details

Cant create composable node for mavros UAS. Router executing normaly.

### MAVROS version and platform

Mavros: ?2.8.0?
ROS: ?Humble?
Ubuntu: ?22.04?

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: ?4.4.0?

### launch.py

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, GroupAction
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import ComposableNodeContainer, Node, PushRosNamespace, SetParameter
from launch_ros.descriptions import ComposableNode
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument('name', default_value='drone1', description='Name of the drone'),
        DeclareLaunchArgument('fcu_url', default_value='udp://:14550@', description='FCU URL'),
        DeclareLaunchArgument('target_system_id', default_value='1', description='Target System ID'),
        DeclareLaunchArgument('respawn', default_value='false', description='Respawn flag'),
        
        SetParameter(name='use_sim_time', value=True),
        
        GroupAction([
            PushRosNamespace([LaunchConfiguration('name')]),
            ComposableNodeContainer(
                name='drone_container',
                namespace='',
                package='rclcpp_components',
                executable='component_container',
                composable_node_descriptions=[
                    ComposableNode(
                        package='mavros',
                        plugin='mavros::uas::UAS',
                        name='uas',
                        parameters=[
                            {'fcu_system_id': 1},
                            {'fcu_component_id': 1},
                            {'target_system_id': 1},
                            {'target_component_id': 1},
                        ],
                    ),
                ],
                output='screen',
            )
        ])
    ])

### Node logs

[INFO] [launch]: All log files can be found below /home/user/.ros/log/2024-08-21-19-21-20-604944-userPC-72709
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [72721]
[component_container-1] [INFO] [1724257280.921034423] [drone1.drone_container]: Load Library: /home/user/ws/install/mavros/lib/libmavros.so
[component_container-1] [INFO] [1724257280.925046203] [drone1.drone_container]: Found class: rclcpp_components::NodeFactoryTemplate<mavros::router::Router>
[component_container-1] [INFO] [1724257280.925065939] [drone1.drone_container]: Found class: rclcpp_components::NodeFactoryTemplate<mavros::uas::UAS>
[component_container-1] [INFO] [1724257280.925071263] [drone1.drone_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<mavros::uas::UAS>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/drone1/uas' in container '/drone1/drone_container'
[... 14 lines cut ...]
[component_container-1] [INFO] [1724257281.007185847] [drone1.uas]: Plugin camera created
[component_container-1] [INFO] [1724257281.007310529] [drone1.uas]: Plugin camera initialized
[component_container-1] [WARN] [1724257281.007470617] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[component_container-1] [INFO] [1724257281.010718948] [drone1.uas]: Plugin cellular_status created
[component_container-1] [INFO] [1724257281.010751379] [drone1.uas]: Plugin cellular_status initialized
[component_container-1] [WARN] [1724257281.010936277] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[component_container-1] [INFO] [1724257281.019837218] [drone1.uas]: Plugin command created
[component_container-1] [INFO] [1724257281.020011255] [drone1.uas]: Plugin command initialized
[component_container-1] [WARN] [1724257281.020252122] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[component_container-1] 
[component_container-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[component_container-1] This error state is being overwritten:
[component_container-1] 
[component_container-1]   'create_subscription() called for existing topic name rt/drone1/drone_container/status with incompatible type mavros_msgs::msg::dds_::CompanionProcessStatus_, at ./src/subscription.cpp:146, at ./src/rcl/subscription.c:109'
[component_container-1] 
[component_container-1] with this new error message:
[component_container-1] 
[component_container-1]   'invalid allocator, at ./src/rcl/subscription.c:219'
[component_container-1] 
[component_container-1] rcutils_reset_error() should be called after error handling to avoid this.
[component_container-1] <<<
[component_container-1] invalid allocator, at ./src/rcl/subscription.c:219
[component_container-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[component_container-1]   what():  could not create subscription: invalid allocator, at ./src/rcl/subscription.c:219
[ERROR] [component_container-1]: process has died [pid 72721, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=drone_container -r __ns:=/drone1 -p use_sim_time:=True'].

astomnbkn, 2024-08-28:
I have the same issue. Does anyone know how to resolve this?

Using composable node from libmavros or custom container node implementation including one Router and two UAS,
either way launching ends up in the same error of node name and topic names screwd up with 'image_container' token.
I don't think having two UAS nodes is not relavant in latter case.
Using python launch file or xml launch file gives the same situation.
And the existance of Router node does not affect the situation neither.

Right after launch failes, node list and topic list shows somehting like below.
And launch.py and logs are attached below from the former case launching 1 UAS.

```console
$ ros2 node list
WARNING: Be aware that there are nodes in the graph that share an exact name, which can have unintended side effects.
/ComponentManager
/image_container
/mavros1
/mavros1/image_container
/mavros1/image_container
/mavros1/image_container
/mavros1/image_container
/mavros1/image_container
/mavros1/image_container
/mavros1/image_container
/mavros1/image_container
/transform_listener_impl_5a6614445e58
$ ros2 topic list
/diagnostics
/mavros1/actuator_control
/mavros1/altitude
/mavros1/image_container/cam_imu_stamp
/mavros1/image_container/image_captured
/mavros1/image_container/send
/mavros1/image_container/status
/mavros1/image_container/vehicle
/mavros1/target_actuator_control
/parameter_events
/rosout
/tf
/tf_static
```

### launch.py

```python
from launch import LaunchDescription
from launch_ros.actions import LoadComposableNodes, Node
from launch_ros.descriptions import ComposableNode
from launch_ros.actions import PushROSNamespace
from launch.actions import GroupAction

def generate_launch_description():
    container = Node(
        name='image_container',
        package='rclcpp_components',
        executable='component_container',
        output='both',
    )

    pluginlists_yaml = "/opt/ros/iron/share/mavros/launch/apm_pluginlists.yaml"
    config_yaml = "/opt/ros/iron/share/mavros/launch/2uas_apm_config.yaml"

    params1 = {
        "target_system_id": 1,
        "target_component_id": 1,
        "system_id": 1,
        "component_id": 191,
        "uas_url": "/uas1",
        "log_output": "screen",
        "fcu_protocol": "v2.0"
[... 55 lines cut ...]
[component_container-1]   'create_subscription() called for existing topic name rt/mavros1/image_container/status with incompatible type mavros_msgs::msg::dds_::CompanionProcessStatus_, at ./src/subscription.cpp:517, at ./src/rcl/subscription.c:112'
[component_container-1]
[component_container-1] with this new error message:
[component_container-1]
[component_container-1]   'invalid allocator, at ./src/rcl/subscription.c:261'
[component_container-1]
[component_container-1] rcutils_reset_error() should be called after error handling to avoid this.
[component_container-1] <<<
[component_container-1] invalid allocator, at ./src/rcl/subscription.c:261
[component_container-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[component_container-1]   what():  could not create subscription: invalid allocator, at ./src/rcl/subscription.c:261
[ERROR] [component_container-1]: process has died [pid 160857, exit code -6, cmd '/opt/ros/iron/lib/rclcpp_components/component_container --ros-args -r __node:=image_container'].
```
### MAVROS version and platform

Mavros: 2.8.0
ROS: Iron
Ubuntu: 22.04

### Autopilot type and version

[X] ArduPilot
[ ] PX4

Version: 4.5.1

amarburg, 2025-08-21:
Apologies that this is a non-answer, but I am able to launch multiple UAS nodes (and a Router node in the same container) when using the [Zenoh middleware](https://github.com/ros2/rmw_zenoh).   

I _was_ able to replicate this issue when I inadvertantly unset `RMW_IMPLEMENTATION` and reverted to the default middleware layer (which is how I found this issue)